ros2/ros2 #635: support building a subset of packages with static libraries

dirk-thomas, 2018-12-27:
This has been worked on in the past in #468 and was working at some point in time.

Currently it seems to be impossible again to build the demos / examples with static libraries, see colcon/colcon-core#140.

It should be possible to build at least the C++ demos and examples with static libraries using an invocation like this (assuming that only one RMW impl is available, e.g. FastRTPS): 

```
colcon build \
  --packages-up-to <demo/example-pkgs> \
  --packages-ignore rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp rosidl_generator_py rmw_fastrtps_dynamic_cpp \
  --cmake-args -DBUILD_SHARED_LIBS=OFF --no-warn-unused-cli
```

dirk-thomas, 2018-12-27:
@eeyang92 FYI

eeyang92, 2018-12-27:
Thanks @dirk-thomas!

dirk-thomas, 2019-11-21:
Latest example command: `colcon build --build-base build-static --install-base install-static --packages-up-to demo_nodes_cpp --packages-ignore rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp rosidl_generator_py rmw_fastrtps_dynamic_cpp --packages-ignore-regex .*cyclonedds.* .*connext.* .*opensplice.* --cmake-args -DBUILD_SHARED_LIBS=OFF --no-warn-unused-cli`

franzrammerstorfer, 2020-08-11:
Unfortunately, the above example command does not work, at least with my settings on a Debian buster docker container, foxy. It seems, that there is a CMake problem when trying to compile a package containing messages, in my case, it's the unique_identifier package:

CMake Error: install(EXPORT "unique_identifier_msgs__rosidl_typesupport_c" ...) includes target "unique_identifier_msgs__rosidl_typesupport_c" which requires target "unique_identifier_msgs__rosidl_typesupport_fastrtps_c" that is not in the export set.
CMake Error: install(EXPORT "unique_identifier_msgs__rosidl_typesupport_cpp" ...) includes target "unique_identifier_msgs__rosidl_typesupport_cpp" which requires target "unique_identifier_msgs__rosidl_typesupport_fastrtps_cpp" that is not in the export set.

In the dynamic case (-DBUILD_SHARED_LIBS=ON), ie. when I do not have to ignore rosidl_typesupport_introspection_* packages, it works out fine.
Is anybody working at that issue, maybe somebody is willing to give me a hint, I'd really love to get a static binary out of ros2. Thanks!

franzrammerstorfer, 2020-09-08:
For debugging I added the flag --executor sequential to colcon build as well as dbg variables to cmake (--trace-source=CMakeLists.txt --trace-expand) and got the following: 

ros2/rmw_dds_common/rmw_dds_common/CMakeLists.txt(105):  ament_package()
CMake Error: install(EXPORT "rmw_dds_common__rosidl_typesupport_c" ...) includes target "rmw_dds_common__rosidl_typesupport_c" which requires target "rmw_dds_common__rosidl_typesupport_fastrtps_c" that is not in the export set.
CMake Error: install(EXPORT "rmw_dds_common__rosidl_typesupport_cpp" ...) includes target "rmw_dds_common__rosidl_typesupport_cpp" which requires target "rmw_dds_common__rosidl_typesupport_fastrtps_cpp" that is not in the export set.

Unfortunately, my cmake knowledge is limited and I could not find any documentation of the build system, so I'm not able to figure out where/how to export the requested targets. Hopefully somebody is willing to give me a hint...

cwyark, 2020-11-06:
> For debugging I added the flag --executor sequential to colcon build as well as dbg variables to cmake (--trace-source=CMakeLists.txt --trace-expand) and got the following: [...]

I guess rosidl_typesupport_fastrtps_c has been excluded from the ignored target (rmw_fastrtps_shared_cpp)

see the package dependency https://github.com/ros2/rmw_fastrtps/blob/4e0fce977c993f840b013c444d603842fb39ad64/rmw_fastrtps_dynamic_cpp/package.xml#L27

helloLoey, 2024-02-01:
I build ros2 static lib on Windows 11, but get the following output:
Summary: 90 packages finished [4min 57s]
  7 packages failed: ament_cmake_vendor_package launch_testing_ament_cmake pluginlib rosidl_dynamic_typesupport rosidl_pycommon uncrustify_vendor urdfdom
  83 packages had stderr output: ament_clang_format ament_clang_tidy ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gen_version_h ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_pclint ament_cmake_pep257 ament_cmake_pycodestyle ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_target_dependencies ament_cmake_test ament_cmake_version ament_cmake_xmllint ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_auto ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_xmllint domain_coordinator eigen3_cmake_module foonathan_memory_vendor gmock_vendor google_benchmark_vendor gtest_vendor iceoryx_binding_c iceoryx_hoofs iceoryx_posh ignition_cmake2_vendor ignition_math6_vendor launch launch_pytest launch_testing launch_xml launch_yaml mcap_vendor osrf_pycommon osrf_testing_tools_cpp pybind11_vendor ros_environment rosidl_cli rpyutils shared_queues_vendor sqlite3_vendor test_interface_files test_osrf_testing_tools_cpp tinyxml_vendor tracetools_read tracetools_trace urdfdom_headers zstd_vendor
  257 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

The build command is : 
colcon build --build-base build-static --install-base install-static --packages-ignore rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp rosidl_generator_py rmw_fastrtps_dynamic_cpp --packages-ignore-regex .*cyclonedds.* .*connext.* .*opensplice.* --cmake-args -DBUILD_SHARED_LIBS=OFF --no-warn-unused-cli --merge-install --continue-on-error --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo

How to solve this problem?