ros2/ros2 #636: failing rosbag2 installation

vkuehn, 2018-12-27:
## Bug report

**Required Info:**

- Operating System:
  NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
- Installation type:
  https://index.ros.org/doc/ros2/Linux-Install-Debians/
- Version or commit hash:
  crystal
- DDS implementation:
  what comes with install instructions
- Client library (if applicable):
  - rosbag

#### Steps to reproduce issue
install on a fresh ubuntu 18.04 according to instructions

```

#### Expected behavior
rosbag2 is installed with full desktop and works fine
#### Actual behavior
even follwing the instructions
https://index.ros.org/r/rosbag2/github-ros2-rosbag2/#crystal
with sudo apt-get install ros2bag rosbag2*
brings lonng error list
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-crystal-rosbag2-test-common' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-default-plugins-dbgsym' for regex 'rosbag2*'
Note, selecting 'python-rosbag' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-dbgsym' for regex 'rosbag2*'
Note, selecting 'librosbag-storage-dev' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-default-plugins' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-converter-default-plugins' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-tests' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-dbgsym' for regex 'rosbag2*'
Note, selecting 'librosbag3d' for regex 'rosbag2*'
Note, selecting 'librosbag-dev' for regex 'rosbag2*'
Note, selecting 'librosbag-storage2d' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-converter-default-plugins-dbgsym' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-transport-dbgsym' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-transport' for regex 'rosbag2*'
E: Unable to locate package ros2bag

or this sudo apt-get install rosbag2*
fails with
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-crystal-rosbag2-test-common' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-default-plugins-dbgsym' for regex 'rosbag2*'
Note, selecting 'python-rosbag' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-dbgsym' for regex 'rosbag2*'
Note, selecting 'librosbag-storage-dev' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-default-plugins' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-converter-default-plugins' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-tests' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-storage-dbgsym' for regex 'rosbag2*'
Note, selecting 'librosbag3d' for regex 'rosbag2*'
Note, selecting 'librosbag-dev' for regex 'rosbag2*'
Note, selecting 'librosbag-storage2d' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-converter-default-plugins-dbgsym' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-transport-dbgsym' for regex 'rosbag2*'
Note, selecting 'ros-crystal-rosbag2-transport' for regex 'rosbag2*'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-rosbag : Depends: python-roslib but it is not going to be installed
                 Depends: python-rospy but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

Karsten1987, 2018-12-27:
This is indeed broken, because I haven't been able to manage to successfully deploy the ros1 dependencies for the rosbag2 converters.

Could you try compiling the rosbag stuff from source?

vkuehn, 2018-12-28:
did try but colcon is also missing
installing it with 
`sudo apt install colcon `
is answered 
`Unable to locate package colcon`
can you kindly also check that and update build instructions ?
sorry I am an absolute noob in terms of ros2 internals

dirk-thomas, 2018-12-28:
> did try but colcon is also missing
> installing it with
> `sudo apt install colcon`
> is answered
> `Unable to locate package colcon`
> can you kindly also check that and update build instructions ?

The Debian package you should install is `python3-colcon-common-extensions` as mentioned in the [install instructions](https://index.ros.org/doc/ros2/Linux-Development-Setup/#install-development-tools-and-ros-tools).

vkuehn, 2018-12-29:
can you please check rosbag2 install from source page if it than makes sense (installed ros2 for debians before)  to research the bug here ?

vkuehn, 2018-12-29:
retried...unfortunatetly colcon build --merge-install fails with
Starting >>> rosbag2_test_common
Starting >>> shared_queues_vendor
Finished <<< shared_queues_vendor [1.19s]                                  
Starting >>> ros1_rosbag_storage_vendor
Finished <<< rosbag2_test_common [1.24s]
Starting >>> rosbag2_storage
--- stderr: ros1_rosbag_storage_vendor
CMake Warning at /opt/ros/crystal/share/class_loader/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'class_loader' exports library 'class_loader' which couldn't be
  found
Call Stack (most recent call first):
  /opt/ros/crystal/share/class_loader/cmake/class_loaderConfig.cmake:38 (include)
  /opt/ros/crystal/share/pluginlib/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/crystal/share/pluginlib/cmake/pluginlibConfig.cmake:38 (include)
  tools/rosbag_storage/CMakeLists.txt:20 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BZip2 (missing: BZIP2_LIBRARIES BZIP2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindBZip2.cmake:49 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  tools/rosbag_storage/CMakeLists.txt:23 (find_package)

make[2]: *** [ros1_rosbag_storage-prefix/src/ros1_rosbag_storage-stamp/ros1_rosbag_storage-configure] Error 1
make[1]: *** [CMakeFiles/ros1_rosbag_storage.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_rosbag_storage_vendor	[ Exited with code 2 ]
Aborted  <<< rosbag2_storage                                          
                                 
Summary: 2 packages finished [2.48s]
  1 package failed: ros1_rosbag_storage_vendor
  1 package aborted: rosbag2_storage
  1 package had stderr output: ros1_rosbag_storage_vendor
  8 packages not processed

dirk-thomas, 2018-12-29:
The error message `Could NOT find BZip2` sounds like you haven't installed the dependencies declared in the [package manifest](https://github.com/ros2/rosbag2/blob/8250dec3b597fdb3f646ed089439a8d9cfd301c8/ros1_rosbag_storage_vendor/package.xml#L14). The install instructions describe how to involve `rosdep` to do so.

vkuehn, 2018-12-29:
bzip2 lowercase is mentioned in the package.xml and installing is anwered with
bzip2 is already the newest version (1.0.6-8.1)
is it possible that there is a typo in one of the install scripts ?

dirk-thomas, 2018-12-30:
The `rosdep` key `bzip2` maps to [libbz2-dev](https://github.com/ros/rosdistro/blob/144de18e16160293d7a3b0930be1f14f386a1f5c/rosdep/base.yaml#L354). Are you sure you have that installed too?

vkuehn, 2018-12-30:
libconsole-bridge-dev is already the newest version (0.4.0+dfsg-2)
checked all the other in package manifest. 
Only libgpgme-dev was missing. Unfortunatetly same result after restarting colcon build

LanderU, 2018-12-30:
I have rosbag2 compiled from sources on Ubuntu 16.04, I installed `libgpgme-dev` via: https://www.ubuntuupdates.org/package/kubuntu-ppa_backports/xenial/main/base/libgpgme-dev

And now I can use rosbag2 without problems.

vkuehn, 2018-12-30:
I am on ubuntu 18.04 as crystal refers to it..though. Did you also install ros2 and nothing else before ?

LanderU, 2018-12-30:
No, have compiled all ros2 packages using the guide. Also, I only have installed the dependencies listed on the installation guide. Additionally I was having some linking problems because of `libgpgme`. But, apart from this, I didn't install nothing else.

I didn't try it using the .deb packages.

vkuehn, 2018-12-30:
for me I get
libgpgme-dev is already the newest version (1.10.0-1ubuntu2).
something else is missing when installing crystal desktop and than rosbag2
@Karsten1987 @dirk-thomas 
can you kindly check ?
maybe you could make rosbag2 also part of the desktop installation ?

Karsten1987, 2019-01-08:
@vkuehn can you tell what's the actual error message you're running into by now? I've noticed that you installed a couple of packages and dependencies and I just wanted to make sure which dependency is missing right now in your setup.

vkuehn, 2019-01-11:
Now I get this
Starting >>> rosbag2_test_common
Starting >>> shared_queues_vendor
Finished <<< rosbag2_test_common [2.12s]                                   
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [8.58s]                                        
Starting >>> ros1_rosbag_storage_vendor
--- stderr: ros1_rosbag_storage_vendor                                      
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BZip2 (missing: BZIP2_LIBRARIES BZIP2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindBZip2.cmake:49 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  tools/rosbag_storage/CMakeLists.txt:23 (find_package)

make[2]: *** [ros1_rosbag_storage-prefix/src/ros1_rosbag_storage-stamp/ros1_rosbag_storage-configure] Error 1
make[1]: *** [CMakeFiles/ros1_rosbag_storage.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_rosbag_storage_vendor	[ Exited with code 2 ]
Aborted  <<< rosbag2_storage                                           

Summary: 2 packages finished [19.2s]
  1 package failed: ros1_rosbag_storage_vendor
  1 package aborted: rosbag2_storage
  1 package had stderr output: ros1_rosbag_storage_vendor
  8 packages not processed

LanderU, 2019-01-11:
You've a missing library, try it installing it: https://launchpad.net/ubuntu/bionic/+package/libbz2-dev

vkuehn, 2019-01-11:
mitigated since sudo apt-get install libbz2-dev

vkuehn, 2019-01-11:
now I get 
Starting >>> rosbag2_test_common
Starting >>> shared_queues_vendor
Finished <<< rosbag2_test_common [0.86s]                                   
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [1.02s]                                        
Starting >>> ros1_rosbag_storage_vendor
--- stderr: ros1_rosbag_storage_vendor                                    
CMake Error at tools/rosbag_storage/CMakeLists.txt:25 (find_package):
  By not providing "Findros1_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ros1_bridge", but CMake did not find one.

  Could not find a package configuration file provided by "ros1_bridge" with
  any of the following names:

    ros1_bridgeConfig.cmake
    ros1_bridge-config.cmake

  Add the installation prefix of "ros1_bridge" to CMAKE_PREFIX_PATH or set
  "ros1_bridge_DIR" to a directory containing one of the above files.  If
  "ros1_bridge" provides a separate development package or SDK, be sure it
  has been installed.

make[2]: *** [ros1_rosbag_storage-prefix/src/ros1_rosbag_storage-stamp/ros1_rosbag_storage-configure] Error 1
make[1]: *** [CMakeFiles/ros1_rosbag_storage.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_rosbag_storage_vendor	[ Exited with code 2 ]
Aborted  <<< rosbag2_storage                                           

Summary: 2 packages finished [2.66s]
  1 package failed: ros1_rosbag_storage_vendor
  1 package aborted: rosbag2_storage
  1 package had stderr output: ros1_rosbag_storage_vendor
  8 packages not processed

LanderU, 2019-01-11:
Do you have ros1_bridge compiled? It looks like you didn't compile it.

vkuehn, 2019-01-11:
after
sudo apt-get install ros2bag rosbag2*
failed I followed these instructions [rosbag2](https://index.ros.org/r/rosbag2/github-ros2-rosbag2/#crystal)
can you kindly check what else is missing ? and why on erth do I need anything from ros 1 ??

dirk-thomas, 2019-01-11:
> and why on earth do I need anything from ros 1 ??

As the page you referenced states:

> rosbag2 provides support for legacy ROS 1 bag data

And in order to reuse the logic already available in ROS 1 to read ROS 1 bags the [ros1_rosbag_storage_vendor](https://index.ros.org/p/ros1_rosbag_storage_vendor/github-ros2-rosbag2/#crystal-deps) depends on the `ros1_bridge` to do the conversion.

vkuehn, 2019-01-11:
ok thanks, for a rooky like me the Instructions just didn't work out for on ubuntu 18.04
I would have thought that there is additional support for ros 1 bag data.
where to go from here ? shouldn't the sudo apt install instructions just work ?
just closing this would not help the community right ?

nuclearsandwich, 2019-01-11:
> maybe you could make rosbag2 also part of the desktop installation ?

The desktop variant is specified in http://www.ros.org/reps/rep-2001.html. You can propose changes on GitHub https://github.com/ros-infrastructure/rep/blob/master/rep-2001.rst

Karsten1987, 2019-01-12:
@vkuehn I just double checked with a fresh ubuntu. I believe this PR should fix it:
https://github.com/ros2/rosbag2/pull/83
Do you mind giving this a try?

You're correct the current instructions fail because `ros2bag` will not be resolved correctly. The `*` will resolve it correctly to `ros-crystal-ros2bag`.

vkuehn, 2019-01-14:
Sorry busy again but will do this week

vkuehn, 2019-01-15:
after sudo apt update && sudo apt dist-upgrade I have tried a suggested
sudo apt install ros2bag* rosbag2*
still unmet dependencies
python-rosbag : Depends: python-roslib but it is not going to be installed
                 Depends: python-rospy but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
python-roslib : Depends: catkin but it is not going to be installed
catkin : Depends: python-catkin-pkg but it is not going to be installed

the machine was the same as before so not clean. I'll check later this week with 18.04 freshly

vkuehn, 2019-01-16:
apt-cache rdepends rosbag*  brings back a lot references to ros-crystal*
to solve this issue you could change the read.me to
sudo apt install **ros-$CHOOSE_ROS_DISTRO**-ros2bag* **ros-$CHOOSE_ROS_DISTRO**-rosbag2*
I found no other solution even on a freshly installed machine

Karsten1987, 2019-01-16:
@vkuehn thanks for iterating over this. Your changes are reflected in https://github.com/ros2/rosbag2/commit/733821089d93b30fa06455d7d6dd1a6291c83e74

I'll close this ticket.

closed 2019-01-16 by Karsten1987