ros2/ros2 #638: ROS 2 Crystal Clemmys Patch Release 1

clalancette, 2019-01-03:
This is a meta-issue to track the changes that are going into the first patch release for Crystal Clemmys.  @ros2/team , please add PRs that are going onto the Crystal branches to this ticket for tracking.  @nuclearsandwich FYI.

PRs merged for Crystal Patch Release 1:
-
-

dirk-thomas, 2019-01-07:
I would propose to separate PRs which actually fix bugs and should go into the patch release from the ones which we might sneak in since it just makes life easier - at least visually. Only the first group is basically the reason why we do patch releases. Especially in the `rmw_connext` and `rmw_fastrtps` repos almost all except one per repo seem to be of the later category.

Also the `rosout` related changes are introducing a new feature. In general we only aim to release those into future distros. Otherwise it would require extensive testing to ensure it does neither change or break any existing behavior. I am not saying we can't make an exception but just want to state that this is the general policy in all ROS distros and we should have a good reason to diverge from that.

nuclearsandwich, 2019-01-07:
> I would propose to separate PRs which actually fix bugs and should go into the patch release from the ones which we might sneak in since it just makes life easier

It seems that we want two different things from this list. It sounds like you want the list of PRs which justify the patch release which serve as a draft "release notes/highlights". Whereas I want a list of every change that would be released in the next sync on the coattails of the PRs from the short list so that they can be reviewed for potentially breaking changes. I'll see about hiding these "coattail riders" behind a `<details>` tag once I finish the complete list.

nuclearsandwich, 2019-01-08:
I split out the main body PR. I haven't yet made an explicit list of coattail riders, but it's everything in the last list that isn't mentioned in the first.

nuclearsandwich, 2019-01-12:
Releases still need to be made for
- ament/ament_lint
- ros2/demos
- ros2/sros2
- ros2/rviz

Other repositories have been released with their desired changes from the list of merged PRs summarized by feature/fix and the PRs which rode their coattails into this release. Except to fix major issues, repositories which have been tagged for this patch release will not be re-released until the next patch that has a desired change for that repository.

wjwwood, 2019-01-15:
Here's the ros2/rviz release: https://github.com/ros/rosdistro/pull/19943

nuclearsandwich, 2019-01-16:
Packages are in the testing repository for Ubuntu and #646 contains a repos file of the prospective patch release for people to try.

nuclearsandwich, 2019-01-16:
Copying @jacobperron's comment from https://github.com/ros2/ros2/pull/646#issuecomment-454881273
> Maybe it was overlooked, but there were a couple fixes added to rmw_fastrtps (at least one listed in the description #638) that are not present in the patch release. If the ship has sailed, I think it's okay if they wait for the next patch release/sync.

I did indeed miss that since I'd already done the rmw-layer releases and there's no notification for edits. I think it's best if that waits for the next patch release since it would require a pretty big rebuild to release now. To protect against this in the future I'll update the template for Patch Release 2 to include announcing edits via comment when necessary.

nuclearsandwich, 2019-01-16:
I've updated the list in the main body to note the PRs which missed my release cut-off. Which can be included in a future Crystal patch release, although I haven't yet audited the rmw_fastrtps changes for Crystal compatibility so that repo requires some investigation.

nuclearsandwich, 2019-01-17:
Packaging jobs:

Ubuntu Bionic AMD64 [image]
Ubuntu Bionic ARM64 [image]
Ubuntu Xenial AMD64 [image]
Ubuntu Xenial ARM64 [image]
macOS [image]
Windows Release [image]
Windows Debug [image]

paulbovbel, 2019-01-22:
I've noticed that Crystal patch 1 will not build from source on xenial, due to the missing python3-lark-parser dependency of rosidl_parser. Is `xenial [s]` still a supported platform for the patch 1 release?

dirk-thomas, 2019-01-22:
> Is `xenial [s]` still a supported platform for the patch 1 release?

Yes, it is. The supported platforms do not change for an already released distribution.

> I've noticed that Crystal patch 1 will not build from source on xenial, due to the missing python3-lark-parser dependency of rosidl_parser.

That shouldn't be any different than the original release. On Xenial you will just need to install the Python package using `pip` since we didn't create a Debian package for Xenial (see the [install instructions](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#install-development-tools-and-ros-tools)).

paulbovbel, 2019-01-22:
That's unfortunate. Since it's the only build/run dependency that's not packaged as a deb in xenial, would you consider adding a debian package for it?

dirk-thomas, 2019-01-22:
We don't see a need for it. What is the problem with installing the package via `pip`?

paulbovbel, 2019-01-22:
We're generating debian packages of ROS2 (bionic) for xenial, and we mirror dependencies from packages.ros.org.

dirk-thomas, 2019-01-22:
Ok, I will make sure we generate one.

nuclearsandwich, 2019-01-23:
With thanks to @j-rivero the python3-lark-parser package has been released for Xenial and is hosted on the ROS 2 repositories.