ros2/ros2 #640: Documentation is missing information needed for rosdep

kutschkem, 2019-01-09:
https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#install-dependencies-using-rosdep

is missing the needed export 

`export ROSDISTRO_INDEX_URL='https://raw.githubusercontent.com/ros2/rosdistro/ros2/index.yaml`

The default index does only contain ROS1 distros. I found this command by inspecting https://github.com/ros2/ros2/issues/536

dirk-thomas, 2019-01-09:
Thanks for reporting this. Yes, currently you need to use that custom environment variable in order to use ROS 2 repos.

We are currently in the process of reintegrating these two repos again which should make this step obsolete (see #492). Since we plan to finish that transition in the coming days I would hold off updating the documentation for the in between time.

gavanderhoorn, 2019-01-10:
@dirk-thomas: is #492 the correct reference?

dirk-thomas, 2019-01-10:
> is #492 the correct reference?

Thanks for pointing it out. While the number is correct my original comment lacked the correct repo reference: ros2/rosdistro#492

kutschkem, 2019-01-11:
Seems to work fine now without the export, I was just unlucky that I tried this right before your reintegration.