ros2/ros2 #642: VCS tool is unreliable, consider repo

ThyMYthOS, 2019-01-14:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Source
- Version or commit hash:
  - Clemmys
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Probably cannot be reproduced reliably:
```
vcs import src < ros2.repos
```

#### Expected behavior

All repositories are cloned and checked out to the right version.

#### Actual behavior

Some repositories are not cloned (e.g. observed for FastCDR).

----
## Feature request

Please consider using [repo](https://source.android.com/setup/develop) instead. Is is used for the same purpose as `vcs` for Android and therefor the tool is much more mature.

dirk-thomas, 2019-01-14:
Please post now information, what version of `vcs` you are using, how frequently you are seeing the problem, etc.

> Please consider using repo instead. It is used for the same purpose as `vcs` for Android ...

First of all `vcs` is not limited to `git` so it doesn't seem that `repo` covers the same feature set but only a subset. So it isn't an adequate replacement for all the use cases (as far as I can tell from briefly looking at the docs of `repo`).

> It is used for the same purpose as `vcs` for Android and therefor the tool is much more mature.

I don't follow the rational why the usage in Android would imply any less or more maturity especially with the combination of reporting this issue for Windows.

dirk-thomas, 2019-01-24:
I will go ahead and close this ticket since this is not something which seems feasible (see my previous comment).

closed 2019-01-24 by dirk-thomas