ros2/ros2 #643: yaml and catkin_pkg modules missing for entry points

maziarz, 2019-01-14:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra 10.13.1 (17B1003)
  - Python 3.7.0b2
- Installation type:
  - https://github.com/ros2/ros2/releases/download/release-crystal-20181214/ros2-crystal-20181214-osx-x86_64.tar.bz2
- Version or commit hash:
  -  5e812b7
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1. Follow https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/
2. Running any ros2 command produces these errors.

#### Expected behavior
No errors. I have verified that pyyaml and catkin_pkg is installed.

#### Actual behavior
```
$ ros2
Failed to load entry point 'launch': No module named 'yaml'
Failed to load entry point 'create': No module named 'catkin_pkg'
Failed to load entry point 'call': No module named 'yaml'
Failed to load entry point 'echo': No module named 'yaml'
Failed to load entry point 'pub': No module named 'yaml'
```

dirk-thomas, 2019-01-14:
It looks like you missed the following step in the [install instructions](https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/#installing-prerequisites):

> Install additional runtime dependencies for command-line tools:
> `python3 -m pip install catkin_pkg empy git+https://github.com/lark-parser/lark.git@0.7b pyparsing pyyaml setuptools argcomplete`

Assuming following that step will resolve the problem I will go ahead and close the ticket. If it doesn't please feel free to comment with more information.

closed 2019-01-14 by dirk-thomas

maziarz, 2019-01-14:
@dirk-thomas i already have this step:

```
$ pip show pyyaml
Name: PyYAML
Version: 3.13
Summary: YAML parser and emitter for Python
Home-page: http://pyyaml.org/wiki/PyYAML
Author: Kirill Simonov
Author-email: xi@resolvent.net
License: MIT
Location: /Users/maziarzamani/.pyenv/versions/3.7.0b2/lib/python3.7/site-packages
Requires: 
Required-by: vcstool, colcon-metadata, colcon-defaults
```

```
$ pip show catkin_pkg
Name: catkin-pkg
Version: 0.4.10
Summary: catkin package library
Home-page: http://wiki.ros.org/catkin_pkg
Author: Dirk Thomas
Author-email: dthomas@osrfoundation.org
License: BSD
Location: /Users/maziarzamani/.pyenv/versions/3.7.0b2/lib/python3.7/site-packages
Requires: pyparsing, python-dateutil, docutils
Required-by: colcon-ros
```

dirk-thomas, 2019-01-14:
Can you post the output of:

```
python3 -c "import catkin_pkg; print(catkin_pkg.__version__)"
python3 -c "import yaml; print(yaml.__version__)"
python3 -c "import ros2cli; print(ros2cli.__file__)"
```

maziarz, 2019-01-14:
> Can you post the output of:
> 
> ```
> python3 -c "import catkin_pkg; print(catkin_pkg.__version__)"
> python3 -c "import yaml; print(yaml.__version__)"
> python3 -c "import ros2cli; print(ros2cli.__file__)"
> ```

```
$ python3 -c "import catkin_pkg; print(catkin_pkg.__version__)"
0.4.10
$ python3 -c "import yaml; print(yaml.__version__)"
3.13
$ python3 -c "import ros2cli; print(ros2cli.__file__)"
/Users/maziarzamani/ros2_install/lib/python3.7/site-packages/ros2cli/__init__.py
```

maziarz, 2019-01-14:
> Can you post the output of:
> 
> ```
> cat `which ros2`
> which python3
> ```
> Are you using any custom Python environment?

```
$ cat `which ros2`
#!/usr/local/bin/python3
# EASY-INSTALL-ENTRY-SCRIPT: 'ros2cli==0.6.2','console_scripts','ros2'
__requires__ = 'ros2cli==0.6.2'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('ros2cli==0.6.2', 'console_scripts', 'ros2')()
    )
```
```
$ which python3
/Users/maziarzamani/.pyenv/shims/python3
```

I use https://github.com/pyenv/pyenv

dirk-thomas, 2019-01-14:
The separate Python environments are certainly causing the problem. While you installed the dependencies in on environment they are not being present in another one which is used to execute `ros2`.

maziarz, 2019-01-14:
> The separate Python environments are certainly causing the problem. While you installed the dependencies in on environment they are not being present in another one which is used to execute `ros2`.

So in essense, ros2cli is using `/usr/local/bin/python3`? In that case, we have found our problem..

maziarz, 2019-01-14:
Basically i fixed it by switching back to system python instead. If anyone else using `pyenv` runs into this, you basically just need to do `pyenv global system` and reinstall the pip dependencies.