ros2/ros2 #647: Crystal Patch Release 2

nuclearsandwich, 2019-01-16:
This issue will track the list of changes that should be included in the next Crystal patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired features ready to be released.

#### PRs merged for this patch release:

<details>
<summary>...Example feature/fix to be released.</summary>
<ul>
<li><a href="#">(links to relevant pull requests)</a></li>
</ul>
</details>

#### Changes still in progress for this patch release

<details>
<summary>...Example feature/fix to be released.</summary>
<ul>
<li><a href="#">(links to relevant pull requests)</a></li>
</ul>
</details>

---

When the patch release date is set, we'll start tracking the list of every repository that needs to be released and review changes to those repositories for compatibility with Crystal to prevent breaking ABI/API unintentionally.

nuclearsandwich, 2019-01-23:
Added the plan to fix #648 in the next patch release.

mjcarroll, 2019-01-28:
On sros2, I would also like to squeeze this in: https://github.com/ros2/sros2/pull/72 but it still requires some CI changes.

dirk-thomas, 2019-01-29:
A bug fix for the problem in the Debian packages described in #653 should be consider.

chapulina, 2019-01-29:
A couple of fixes for node graph to consider:

* https://github.com/ros2/rmw_opensplice/pull/255
* https://github.com/ros2/rcl/pull/368

sloretz, 2019-02-02:
ros2/examples#226 would be a good one. It changes where the examples install their entry points so they're not accidentally included.

dirk-thomas, 2019-02-05:
The following PR needs to be in to fix action which have fields using other packages: ros2/rosidl#343

nuclearsandwich, 2019-02-05:
I've updated the issue body.

> @jacobperron: I think we can consider the new package `sros2_cmake` for the patch release ([ros2/sros2#75](https://github.com/ros2/sros2/pull/75)).

I've added this to the to-be-included list.

> @mjcarroll: On sros2, I would also like to squeeze this in: [ros2/sros2#72](https://github.com/ros2/sros2/pull/72) but it still requires some CI changes.

I haven't pulled this in above because I think the CI configuration still has open questions. It will have to wait for a future patch release.

> @dirk-thomas: A bug fix for the problem in the Debian packages described in #653 should be consider.

This has been added to the release's in progress changes with the anticipation that it will merge before the end of the week (ball is in my court to create an off-farm test package)

> @chapulina: A few more fixes to consider:
> 
>     * [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255)
> 
>     * [ros2/rcl#368](https://github.com/ros2/rcl/pull/368)

These two changes will require backport PRs to the respective `crystal` branches as mainline on those packages have broken compatibility in preparation for the Dashing.
> 
>     * [ros2/ros2cli#158](https://github.com/ros2/ros2cli/pull/158)

This change has been added to the release queue.

> @sloretz: [ros2/examples#226](https://github.com/ros2/examples/pull/226) would be a good one. It changes where the examples install their entry points so they're not accidentally included.

This change has been added to the release queue.

> @dirk-thomas: The following PR needs to be in to fix action which have fields using other packages: [ros2/rosidl#343](https://github.com/ros2/rosidl/pull/343)

This change was somehow missed in the previous patch release process along with some of the uncrustify updates and is in the queue for this one.

@jacobperron if you think there's a possibility of the Actions in Python work being ready by the end of the day Friday could you please link those PRs here. If not then we'll target them for patch release 3.

dirk-thomas, 2019-02-05:
> [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255)

@chapulina Why should this be backported? What exact problems does it fix?

ross-desmond, 2019-02-06:
> > [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255)
> 
> @chapulina Why should this be backported? What exact problems does it fix?

This fixes tests for opensplice when searching for a topic via a ros node in the dynamic node graph. Prior to this fix in rmw-opensplice, a node was unable to lookup it's own topics and services discovered through the Simple Discovery Protocol. This enables rmw-opensplice to behave the way rmw-connext and rmw-fastrtps in regards to the node graph API.

dirk-thomas, 2019-02-06:
> This fixes tests for opensplice when searching for a topic via a ros node in the dynamic node graph.

This on its own is not a reason for backporting it.

> Prior to this fix in rmw-opensplice, a node was unable to lookup it's own topics and services discovered through the Simple Discovery Protocol. This enables rmw-opensplice to behave the way rmw-connext and rmw-fastrtps in regards to the node graph API.

This certainly is a good reason to backport it. (It might be good to mention this on the PR itself in case future reader wonder.) Thanks for clarifying :+1:

jacobperron, 2019-02-06:
> @jacobperron if you think there's a possibility of the Actions in Python work being ready by the end of the day Friday could you please link those PRs here. If not then we'll target them for patch release 3.

I'd say there is a chance for Actions in Python to be ready.
Here are the relevant PRs:

* Action Client ros2/rclpy#262 (merged)
* Refactor extension module  ros2/rclpy#269 (needs review)
* Action Server ros2/rclpy#270 (almost ready for review)

There may also be another PR related to client adding some nice-to-have feature that was missed.

nuclearsandwich, 2019-02-06:
As an update, the rosout logging may either be dropped from the patch release or need further changes as it makes new assumptions about node name uniqueness. The conversation starts with https://github.com/ros2/rcl/pull/350#issuecomment-460877964 and is so far continuing there.

At this time my intention is to continue with the patch release as scheduled and drop the rcl changes from it pending the outcome of that discussion. I don't believe the backport of https://github.com/ros2/rcl/pull/368 is critical and I also believe we can backport https://github.com/ros2/rmw_opensplice/pull/255 without it but please correct me if I am mistaken.

dirk-thomas, 2019-02-06:
I am wondering if ros2/rmw_fastrtps#238 and ros2/rmw_fastrtps#254 are fixing an important enough issue (avoid a race when waiting for a service) which is worth being backported to Crystal?

ross-desmond, 2019-02-06:
>  I don't believe the backport of [ros2/rcl#368](https://github.com/ros2/rcl/pull/368) is critical and I also believe we can backport [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255) without it but please correct me if I am mistaken.

You can backport [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255) without [ros2/rcl#368](https://github.com/ros2/rcl/pull/368) though the rcl changes are simply 2 deletions.

nuclearsandwich, 2019-02-06:
> I am wondering if [ros2/rmw_fastrtps#238](https://github.com/ros2/rmw_fastrtps/pull/238) and [ros2/rmw_fastrtps#254](https://github.com/ros2/rmw_fastrtps/pull/254) are fixing an important enough issue (avoid a race when waiting for a service) which is worth being backported to Crystal?

Given that Fast-RTPS is our default rmw provider I'd be inclined to backport improvements. My biggest concern would be if those changes are compatible with Fast-RTPS 1.7.0 rather than master.

nuclearsandwich, 2019-02-06:
> You COULD backport [ros2/rmw_opensplice#255](https://github.com/ros2/rmw_opensplice/pull/255) without [ros2/rcl#368](https://github.com/ros2/rcl/pull/368), but this would result in code in rmw_opensplice that would not be tested.

Thanks for the feedback here @ross-desmond now that we've got a path to resolving the rosout node name kerfuffle (see https://github.com/ros2/rcl/pull/350#issuecomment-461227553) my updated intention (see https://github.com/ros2/ros2/issues/647#issuecomment-461156697 for previous) is to release rcl once the amended rosout behavior has been merged and backported, so we can backport this rcl change along with it. Although it is potentially moot I am not too concerned about code not being tested directly in crystal since we know it's getting tested on master. But that concern will increase as master and crystal diverge so I'd prefer to backport the test change as well and would only have avoided it to avoid releasing on an unstable branch.

nuclearsandwich, 2019-02-07:
While reviewing [devel jobs](http://build.ros2.org/view/Cdev/) yesterday I saw that ros2cli relies on unreleased changes in rclpy. Which reminds me to run my script checking for other unreleased changes.

nuclearsandwich, 2019-02-08:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

This is a test of the current state of Crystal's development branches with the backport PRs I just opened. At least it will be if I set it up correctly. There are some repositories that have changes on their development branch which won't be released in this patch release but this is the easiest way to test things before actually starting to make releases.

nuclearsandwich, 2019-02-08:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

The failure in the previous build is related to https://github.com/ament/ament_cmake/commit/06f842e39b1786aeb4fdb8366eac157e9badf495 which isn't scheduled for release in this patch releas. I've started a new round of builds pinning that repo to it's Crystal release.

nuclearsandwich, 2019-02-14:
* Linux Bionic [image]
* Linux-aarch64 Bionic [image]
* Linux Xenial [image]
* Linux-aarch64 Xenial [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-02-14:
Flake8 tests are failing in these builds because the Crystal track doesn't include https://github.com/ament/ament_lint/pull/124 which is required to use flake8 >= 3.7.0. When installing from debs Xenial uses 2.5.4 and Bionic uses 3.5.0. I don't feel like this is a release blocker but we ought to consider the ament_lint PR for backport in a future patch release to make it easier for our CI to test Crystal.

nuclearsandwich, 2019-02-15:
Packaging jobs:

* Linux AMD64 Bionic [image]
* Linux ARM64 Bionic [image]
* Linux AMD64 Xenial [image]
* Linux ARM64 Xenial [image]
* macOS [image]
* Windows [image]
* Windows Debug [image]

nuclearsandwich, 2019-02-15:
:ship:'d and :mega:'d https://discourse.ros.org/t/patch-release-and-new-packages-for-ros-2-crystal-clemmys-2019-02-14/7867

PR #655 (2019-02-15): Update Crystal repos file for patch release 2.
This is the proposed release state for Crystal patch 2.

This PR will be updated as changes to the release are made and is to be merged when Crystal patch 2 is released.

Closes #647
closed 2019-02-15 by nuclearsandwich