ros2/ros2 #648: uncrustify failed with error code 1 while colcon test for msg packages

yechun1, 2019-01-22:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04 
- Installation type:
  -  binaries installation, Issue find after Update Crystal repos file for Patch release 1. #646 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
apt-get install ros-crystal-dekstop (and apt-get upgrade to latest version)
source /opt/ros/crystal/setup.bash
<clone crystal source to src_crystal>
colcon build --base-paths src_crystal/ros2/rcl_interfaces/
colcon test --base-paths src_crystal/ros2/rcl_interfaces/

```

#### Expected behavior
build without failures
#### Actual behavior
build with test failures
#### Additional information

----
```
ros2_ws$ colcon build --base-paths src_crystal/ros2/rcl_interfaces/
Starting >>> builtin_interfaces
Starting >>> lifecycle_msgs
Finished <<< builtin_interfaces [10.6s]                                                                        
Starting >>> action_msgs
Starting >>> rcl_interfaces
Starting >>> rosgraph_msgs
Finished <<< rosgraph_msgs [13.7s]                                                                                                
Finished <<< action_msgs [27.5s]                                                                            
Starting >>> test_msgs
Finished <<< lifecycle_msgs [49.4s]                                                                                              
[Processing: rcl_interfaces, test_msgs]                                                                     
Finished <<< rcl_interfaces [1min 15s]                                                                          
Finished <<< test_msgs [1min 5s]                                

Summary: 6 packages finished [1min 43s]
ros2_ws$ colcon test --base-paths src_crystal/ros2/rcl_interfaces/
Starting >>> builtin_interfaces
Starting >>> lifecycle_msgs
--- stderr: builtin_interfaces                                                       
Errors while running CTest
---
Finished <<< builtin_interfaces [9.57s]	[ with test failures ]
Starting >>> action_msgs
Starting >>> rcl_interfaces
Starting >>> rosgraph_msgs
Finished <<< rcl_interfaces [0.85s]                                                                                            
--- stderr: lifecycle_msgs                                                                             
Errors while running CTest
---
Finished <<< lifecycle_msgs [16.4s]	[ with test failures ]
--- stderr: rosgraph_msgs                                                     
Errors while running CTest
---
Finished <<< rosgraph_msgs [9.55s]	[ with test failures ]
--- stderr: action_msgs                     
[... 30 lines cut ...]
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__traits.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array__struct.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array__traits.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request__struct.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request__traits.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response__struct.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response__traits.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__struct.hpp as language CPP
3: Parsing: /data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__traits.hpp as language CPP
3:-
3: The invocation of 'uncrustify' failed with error code 1: Command '['/opt/ros/crystal/bin/uncrustify', '-c', '/tmp/uncrustify_34uitw5v', '--prefix', '/tmp/uncrustify_gvx16fog', '--suffix', '.uncrustify', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_info__traits.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status__traits.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/msg/goal_status_array__traits.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__request__traits.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__response__traits.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__struct.hpp', '/data/proj/ros2_ws/build/action_msgs/rosidl_generator_cpp/action_msgs/srv/cancel_goal__traits.hpp']' returned non-zero exit status 1.
3: -- run_test.py: return code 1
3: -- run_test.py: generate result file '/data/proj/ros2_ws/build/action_msgs/test_results/action_msgs/uncrustify_rosidl_generated_cpp.xunit.xml' with failed test
3: -- run_test.py: verify result file '/data/proj/ros2_ws/build/action_msgs/test_results/action_msgs/uncrustify_rosidl_generated_cpp.xunit.xml'
 3/31 Test  #3: uncrustify_rosidl_generated_cpp ...................***Failed    0.31 sec
```
## Feature request

#### Feature description

#### Implementation considerations

yechun1, 2019-01-22:
seems all the msgs packages with "ADD_LINTER_TESTS" will test with failures.

dirk-thomas, 2019-01-22:
What version of the `ros-crystal-uncrustify-vendor` do you have installed?

yechun1, 2019-01-22:
$ dpkg -l |grep uncrustify-vendor
ii  ros-crystal-uncrustify-vendor                      1.1.0-0bionic.20190114.183025                amd64        Wrapper around uncrustify, providing nothing but a dependency on uncrustify, on some systems.

dirk-thomas, 2019-01-23:
I can reproduce the test failures (using the same version of the Debian package). The problem seems to be that the new `uncrustify` version was released into Crystal but not the changes in the message generator packages related to it (e.g. ros2/rosidl#341).

@nuclearsandwich I think this needs a patch release for all the repos referenced from ros2/rosidl#340 which haven't been released since the `MAX_LINE_LENGTH` argument was changes, e.g. `rosidl`, `rcutils`, `rclcpp`.

yechun1, 2019-02-22:
Issue fixed by Cyrstal Patch Release 2, close it.

closed 2019-02-22 by yechun1