ros2/ros2 #653: install.sh in overlay workspace does not source binaries

Karsten1987, 2019-01-29:
## Bug report
When working with the binary installation and a custom overlay workspace on top of it, sourcing the `setup.bash` of the overlay workspace does not recursively source the binary installation. 

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries with custom overlay workspace
- Version or commit hash:
  - latest binaries (crystal)

#### Steps to reproduce issue
Steps are inspired from this [Dockerfile](https://github.com/ros-controls/ros2_controllers/blob/crystal/travis/Dockerfile)

```
sudo apt-get install ros-crystal-desktop  #install binaries
mkdir -p ~/ros2_overlay/src
cd $_
git clone <repo name>
cd -
source /opt/ros/crystal/setup.bash
colcon build
```
in a fresh terminal now:
```
cd ~/ros2_overlay
source install/setup.bash
ros2
```

#### Expected behavior
ros2 entry point found
```
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

optional arguments:
  -h, --help            show this help message and exit

Commands:
  daemon  Various daemon related sub-commands

  Call `ros2 <command> -h` for more detailed usage.
```

#### Actual behavior
ros2 is not found, meaning the binaries workspace was not recursively sourced when sourcing the workspace install `setup.bash`
```
bash: ros2: command not found
```

#### Additional information

When sourcing the binary workspace `/opt/ros/crystal/setup.bash` beforehand (manually) followed by sourcing the workspace it works fine. 

Also when compiling everything from source, the chaining of workspaces works as expected. I believe something during the packaging process might prevent this functionality.

dirk-thomas, 2019-01-29:
The prefix level setup files generated by `colcon` in the overlay workspace only consider workspaces listed in the `COLCON_PREFIX_PATH` environment variable. Since the Debian packages under `/opt/ros/...` are not being built by `colcon` they don't set that environment variable - only `AMENT_PREFIX_PATH`.

That is also the reason why it works when the underlay workspace was built locally with `colcon` and only fails with the Debian packages.

I am not sure what the best long term solution for this is atm but as a short term fix I would suggest to add an environment hook to the [ros_workspace](https://github.com/ros2/ros_workspace/blob/e52a32fe8ffe4ba3c9a9a3b92a8633b17a618024/CMakeLists.txt#L17) package whicih sets `COLCON_PREFIX_PATH` to the same value as `AMENT_PREFIX_PATH`.

codebot, 2019-02-04:
Thanks for looking into this. I'm in the same situation as @Karsten1987  :+1:

dirk-thomas, 2019-02-07:
Addressed by ros2/ros_workspace#11.

closed 2019-02-07 by dirk-thomas