ros2/ros2 #656: Crystal Patch Release 3

nuclearsandwich, 2019-02-15:
This issue will track the list of changes that should be included in the next Crystal patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired features ready to be released.

#### PRs merged for this patch release:

#### PRs which need a backport for this patch release:

#### Changes in progress for this patch release

<details>
<summary>Enable rcl_logging_log4cxx build</summary>
<ul>
<li><a href="https://github.com/ros2/choco-packages/pull/7">ros2/choco-packages#7</a></li>
<li><a href="https://github.com/ros2/rcl_logging/pull/3">ros2/rcl_logging#3</a></li>
</ul>
</details>

---

When the patch release date is set, we'll start tracking the list of every repository that needs to be released and review changes to those repositories for compatibility with Crystal to prevent breaking ABI/API unintentionally.

jacobperron, 2019-02-15:
PRs for Python Actions in merge order:

* Action Client ros2/rclpy#262
* Bugfix ros2/rclpy#272
* Bugfix ros2/rclpy#275
* Add `rclpy_common` extension ros2/rclpy#269
* Action Server ros2/rclpy#270

dirk-thomas, 2019-02-20:
PR fixing the include dir order of overlay workspaces: ament/ament_cmake#157

Karsten1987, 2019-02-20:
PR exporting the cmake logic to find ros1 packages through the ros1_bridge:
https://github.com/ros2/ros1_bridge/pull/164

Karsten1987, 2019-02-25:
display the robot description correctly in rviz: https://github.com/ros2/rviz/pull/378

dirk-thomas, 2019-02-28:
PR fixing naming collisions with OpenSplice: ros2/rosidl_typesupport_opensplice#20

Karsten1987, 2019-03-04:
PR for `ament_lint` as discussed on the last patch release ticket:
https://github.com/ament/ament_lint/pull/124

see discussion here: https://github.com/ros2/ros2/issues/647#issuecomment-463814096

nuclearsandwich, 2019-03-07:
> PR fixing naming collisions with OpenSplice: [ros2/rosidl_typesupport_opensplice#20](https://github.com/ros2/rosidl_typesupport_opensplice/pull/20)

This pull request has since been [reverted](https://github.com/ros2/rosidl_typesupport_opensplice/pull/22) and the [re-application](https://github.com/ros2/rosidl_typesupport_opensplice/pull/23) is still in progress.

sloretz, 2019-03-07:
Backport PRs that fix ros2/rosidl#348

* https://github.com/ros2/rosidl_typesupport/pull/45
* https://github.com/ros2/rcl_interfaces/pull/66
* https://github.com/ros2/rosidl_python/pull/31
* https://github.com/ros2/system_tests/pull/335

jacobperron, 2019-03-08:
Additional backports needed for Python Action support (added to the list):
* ros2/rosidl_python#21
* ros2/rosidl_python#28

nuclearsandwich, 2019-03-08:
Updated the issue body with the open backport PRs. Rather than try and track the backports as both "in progress" work and backports of needed features from development branches I organized them all by what they backport. That leaves the In progress list for tasks which may still have development required before even a backport is ready.

I've also put up a link to a gist I'll try to keep in sync with the current status of backport branches so we can use it to test incoming PRs against Crystal patch 3.

nuclearsandwich, 2019-03-08:
@sloretz Here's CI for the backport branches fixing https://github.com/ros2/rosidl/issues/348 testing test_communication.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-03-08:
The macOS failure looks like a bad interaction with Fast-RTPS and TinyXML2.

nuclearsandwich, 2019-03-08:
@jacobperron here's CI for the rclpy Actions branches testing test_communication and rclpy.

* Linux [image]
* Linux-aarch64 [image]
* macOS (crystal infra) [image]
* Windows [image]

nuclearsandwich, 2019-03-08:
FYI I just somehow managed to mangle a bunch of the recent edits in the issue body and GitHub doesn't appear to give me a way to undo that damage. Please refrain from making edits while I fix the mess.

nuclearsandwich, 2019-03-08:
My browser tab is cursed. Somehow destroyed all the edits again :sob: :sob: :sob:

jacobperron, 2019-03-09:
Backporting ament/ament_lint#122 should resolve the lint errors. We can either put it on top of ament/ament_lint#130 or wait for that to be merged and open a follow up PR.

jacobperron, 2019-03-09:
CI for rclpy Actions, this time with backport ament/ament_lint#131:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-03-09:
Rebuild of https://github.com/ros2/ros2/issues/656#issuecomment-471131012 with the updated branch now that the rcl_logging backport has merged.

* Linux [image]
* Linux-aarch64 [image]
* macOS (Crystal infra) [image]
* Windows [image]

jacobperron, 2019-03-13:
Backport for Python actions communication tests: ros2/system_tests#337

mjcarroll, 2019-03-13:
The backported changes in `sros2` introduce a dependency on the rosdep key: `python3-lxml`

The installation documentation has been updated in ros2/ros2_documentation#138, but I'll highlight the changes here: 

Depending on the platform, users may need to get the additional dependency:
* With Ubuntu debs: apt will install suitable the suitable dependency
* With Ubuntu binary releases: The `rosdep` tool should resolve the dependency (installs `python3-lxml`)
* With macOS: The dependency should be resolved either with `rosdep` or by installing `lxml` via pip
* With Windows: The dependency should be resolved by installing `lxml` via pip
* With Windows (debug): The dependency should be resolved by installing a custom-built `lxml` via pip: https://github.com/ros2/ros2/releases/tag/lxml-archives

mikaelarguedas, 2019-03-13:
> With macOS: The dependency should be resolved either with rosdep or by installing lxml via pip

There are currently no rosdep macos rules for `python3-lxml` so it will not be resolved.

Based on https://github.com/ros/rosdistro/pull/20560#issuecomment-472628304 it looks like it will stay the case for now until we have a way to specify / test which pip is used to install a package via rosdep.
The instructions linked above do mention to install it via pip so users following the instructions should not face any issue :+1:

nuclearsandwich, 2019-03-14:
Thanks for the edit @mikaelarguedas I've amended https://github.com/ros2/ros2/issues/656#issuecomment-472614723 which will go into the release notes imminently.

mjcarroll, 2019-03-14:
@mikaelarguedas You are correct, I mis-remembered the rosdep keys, I'll also make sure that the documentation is in sync with that.

nuclearsandwich, 2019-03-14:
* Linux Bionic [image]
* Linux Bionic ARM64 [image]
* Linux Xenial [image]
* Linux Xenial ARM64 [image]
* macOS [image]
* Windows[image]
* Windows Debug [image]

dejanpan, 2019-03-14:
@nuclearsandwich I am sorry for a late chime in but I think that these 2 bugs area serious:
1. Race condition in rmw_fastrtps: https://github.com/ros2/rmw_fastrtps/issues/258. We confirmed that the fix works: https://github.com/ros2/rmw_fastrtps/issues/258#issuecomment-473073537.
2. https://github.com/eProsima/Fast-RTPS/pull/444 aka FastRTPS 1.7.2 release. What is concretely serious here is a memory leak reported here https://github.com/ros2/rmw_fastrtps/issues/257 and fixed here https://github.com/eProsima/Fast-RTPS/pull/434. Should including 1.7.2 not be acceptable by you since according to @wjwwood there are API breaking changes since 1.7.0, then I think at least a known issue must be clearly documented in [ROS 2 Crystal Patch 3 Release](https://github.com/ros2/rmw_fastrtps/issues/257#issuecomment-470406047): 
    1. TRANSIENT_LOCAL QoS should not be used in current ROS 2 Crystal Patch 3 when `RMW_IMPLEMENTATION=rmw_fastrtps_cpp`
    1. https://github.com/ros2/rmw_fastrtps/issues/257#issuecomment-464671472

nuclearsandwich, 2019-03-14:
@dejanpan the window is definitely closed for new changes in patch release 3. Since the Fast-RTPS issues have been present in Crystal since its release rather than recent regressions, I am opting not to delay this patch release and instead we can target these fixes for the next one (#668).

Open Robotics had a brief discussion about whether or not the improvements in Fast-RTPS 1.7.1/1.7.2 are worth allowing breaking changes and I expect that discussion to continue in #668.

In the meantime, documenting these issues is a important, although I think adding them to [the documentation](https://github.com/ros2/ros2_documentation/blob/master/source/Releases/Release-Crystal-Clemmys.rst#id4) for Crystal makes more sense than putting them in the release notes since they're not new, just newly reported. I will definitely add a remark re-linking the Crystal release page and known issues in the release announcement.

nuclearsandwich, 2019-03-15:
The patch release has [shipped](https://discourse.ros.org/t/patch-release-and-new-packages-for-ros-2-crystal-clemmys-2019-03-14/8311)! :boat:

PR #667 (2019-03-14): Crystal Patch Release 3
This is the proposed (and likely final) release state for Crystal patch 3.

Closes #656
closed 2019-03-15 by nuclearsandwich