ros2/ros2 #657: Python path seems to be hard-coded to C:\Python37 in Windows binary installation

rlkamalapurkar, 2019-02-17:
## Bug report

**Required Info:**

- Operating System:
Windows 10
- Installation type:
Binaries
- Version or commit hash:
ROS 2 Crystal Clemmys - Patch Release 2
- DDS implementation:
 OpenSplice
- Client library (if applicable):
N/A

#### Steps to reproduce issue
Install everything but Python according to the instructions [here] https://index.ros.org/doc/ros2/Installation/Windows-Install-Binary/). Install Python to any directory **other than C:\Python37**
Run
ros2 run demo_nodes_cpp talker

#### Expected behavior
Talker starts publishing

#### Actual behavior
A prompt that says "failed to create process."

#### Additional information
Keep everything else the same, copy and paste the Python installation from wherever you installed it to C:\Python37 and modify the PATH variable to reflect the new location of Python.

Run
ros2 run demo_nodes_cpp talker
Result: 
Talker now works as expected.

ROS2 should work as long as Python paths are set up correctly, it should not matter where the installation is.

Hurix, 2021-02-23:
Is there a workaround to this that doesnt involve me having to manage multiple python installations?
Can I change the shebang of (which?) files to make the ros2 cmd work for me and my weird python path, or is that baked into binaries?

2catycm, 2022-03-04:
Hi, I am a ROS beginner. I am trying to install ROS on Windows too, and don't like the location requirement by ROS installation guide. I searched Stackoverflow and found that it is useful to use `mklink` to predend that C:\python37 to be real path, while actually it is installed somewhere else.

mjcarroll, 2025-03-29:
For Windows installations moving forward, we suggest using `pixi` as the dependency management system.  As part of this effort, there has been a lot of work to make the Windows installation relocatable as well: https://discourse.ros.org/t/upcoming-switch-of-windows-installation-to-pixi-conda/41916

closed 2025-03-29 by mjcarroll