ros2/ros2 #658: use ament_target_dependencies where possible to order include dirs

dirk-thomas, 2019-02-20:
Follow up of ament/ament_cmake#130.

ivanpauno, 2019-03-13:
Am I in the right direction here https://github.com/ros2/rosidl_typesupport/pull/46?
And using target_include_directories shouldn't be a problem in tests, right? Like this one: https://github.com/ros2/geometry2/blob/aaae4cf046e44787be12e4f5209cef34a8123a39/tf2_eigen/CMakeLists.txt#L24-L35

If it is ok, I can take this issue

ivanpauno, 2019-03-14:
> And using target_include_directories shouldn't be a problem in tests, right? Like this one: https://github.com/ros2/geometry2/blob/aaae4cf046e44787be12e4f5209cef34a8123a39/tf2_eigen/CMakeLists.txt#L24-L35

Checking the comments again, it should be modified to use "ament_include_directories_order" manually in this case, I will start working in that direction if it is ok

ivanpauno, 2019-03-18:
@dirk-thomas This three PRs need a review:
https://github.com/ros/class_loader/pull/121
https://github.com/ros2/rosidl_typesupport/pull/46
https://github.com/ros2/rcl/pull/400

I will run a CI job with this complete update after approval of the PRs.

wjwwood, 2019-03-18:
@ivanpauno do you want to do CI for https://github.com/ros2/rviz/pull/384 as well when you do the others? There's some test failures in the CI I ran, but it might be known issues.

dirk-thomas, 2019-03-18:
> This three PRs need a review:

I have already reviewed the second and third. @nuclearsandwich will review the first since he is the maintainer of that package/repo.

Please also set the label to match the status a ticket is supposed to have ("in review" rather than "in progress").

jacobperron, 2019-04-25:
@ivanpauno It looks like all referenced PRs have been merged. Is there still more to do for this ticket or can it be closed?

ivanpauno, 2019-04-25:
> @ivanpauno It looks like all referenced PRs have been merged. Is there still more to do for this ticket or can it be closed?

I will check again before closing it.

clalancette, 2019-04-29:
I'm actually going to close this out just to get it out of our review column.  Feel free to reopen if you find more issues that need to be addressed as part of this.

dirk-thomas, 2019-04-29:
@clalancette You do we ensure that this will be checked with the ticket closed?

clalancette, 2019-04-29:
> @clalancette You do we ensure that this will be checked with the ticket closed?

Nothing, but there is nothing to ensure it will be checked with it open either.  Basically there is nothing actionable here, and having it just hanging out in the review column with everything merged means it is just wasting reviewer time looking at it for no reason.  I could alternatively open it back up and switch to "in progress", but it definitely shouldn't be "in review".

dirk-thomas, 2019-04-29:
Currently there is no ticket to follow up on "I will check again before closing it.". Therefore I will reopen it.

ivanpauno, 2019-05-06:
I opened the two leftovers I've found. I will close this issue after merging them.

closed 2019-05-07 by ivanpauno