ros2/ros2 #660: Errors building qt_gui_cpp

JacobJBeck, 2019-02-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - qt_gui_core 968affd39a5db52c8b688fc718e2f83e07341308
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Roughly following instructions [here](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/).  Key steps are:
```
$ pip3 install pyqt5
$ colcon build --symlink-install --packages-select qt_gui_cpp
```

#### Expected behavior
Package builds.

#### Actual behavior
Build fails with
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                             
Traceback (most recent call last):
  File "/home/jacob/ros2_core_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 50, in get_sip_dir_flags
    sip_dir = config.pyqt_sip_dir
  File "/usr/lib/python3/dist-packages/sipconfig_nd6.py", line 191, in __getattr__
    raise AttributeError("\"%s\" is not a valid configuration value or user option" % name)
AttributeError: "pyqt_sip_dir" is not a valid configuration value or user option

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jacob/ros2_core_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 85, in <module>
    sip_dir, sip_flags = get_sip_dir_flags(config)
  File "/home/jacob/ros2_core_ws/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 66, in get_sip_dir_flags
    ' that PyQt5 is installed')
FileNotFoundError: The sip directory for PyQt5 could not be located. Please ensure that PyQt5 is installed
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]

Summary: 0 packages finished [1.45s]
  1 package failed: qt_gui_cpp
  1 package had stderr output: qt_gui_cpp
```

#### Additional information
Possibly related to #623.

Configuration info
```
$ which -a python3
/usr/bin/python3
$ which -a sip
/usr/bin/sip
$ sip -V
4.19.13
$ python3 -c "import sipconfig; print(sipconfig._pkg_config['default_sip_dir'])"
/usr/share/sip
```
Note that `/usr/share/sip` does not exist!

dirk-thomas, 2019-02-22:
You shouldn't install dependencies via `pip` on Ubuntu. Please use `rosdep install ...` as described in this [tutorial](https://index.ros.org/doc/ros2/Tutorials/RQt-Source-Install/#install-dependencies) to install the proper Debian packages.

JacobJBeck, 2019-02-22:
Using `rosdep install --from-paths src --ignore-src --rosdistro crystal -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 rti-connext-dds-5.3.1 urdfdom_headers"` does not resolve the issue.

dirk-thomas, 2019-02-22:
Did you remove all (recursive) packages previous installed via `pip`?

JacobJBeck, 2019-02-22:
I used `pip3 uninstall pyqt5`, are there additional steps?

dirk-thomas, 2019-02-23:
It could have installed recursive packages during `install` which won't be removed during `uninstall`. If the problem persists I would suggest trying the instructions in a clean environment - e.g. a Docker container - to check if it works there.

JacobJBeck, 2019-02-25:
Ok, I think I was able to resolve the issue by using `sudo pip3 uninstall pyqt5` to ensure removal from system space. Thanks for your help.

closed 2019-02-25 by JacobJBeck

dbHub20092758, 2021-12-07:
python version is error ? default python version of foxy is 3.8, but python version of ubuntu18.04 is 3.6