ros2/ros2 #663: [ROS2 QA] Enable symbol visibility for all RMW and RCL packages

thomas-moulard, 2019-02-27:
Enable symbol visibility for all RMW and RCL packages

See doc here: https://gcc.gnu.org/wiki/Visibility

## Exit Criteria

* Symbol visibility enabled for all RMW and RCL packages (ensuring the set of macros - see https://gcc.gnu.org/wiki/Visibility is defined, `CMakeLists.txt` is updated accordingly)
* All RMW and RCL are compiling with `-Wvisibility=hidden` enabled by default when using GCC and Clang on Linux and Mac OS X.

## Links

* Discourse: https://discourse.ros.org/t/set-symbol-visibility-to-hidden-for-rmw-and-rcl-packages/7981/

emersonknapp, 2019-09-06:
@thomas-moulard is this done?

thomas-moulard, 2019-09-06:
Partially only - we can close for now as we don't have the bandwidth to do that currently.

closed 2019-09-06 by thomas-moulard