ros2/ros2 #664: [ROS2 QA] Implement PoC for Thread-Safety annotation

thomas-moulard, 2019-02-27:
Implement PoC for Thread-Safety annotation

Clang doc here: https://clang.llvm.org/docs/ThreadSafetyAnalysis.html

Candidate class for the PoC is https://github.com/ros2/rmw_fastrtps/blob/705a0c07ca9f3708028660c41de6eadfad16e9f5/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/topic_cache.hpp

## Exit Criteria

* Add Thread-Safety annotations macro to e.g. rcutils (see `mutex.h` in https://clang.llvm.org/docs/ThreadSafetyAnalysis.html) - which repo should be cleared with OSRF first
* Modify `rmw_fastrtps*` packages to build with -Wthread-safety by default when Clang is chosen.
* Modify topic_cache.hpp to use thread-safety annotations.

emersonknapp, 2019-02-27:
Adding rcpputils to repos list - https://github.com/ros2/ros2/pull/661

emersonknapp, 2019-03-11:
The rmw (ROS2 Middleware) implementations have been identified as prime candidates for TSA, due to the inherently threaded nature of pub/sub callbacks. Additionally, the C++ client library rclcpp has a good deal of threaded code, and is likely to be a widely used language client in ROS2 projects.

- [ ] rmw_fastrtps
    - [X] rmw_fastrtps_shared_cpp https://github.com/ros2/rmw_fastrtps/pull/259
    - [ ] rmw_fastrtps_dynamic_cpp
    - [ ] rmw_fastrtps_cpp
- [ ] rmw_opensplice
    - [ ] rmw_opensplice_cpp
- [ ] rmw_connext
    - [ ] rmw_connext_shared_cpp
    - [ ] rmw_connext_dynamic_cpp
    - [ ] rmw_connext_cpp
- [ ] rclcpp
    - [ ] rclcpp
    - [ ] rclcpp_action

nuclearsandwich, 2019-03-22:
I've deployed the updated [nightly_linux_clang_libcxx](https://ci.ros2.org/job/nightly_linux_clang_libcxx/) job to https://ci.ros2.org. I've also disabled the earlier nightly_linux_clang job and will likely delete it in the coming week.

thomas-moulard, 2019-04-09:
@emersonknapp is there any work left there?

emersonknapp, 2019-04-09:
Nothing left to do, the described task "implement poc" is complete. There is still ongoing work to make the nightly build happy, and followup to annotate more packages, but I feel these should be tracked separately

closed 2019-04-17 by thomas-moulard