ros2/ros2 #666: [ROS 2 QA] Add TSan sanitizer nightly job to ROS2 CI

thomas-moulard, 2019-02-27:
Adding a GCC Linux nightly job to compile ROS 2 with TSAN and run core packages tests with TSAN enabled.

Owner: @bhatsach

## Exit Criteria

* TSAN Job is added to `ci.ros2.org`

thomas-moulard, 2019-03-18:
Closing as https://github.com/ros2/ci/pull/254 has been merged.

dirk-thomas, 2019-03-18:
I would suggest to keep this ticket open until the new job actually passed.

thomas-moulard, 2019-03-18:
You mean until the job passes w/o Jenkins running out of memory, right?

nuclearsandwich, 2019-03-18:
Opened https://github.com/ros2/ci/issues/265 tracking my investigation.

nuclearsandwich, 2019-03-22:
With [last night's build](https://ci.ros2.org/view/nightly/job/nightly_linux_thread_sanitizer/10) the thread sanitizer job now completes. There are a large number of tests with thread sanitizer output. Once that is under control it will be possible to explore widening the scope of tested packages.

closed 2019-03-22 by nuclearsandwich