ros2/ros2 #668: Crystal Patch Release 4

nuclearsandwich, 2019-03-14:
This issue will track the list of changes that should be included in the next Crystal patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

---

#### Changes proposed for backport this patch release

#### Changes in progress for this release

---

When the patch release date is set, we'll start tracking the list of every repository with changes that need to be backported and released. Proposed changes should be reviewed for compatibility with Crystal to prevent breaking ABI/API unintentionally.

nuclearsandwich, 2019-03-14:
As [mentioned previously](https://github.com/ros2/ros2/issues/656#issuecomment-473076708) and discussed internally, (calling @clalancette as the current leader of that discussion), we should consider updating the version of Fast-RTPS in Crystal to resolve ros2/rmw_fastrtps#257 and ros2/rmw_fastrtps#258.

clalancette, 2019-03-15:
I'm now tracking the Fast-RTPS issue at https://github.com/ros2/ros2/issues/669; I'll put further notes in there while we figure out what to do for Crystal patch release 4.

wjwwood, 2019-03-27:
Can we add a partial cherry-pick on the launch repository?

https://github.com/ros2/launch/commit/632ef1f3eca7dbc54fbed85ab65a5efc512d83ed#diff-970fc3c5d394ad33c14e4f9fe55288ceR113

Via

- https://github.com/ros2/launch/pull/220
- https://github.com/ros2/launch/pull/167#issuecomment-476960691

wjwwood, 2019-03-27:
@stonier FYI

nuclearsandwich, 2019-04-04:
A possible regression in rclcpp_action, which wasn't released this round and thus why it wasn't caught during the backport process, has been found https://github.com/ros2/rclcpp/issues/683.

nuclearsandwich, 2019-04-05:
Full CI with a fix for ros2/rclcpp#683. If this comes back with no unexpected failures we'll re-release rclcpp and include it in this patch release.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-04-06:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-04-08:
Packaging jobs

* Linux Bionic [image]
* Linux Xenial [image]
* Linux Bionic (ARM64) [image]
* Linux Xenial (ARM64) [image]
* macOS [image]
* Windows [image]
* Windows Debug [image]

nuclearsandwich, 2019-04-10:
I keep forgetting that PRs to the `crystal` branch won't auto-close issues.

PR #683 (2019-04-08): Crystal patch 4
Update the crystal branch to the patch 4 releases.

Connects to #668. Closes #668.
commit e0e4b6b4 (2019-04-08): Crystal patch 4 (#683)

Update the crystal branch to the patch 4 releases.

Connects to #668. Closes #668.
closed 2019-04-10 by nuclearsandwich