ros2/ros2 #669: Crystal Fast-RTPS fix backport tracking issue

clalancette, 2019-03-15:
As detailed in https://github.com/ros2/ros2/issues/656#issuecomment-473076708 and https://github.com/eProsima/Fast-RTPS/issues/359 , there are some serious RTPS discovery issues with the version of Fast-RTPS released in Crystal (1.7.0).  This bug is a tracking bug for investigations into backporting those fixes into Crystal for all users.

# Problem Statement
When using Fast-RTPS with ROS 2, it can sometimes take quite a while for nodes to discover each other, particularly when there are many topics or the machine is under stress.  In testing, I've sometimes seen it take upwards of 30 seconds to do discovery.  There also seem to be some memory leaks that may need addressing in certain situations.

# Investigation
Several bugs in Fast-RTPS have been discovered that contribute to this problem:
* Discovery takes too long, https://github.com/eProsima/Fast-RTPS/issues/359 - Fixed by https://github.com/eProsima/Fast-RTPS/pull/411
* Race condition, https://github.com/ros2/rmw_fastrtps/issues/258 - Fixed by https://github.com/richiware/rmw_fastrtps/commit/b262caa9740cb714f54c4cb80379b82a7e261fc7 , no PR yet
* Memory leak, https://github.com/ros2/rmw_fastrtps/issues/257 - Fixed by https://github.com/eProsima/Fast-RTPS/pull/434

# Potential Solutions
1. Take Fast-RTPS 1.7.2 into Crystal Patch release 4
   * Release Fast-RTPS 1.7.2 into Crystal Patch release 4
   * Get the fix for https://github.com/ros2/rmw_fastrtps/issues/258 into master, then backport to crystal
   * Pros
     * Uses the tested release from eProsima
     * Least amount of backporting work
   * Cons
      * Causes an ABI break

alsora, 2019-03-18:
I would add an additional point to the discussion:

- Memory regression: FastRTPS v1.7.0 has a much higher memory usage than v.1.6.0, v.1.7.1, v.1.7.2 

This is also evident from the [FastRTPS HelloWorld example](https://github.com/eProsima/Fast-RTPS/tree/master/examples/C%2B%2B/HelloWorldExample).

| System    | FastRTPS version |  RAM 
| -------------  | ------------- | ------------- | 
| Process with 1 participant   | 1.6.0 / 1.7.1 / 1.7.2    | 10MB        
| Process with 20 participants   |  1.6.0 / 1.7.1 / 1.7.2    | 100MB        
| Process with 1 participant  | 1.7.0   | 15MB        
| Process with 20 participants   |  1.7.0     | 190MB   

I haven't created any issue for that since when I discovered it was already fixed in the newer release.
I don't know if they are aware of it (@richiprosima ?).

clalancette, 2019-03-19:
@alsora I can reproduce the memory problem; here is my table:

| Example | FastRTPS version |  VIRT | RES | SHR |
| ----------- | ----------------------- | -------- | ------ | ------ | 
| ./HelloWorldExample publisher 1 | 1.7.2 | 276188 | 8400 | 6568 |
| ./HelloWorldExample subscriber | 1.7.2 | 398736 | 8652 | 6840 |
| ./HelloWorldExample publisher 1 | 1.7.1 | 210656 | 8484 | 6660 |
| ./HelloWorldExample subscriber | 1.7.1 |  398740 | 8904 | 7100 |
| ./HelloWorldExample publisher 1 | 1.7.0 | 213364 | 11316 | 6856 |
| ./HelloWorldExample subscriber | 1.7.0 | 401540 | 11416 | 6932 |

I tried to do a bisect to find it, but I couldn't figure it out that way.  So maybe @richiprosima has an idea, otherwise we know that it is much better in 1.7.2 than in 1.7.0.

clalancette, 2019-03-20:
We had a discussion about this yesterday, and the conclusion was that there are enough problems in Fast-RTPS 1.7.0 that we are willing to break ABI to get the fixes.  Assuming there are no major problems in testing, the plan is to get Fast-RTPS 1.7.2 into the next Crystal patch release.

richiware, 2019-03-20:
FYI In v1.7.1 I've improved the times on discovery phases, but I took also the change of improving the usage of memory in discovery phases.

What I cannot answer you right now is why v1.6.0 uses less memory than v1.7.0.

alsora, 2019-03-20:
@richiware the reason may be due to the creation of secured endpoints.

In all the aforementioned versions, when I run something, being it ROS2 related or just FastRTPS, both standard as well as secured endpoints are created.

The increase of memory usage in 1.7.0 to 1.6.0 it looks like it is mostly due to the fact that you added the `createSecureEndPoints` also in the WLP builtin interface https://github.com/eProsima/Fast-RTPS/blob/release/1.7.0/src/cpp/rtps/builtin/liveliness/WLP.cpp#L93 .

clalancette, 2019-03-20:
Here's some CI using the crystal version of everything, with https://github.com/ros2/rmw_fastrtps/pull/264 and https://github.com/ros2/rmw_fastrtps/pull/256 backported, and using Fast-RTPS 1.7.2:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

clalancette, 2019-03-22:
Almost all of the CI warnings were because of flake8; I'm going to ignore those.  There were also a few cppcheck failures that I'm also going to ignore.  That leaves us with a few actual failures:

* rclcpp_action client test seems to have failed by timing out on Linux (https://ci.ros2.org/job/ci_linux/6480/consoleFull#console-section-675); @jacobperron any thoughts?
* test_params_yaml failed by timing out on both Linux amd64 and arm64 (https://ci.ros2.org/job/ci_linux-aarch64/2843/consoleFull#console-section-665); not sure yet what is going on.
* Linux aarch64 failed Opensplice pendulum test; this is a known failure, so we can ignore.
* There's a potential null-dereference warning from clang on macOS (https://ci.ros2.org/job/ci_osx/5318/consoleFull#console-section-648).  While a potential problem, it is not because of Fast-RTPS, so ignoring.

So there seem to be 2 issues that need looking into here.

jacobperron, 2019-03-22:
@clalancette The `rclcpp_action_client` and `params_yaml` test failures are known issues (see https://github.com/ros2/build_cop/issues/166).

jacobperron, 2019-03-22:
You could try backporting https://github.com/ros2/rmw_fastrtps/pull/262 (but I think the tests are still flaky).

clalancette, 2019-03-25:
> @clalancette The `rclcpp_action_client` and `params_yaml` test failures are known issues (see [ros2/build_cop#166](https://github.com/ros2/build_cop/issues/166)).

Awesome, thanks.

> You could try backporting [ros2/rmw_fastrtps#262](https://github.com/ros2/rmw_fastrtps/pull/262) (but I think the tests are still flaky).

In that case, it is probably not worth it.  I just wanted to make sure that these weren't new failures caused by this change, and it seems they aren't.

nuclearsandwich, 2019-03-27:
Upgrading to Fast-RTPS 1.7.2 also lifts the restriction on using TinyXML 7.x for macOS with the inclusion of https://github.com/eProsima/Fast-RTPS/pull/349

MiguelCompany, 2019-04-23:
As the patch release is already out, I think this issue can be closed

nuclearsandwich, 2019-04-23:
Agreed. Thanks for the bump @MiguelCompany

closed 2019-04-23 by nuclearsandwich