ros2/ros2 #670: Corrupt files for OS X when downloading from release: Crystal patch3

LanderU, 2019-03-16:
## Bug report

**Required Info:**

- Operating System:
OS X
- Installation type:
  - Downloaded ROS2 release from here: https://github.com/ros2/ros2/releases/download/release-crystal-20190314/ros2-crystal-20190314-macos-amd64.tar.bz2
  - https://github.com/ros2/ros2/releases/tag/release-crystal-20190314
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/8d34d6550a77dcfe4828466e53f0cc5401be77fe
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
wget https://github.com/ros2/ros2/releases/download/release-crystal-20190314/ros2-crystal-20190314-macos-amd64.tar.bz2
bzip2 -t ros2-crystal-20190314-macos-amd64.tar.bz2
```

#### Expected behavior
I'm trying to download the compiled version of ROS2 for OS X, when I downloaded the files using the above command `wget`... The expected behavior when I use the command `bzip2 -t ros2-crystal-20190314-macos-amd64.tar.bz2 && echo $?` should be 0.

**Note:** This is the output that was have with the latest release version.
 
#### Actual behavior

```bash
lander@wks4:/tmp$ bzip2 -t ros2-crystal-20190314-macos-amd64.tar.bz2 
bzip2: ros2-crystal-20190314-macos-amd64.tar.bz2: data integrity (CRC) error in data

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

lander@wks4:/tmp$ echo $?
2
```

I also try to recover the data using `bzip2recover`but not luck.

#### Additional information

N/A

nuclearsandwich, 2019-03-17:
It looks like the archive didn't upload completely. For the last two releases I've been uploading artifacts via GitHub API directly from our packaging host which is significantly more efficient and, in theory, reliable than copying them to my workstation and uploading them again but clearly has some integrity issues.

nuclearsandwich, 2019-03-17:
I'm currently re-uploading the releases. I'll report back when that's completed.

nuclearsandwich, 2019-03-17:
macOS and Linux releases are all up (I've also added sha256 and md5 checksums to the release description) Windows releases are taking longer due to the artifact size. ETA 90-120 minutes

LanderU, 2019-03-17:
```bash
lander@MacBook-Pro-de-Lander:~$ md5 ros2-crystal-20190314-macos-amd64.tar.bz2 
MD5 (ros2-crystal-20190314-macos-amd64.tar.bz2) = 6ff524af3b2e09b9683c8328fc8dd585
```

Solved my issue, thanks a lot @nuclearsandwich.

closed 2019-03-17 by LanderU

nuclearsandwich, 2019-03-17:
The Windows releases have now also been updated.