ros2/ros2 #671: ROS 2 fails to build against tinyxml2 7.0.0

vmayoral, 2019-03-17:
## Bug report

**Required Info:**

- Operating System:
  - OS X Mojave 10.14
- Installation type:
  - from source
- Version or commit hash:
  - patch 3 release
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - not applicable

#### Steps to reproduce issue
Official instructions to build from source in OS X

#### Actual behavior

```bash
--- stderr: fastrtps
Undefined symbols for architecture x86_64:
  "tinyxml2::XMLDocument::Parse(char const*, unsigned long)", referenced from:
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(char const*, unsigned long, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
  "tinyxml2::XMLDocument::LoadFile(char const*)", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
  "tinyxml2::XMLDocument::XMLDocument(bool, tinyxml2::Whitespace)", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(char const*, unsigned long, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
  "tinyxml2::XMLDocument::~XMLDocument()", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::loadXML(char const*, unsigned long, std::__1::unique_ptr<eprosima::fastrtps::xmlparser::BaseNode, std::__1::default_delete<eprosima::fastrtps::xmlparser::BaseNode> >&) in XMLParser.cpp.o
  "tinyxml2::XMLElement::QueryIntText(int*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLParser::getXMLInt(tinyxml2::XMLElement*, int*, unsigned char) in XMLElementParser.cpp.o
  "tinyxml2::XMLElement::FindAttribute(char const*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
  "tinyxml2::XMLElement::QueryBoolText(bool*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLParser::getXMLBool(tinyxml2::XMLElement*, bool*, unsigned char) in XMLElementParser.cpp.o
  "tinyxml2::XMLElement::QueryUnsignedText(unsigned int*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLParser::getXMLUint(tinyxml2::XMLElement*, unsigned int*, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLUint(tinyxml2::XMLElement*, unsigned short*, unsigned char) in XMLElementParser.cpp.o
  "tinyxml2::XMLElement::GetText() const", referenced from:
      eprosima::fastrtps::xmlparser::XMLParser::getXMLBuiltinAttributes(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::BuiltinAttributes&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLDuration(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::Time_t&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLLocatorList(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::LocatorList_t&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLString(tinyxml2::XMLElement*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLHistoryMemoryPolicy(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::MemoryManagementPolicy&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLTransports(tinyxml2::XMLElement*, std::__1::vector<std::__1::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface>, std::__1::allocator<std::__1::shared_ptr<eprosima::fastrtps::rtps::TransportDescriptorInterface> > >&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::getXMLTopicAttributes(tinyxml2::XMLElement*, eprosima::fastrtps::TopicAttributes&, unsigned char) in XMLElementParser.cpp.o
      ...
  "tinyxml2::XMLElement::Attribute(char const*, char const*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::parseXMLStructDynamicType(tinyxml2::XMLElement*) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::parseXMLUnionDynamicType(tinyxml2::XMLElement*) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::parseXMLEnumDynamicType(tinyxml2::XMLElement*) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::parseXMLAliasDynamicType(tinyxml2::XMLElement*) in XMLParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLParser::parseXMLMemberDynamicType(tinyxml2::XMLElement*, eprosima::fastrtps::types::DynamicTypeBuilder*, unsigned int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in XMLParser.cpp.o
      ...
  "tinyxml2::XMLAttribute::QueryIntValue(int*) const", referenced from:
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
  "tinyxml2::XMLAttribute::Name() const", referenced from:
      void eprosima::fastrtps::xmlparser::XMLParser::addAllAttributes<eprosima::fastrtps::TopicAttributes>(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::DataNode<eprosima::fastrtps::TopicAttributes>&) in XMLParser.cpp.o
      void eprosima::fastrtps::xmlparser::XMLParser::addAllAttributes<eprosima::fastrtps::ParticipantAttributes>(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::DataNode<eprosima::fastrtps::ParticipantAttributes>&) in XMLParser.cpp.o
[... 20 lines cut ...]
      eprosima::fastrtps::xmlparser::XMLParser::getXMLPropertiesPolicy(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::PropertyPolicy&, unsigned char) in XMLElementParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLParticipantEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::xmlparser::StaticRTPSParticipantInfo*) in XMLEndpointParser.cpp.o
      eprosima::fastrtps::xmlparser::XMLEndpointParser::loadXMLNode(tinyxml2::XMLDocument&) in XMLEndpointParser.cpp.o
      ...
[... cut ...] traits<char>, std::__1::allocator<char> > const&) in XMLProfileManager.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/cpp/libfastrtps.1.7.0.dylib] Error 1
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make: *** [all] Error 2
```

### Additional information
Workaround: get back to tinyxml2 version 5.0.1 as explained at https://github.com/ros2/ros2/issues/440#issuecomment-353702899

vmayoral, 2019-03-17:
More than a ROS2 issue, AFAIK this affects fastrtps package thereby pinging @richiware.

nuclearsandwich, 2019-03-17:
The requirement for TinyXML2 <7 is documented on the [MacOS install instructions](https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/). Support was added in Fast-RTPS in https://github.com/eProsima/Fast-RTPS/pull/349 but Crystal is still using Fast-RTPS 1.7.0. Updating Fast-RTPS, which would remove the restriction is being discussed in https://github.com/ros2/ros2/issues/669

nuclearsandwich, 2019-03-17:
> Operating System:
> 
>     * OS X Mojave 10.14

As an additional note, Crystal is only supported on macOS Sierra. See [REP-2000](http://www.ros.org/reps/rep-2000.html#crystal-clemmys-december-2018-december-2019) for the full list of supported platforms.

vmayoral, 2019-03-17:
Thanks @nuclearsandwich, that seemed to do it and it's actually much better than the workaround I was producing:

```bash
brew uninstall --ignore-dependencies tinyxml2
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/0b3ee2f67043af1b4270096c13350420e3290399/Formula/tinyxml2.rb
ln -s /usr/local/opt/tinyxml2/lib/libtinyxml2.5.0.1.dylib /usr/local/opt/tinyxml2/lib/libtinyxml2.7.dylib
```
which then obliged me to get gazebo back to a version where libtinyxml 5.0.1 was accepted...

I'll close this.

closed 2019-03-17 by vmayoral