ros2/ros2 #672: ROS 2 Threat Model

thomas-moulard, 2019-03-18:
As a ROS 2 developer, I should have access to best practices to build a secure robotic product built using ROS 2.

Owner: @thomas-moulard 

## Exit Criteria

* A ROS 2 threat model document is contributed back to the community.

thomas-moulard, 2019-03-20:
Sent PR for review: https://github.com/ros2/design/pull/218

tfoote, 2019-03-28:
The first draft has been merged and the article is rendered at: http://design.ros2.org/articles/ros2_threat_model.html Please open new PRs for further discussion.

thomas-moulard, 2019-04-09:
Follow-up PR: https://github.com/ros2/design/pull/229

thomas-moulard, 2019-04-09:
@ryanewel Ryan, would you mind linking your PRs here as we open those follow-up PRs?

ryanewel, 2019-04-09:
Yep,
Combine assets and business goals: https://github.com/ros2/design/pull/227
Consider XRCE attacks: https://github.com/ros2/design/pull/231

thomas-moulard, 2019-04-17:
All PRs have been merged.

closed 2019-04-17 by thomas-moulard