ros2/ros2 #673: no colcon binary in osx binary package

snowdeer-seo, 2019-03-19:
## Bug report

There is no colcon in crystal binary package for OSX.
The previous version(bouncy) has executable file 'ament' in package,
but current version(crystal) has no `colcon`

- Operating System: OSX(Mojave)
- Installation type:Binary
- Version or commit hash:0314 release version

sloretz, 2019-03-19:
I assume you installed from [the instructions here](https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/#downloading-ros-2). While colcon is helpful for building packages, the binary archive is useful without it. Tools like rqt or rviz2 or the ros2 command line tools can still be used. I'm not sure colcon will ever be added to it.

You might be interested in [how to install colcon using pip](https://colcon.readthedocs.io/en/released/user/installation.html#using-pip-on-any-platform).

dirk-thomas, 2019-03-21:
I will go ahead and close this ticket assuming the question has been answered by the comment.

closed 2019-03-21 by dirk-thomas