ros2/ros2 #674: How to fix compile error

huchhong, 2019-03-19:
I use ros2.repos in master branch (commit id: a76d265cd57d1b0) to get source and compile it and get following error:

```
--- stderr: rosidl_generator_py
/usr/bin/ld: cannot find -lrosidl_generator_py_custom__rosidl_typesupport_connext_c
collect2: error: ld returned 1 exit status
make[2]: *** [rosidl_generator_py/rosidl_generator_py/rosidl_generator_py_s__rosidl_typesupport_connext_c.cpython-36m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_py__rosidl_typesupport_connext_c__pyext.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lrosidl_generator_py_custom__rosidl_typesupport_opensplice_c
collect2: error: ld returned 1 exit status
make[2]: *** [rosidl_generator_py/rosidl_generator_py/rosidl_generator_py_s__rosidl_typesupport_opensplice_c.cpython-36m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_py__rosidl_typesupport_opensplice_c__pyext.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_py        [ Exited with code 2 ]
Aborted  <<< rosidl_typesupport_cpp
Aborted  <<< rviz_rendering                                                                                                                                                                                 
                                                                                                                                                                                                            
Summary: 118 packages finished [39.5s]                                                                                                                                                                      
  1 package failed: rosidl_generator_py                                                                                                                                                                     
  2 packages aborted: rosidl_typesupport_cpp rviz_rendering                                                                                                                                                 
  1 package had stderr output: rosidl_generator_py                                                                                                                                                          
  112 packages not processed

```

It seems to me that it can't find dependency of openslice and rti context which i don't care. I prefer Fast-RTPS for its free license. So if fast-rtps is the default dds implementation, why it takes this dependency failed as error instead of warning.  

And 112 packages not processed ? This scared me, where can i find which packages not being processed ?

jacobperron, 2019-03-19:
Please provide the required info in the issue template (e.g. which operating system and commands used to reproduce the error).

> And 112 packages not processed ? This scared me, where can i find which packages not being processed ?

The unprocessed packages are due to the build error (packages below `rosidl_generator_py` in the dependency tree are skipped).

huchhong, 2019-03-20:
**Required Info:**

- Operating System:
  - ubuntu
- Installation type:
  - from source

After make sure i installed all the required package, i make a fresh compile again,  then i get this output:
```
Summary: 233 packages finished [25min 18s]
  10 packages had stderr output: qt_gui_cpp rmw_connext_cpp rmw_connext_shared_cpp rmw_opensplice_cpp ros1_bridge rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp rqt_gui_cpp
```
this seems ok to me now.  So i may close this issue.

closed 2019-03-20 by huchhong

jacobperron, 2019-03-20:
I think the stderr output you see is because you don't have Connext or OpenSplice installed (which is fine). 
`qt_gui_cpp` and `rqt_gui_cpp` do not build on Ubuntu 16.04, and so the [build instructions](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#build-the-code-in-the-workspace) recommend ignoring them with the following command:

        colcon build --symlink-install --packages-ignore qt_gui_cpp rqt_gui_cpp