ros2/ros2 #675: Question about binary release version in ubuntu

huchhong, 2019-03-19:
Today, i updated crystal by 'apt update && apt install ros-crystal-desktop', the output is:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be upgraded:
  ros-crystal-desktop
1 upgraded, 0 newly installed, 0 to remove and 575 not upgraded.
Need to get 0 B/5,448 B of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 277815 files and directories currently installed.)
Preparing to unpack .../ros-crystal-desktop_0.6.1-0bionic.20190312.220507_amd64.deb ...
Unpacking ros-crystal-desktop (0.6.1-0bionic.20190312.220507) over (0.6.1-0bionic.20181214.045017) ...
Setting up ros-crystal-desktop (0.6.1-0bionic.20190312.220507) ...
```

In the above output, there is version that is 0.6.1-0bionic.20190312.220507, my question is what is this 0.6.1 version? I can't find related tag or release number in ros2 repo, and i also noticed that there was a Crystal Clemmys Patch Release 3 which also named ros2-crystal-20190314. So what does this 20190312.220507 means in my output? 

Because of this chaotic  naming,  i can't figure out which version i have updated to.

sloretz, 2019-03-19:
I believe `0.6.1` is the version of the package `desktop` https://github.com/ros2/variants/tree/master/desktop , and `20190312.220507` is a time stamp that says when the package was built by the build farm.

huchhong, 2019-03-19:
So ros-crystal-desktop is not directly related to release of this ros2 repo?  Otherwise why was desktop build earlier than patch3 release.

sloretz, 2019-03-19:
I'm not sure how to answer that. @nuclearsandwich mind correcting me where I answer this wrong?

A patch release is an announcement that updates for a lot of packages have been made available. The artifacts have to be built before that announcement is made. I believe `20190314` is both the date that the patch release was announced and the date the big archives containing all packages were built. What you updated was a debian package, which has a different date from the big archives because it was built earlier.

huchhong, 2019-03-19:
ok, i understand now. thanks.

closed 2019-03-19 by huchhong

nuclearsandwich, 2019-03-19:
No corrections needed @sloretz.

ROS 2 overall does not use a numbered version scheme since it's an amalgam of many packages. An example of a fully qualified ROS 2 release would be "ROS 2 Crystal Clemmys 20190314. The numbers  (patch 1, patch 2, etc) on the patch releases are really just there to give us a handle on a specific patch before the release date is set. The [tags](https://github.com/ros2/ros2/releases) on this repository take the form `release-$ROS_DISTRO-$YYYMMDD` which identify the release versions of packages in the ros2.repos file at the time of release. Packages on the ROS buildfarm can be re-released and built at any time up-to that date. Releasing and building debian packages earlier than the release date gives us time to use the packages in the testing repo before the full release. We encode the build timestamp into the version number to make it easier to invalidate packages when a newer version of a dependency is added to the repository.