ros2/ros2 #677: Message namespaces returned from rmw layer are always 'msg'

jacobperron, 2019-03-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - Latest [ros2.repos](https://github.com/ros2/ros2/blob/035165820eed3f1b760c8868162e5bfd5bd8c299/ros2.repos)
- DDS implementation:
  - Fast-RTPS and RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Not the most succinct example, but demonstrates the roadblock for the forthcoming action graph API:

1. Check out new action graph API [rcl [branch: action_graph]](https://github.com/ros2/rcl/tree/action_graph).

    mkdir -p action_graph_ws/src
    cd action_graph_ws
    git clone https://github.com/ros2/rcl -b action_graph src/rcl

2. Build (overlay on existing ros2 workspace)

        . ~/ros2_ws/install/setup.bash
        colcon build

3. Run the `rcl_action` graph test with Fast-RTPS, Connext, and Opensplice

        . install/local_setup.bash
        build/rcl_action/test_graph__rmw_fastrtps_cpp
        RMW_IMPLEMENTATION=rmw_connext_cpp build/rcl_action/test_graph__rmw_connext_cpp
        RMW_IMPLEMENTATION=rmw_opensplice_cpp build/rcl_action/test_graph__rmw_opensplice_cpp

#### Actual behavior

The Fast-RTPS and Connext tests pass, but the OpenSplice test fails due to unexpected message types, e.g.

        Expected equality of these values:
          nat.types[0].data[0]
            Which is: "test_msgs::action::dds_::Fibonacci_"
          "test_msgs/Fibonacci"

This occurs because the type demangling in the rmw layer, expects to find the `msg` namespace, but finds `action`:

https://github.com/ros2/rmw_opensplice/blob/b38080cf4927aad0313edeab16e5eadab6f15cf9/rmw_opensplice_cpp/src/demangle.cpp#L39

#### Expected behavior

It turns out that OpenSplice is exhibiting the correct behavior, since the aciton message types **are** defined in the `action` namespace.

The tests are incorrectly passing for Fast-RTPS and Connext as they report that the action types are in the `msg` namespace (which they are not).

#### Additional information

Looking closer at the `rmw_fastrtps` and `rmw_connext`, we see that they hard-code registered topics and servers with the namespace `msg` and `srv` respectively:

https://github.com/ros2/rmw_fastrtps/blob/c9e496560d098b0de2e12aee9af31b1b990156ce/rmw_fastrtps_cpp/src/rmw_publisher.cpp#L104

https://github.com/ros2/rmw_fastrtps/blob/c9e496560d098b0de2e12aee9af31b1b990156ce/rmw_fastrtps_cpp/src/rmw_service.cpp#L118-L119

https://github.com/ros2/rmw_connext/blob/43d6833977ddfc3d8fa599a51c7b01bcb3008fa3/rmw_connext_cpp/src/rmw_publisher.cpp#L87

Although OpenSplice has a similar implementation:

https://github.com/ros2/rmw_opensplice/blob/b38080cf4927aad0313edeab16e5eadab6f15cf9/rmw_opensplice_cpp/src/rmw_publisher.cpp#L105

It is not immediately clear why it exhibits the correct behavior; more investigation is required.

jacobperron, 2019-03-27:
The proposed solution involves two major changes:
1. Add a field for message namespace to the typesupport structs passed to the `rmw` layer. Related PRs:
  - https://github.com/ros2/rosidl/pull/355
  - https://github.com/ros2/rosidl_typesupport_fastrtps/pull/18
2. Updating the demangling logic in the `rmw` layer to return a tuple for topic/service type names. This consequently affects many repos as a central datatype was changed in https://github.com/ros2/rmw/pull/170.

I've opened an initial set of PRs for resolving this issue that focus on changes to `rmw_fastrtps`:

- https://github.com/ros2/rmw_fastrtps/pull/266
- https://github.com/ros2/rosidl_typesupport_fastrtps/pull/18
- https://github.com/ros2/rcutils/pull/144
- https://github.com/ros2/rmw/pull/170
- https://github.com/ros2/rosidl/pull/355
- https://github.com/ros2/rcl/pull/407
- https://github.com/ros2/rclpy/pull/294
- https://github.com/ros2/rclcpp/pull/670
- https://github.com/ros2/rviz/pull/387

If the proposal is approved, then I will make similar patches to `rmw_connext` and `rmw_opensplice`, before merging everything.

CI for the changes is tracked here https://github.com/ros2/rmw_fastrtps/pull/266#issuecomment-476906308

hidmic, 2019-04-04:
@jacobperron this issue has the `in review` label but many of the connected PRs don't, is that on purpose?

jacobperron, 2019-05-04:
I've updated the `rmw_fastrtps` and `rmw_connext` and typesupport so they provide an arbitrary message namespace. I've also renabled a previously failing test related to OpenSplice, which originally brought about this issue (https://github.com/ros2/rcl/pull/407/commits/15f38afa069db8aa16d9dceda785a860da430c85).

CI up to `rcl_action`: [image]

jacobperron, 2019-05-06:
Full CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

jacobperron, 2019-05-08:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2019-05-08:
> * Linux

Addressed linter warnings in https://github.com/ros2/rmw_fastrtps/pull/266/commits/419a3cf8c66286108d5e57c496ee0b054dec41cf

dirk-thomas, 2019-05-08:
While all but one referenced PRs have been merged I will leave this open until ros2/rviz#387 has been closed.

jacobperron, 2019-05-10:
> While all but one referenced PRs have been merged I will leave this open until ros2/rviz#387 has been closed.

Merged.

closed 2019-05-10 by jacobperron