ros2/ros2 #680: [ROS2 QoS meta-ticket] Dealine, Liveliness, and Lifespan

nburek, 2019-03-29:
As a ROS 2 developer, I want to have some guarantees on the delivery of my messages and notification of when my system fails to meet those policies. These new QoS settings provide additional methods of specifying the profile for the messages in your system. Additionally they provide a feedback mechanism to know when your specified deadline and liveliness have been violated. 

Owner: @ros2/aws-robotics

# Exit Criteria
The new QoS features are implemented and available using the rclcpp and rclpy client libraries with either rmw_fastrtps, rmw_connext, or rmw_opensplice. 

# Tasks
[] Interface changes for optional parameters on Publisher/Subscriber creation in rclcpp
[] New interfaces for event callbacks available in rclcpp
[] New interfaces for event callbacks available in rclpy
[] New functions for getting event occurrences available in rcl
[] New functions for getting event occurrences available in rmw
[] New functions for getting event occurrences available in rmw_connext
[] New functions for getting event occurrences available in rmw_opensplice
[] New functions for getting event occurrences available in rmw_fastrtps
[] Support for Deadline in FastRTPS
[] Support for Lifespan in FastRTPS
[] Support for Liveliness manual by Topic/Participant in FastRTPS
[] Integrate new release of these features in FastRTPS 1.8.0 with rmw_fastrtps
[] New demos for deadline, liveliness, and lifespan
[] New system tests for deadline, liveliness, and lifespan

wjwwood, 2019-04-08:
I find it pretty difficult to track which pr's are meant to be reviewed together and to be merged together. For instance, have a look at the issues for this feature on waffle.io:

[image]

None of them are connected (see the connects to feature: https://help.waffle.io/faq/waffle-workflow/use-waffles-connect-keyword-to-connect-prs-to-issues).

Here's an example of that:

[image]

Also, many of the issues/pr's are in progress and some are in review (as denoted by the labels on the issues/pr's). This issue isn't in progress or in review, I just moved it to in progress so it shows up next to it's peers.

Waffle will be going away soon, but it's still how we organize these for now, and hopefully in the future our new solution will work in a similar fashion.

Aside from the tool specifics, it would be nice to have pr's that should be reviewed and merged together be grouped somehow. I see in some places but not others a `QoS N` where N might be 0, 1, 2, etc... but I don't see that in other places. There are the two phases above, but they seems to be higher level that what I'm looking for in a grouping, as (for example) there are several pr's in flight for both rcl and rclpy, which presumably should not be merged a the same time?

nburek, 2019-04-09:
Hey William, 

Sorry for the organization difficulties. Outside of rclpy, I think that most of these are pretty tightly coupled at the moment. Here is my initial breakdown of what is coupled and I'll work with the people who opened each PR to get them marked with the proper "Connects to *" syntax. 

### Make parameters optional
ros2/rclcpp#673

### New QoS Features and Events in rcl layer
ros2/rcl#408
aws-robotics/rcl#6
ros2/rclpy#305
ros2/rmw#171
https://github.com/ros2/rmw_implementation/pull/60
ros2/rmw_connext#352
ros2/rmw_fastrtps#271
rclcpp - need a new PR open for this

### Assert Liveliness Functions
aws-robotics/rcl#7
aws-robotics/rmw_fastrtps#4

### Event Callbacks available
ros2/demos#320

Unfortunately our team cannot change the tags on the PRs ourselves, but I'll make sure that we are active about making a comment stating when a tag should be added/changed.

nburek, 2019-05-16:
The latest update on this is that all of the PRs for the features have been merged. The one feature that was not able to be done in time is Liveliness support for rmw_fastrtps. Liveliness support is currently planned for Fast-RTPS release 1.8.1 and we look at enabling support for it in rmw_fastrtps at that point. 

The only outstanding PRs are for the demo and system_test packages, which we hope to get merged in after feature freeze.

dirk-thomas, 2019-05-16:
> The only outstanding PRs are for the demo and system_test packages, which we hope to get merged in after feature freeze.

Please try to land these as soon as possible since especially the `demo` PR will be necessary for the upcoming testing phase.

nburek, 2019-05-16:
@dirk-thomas Both the PRs are in a place where they're ready for follow up reviews. I assumed they would be lower priority for the next couple of days since they aren't blocking feature freeze, but as soon as OSRF has bandwidth to review them again we can get them merged.
https://github.com/ros2/system_tests/pull/347
https://github.com/ros2/demos/pull/320

emersonknapp, 2019-09-06:
@nburek do we think this ticket is done, or only once we land the design pr?

wjwwood, 2019-09-09:
Can these be checked?

> system_tests - New system tests for deadline, liveliness, and lifespan ros2/system_tests#347

> rmw_fastrtps - Support for Liveliness

emersonknapp, 2019-09-09:
Yes - all of the Phase 2 checkboxes have been completed. Nick's on vacation though so I'm not sure how we can edit it.

wjwwood, 2019-09-09:
I checked them off, I'll have another look at the design document asap.

wjwwood, 2019-09-24:
I'm going to close this since all the original tasks laid out in it have been accomplished.

closed 2019-09-24 by wjwwood