ros2/ros2 #682: Publishing in Python with custom message fails.

capaulson, 2019-04-02:
## Bug report

**Required Info:**

- Operating System:
  - Mac OSX 
- Installation type:
  - Latest binary
- Version or commit hash:
  - 8d34d65
- DDS implementation:
  - 
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Trying to publish on a custom message type (float32 for testing, built using demos/pendulum_msgs example).

The python code is: 

```
import rclpy
from rclpy.node import Node
from btmsg.msg import Bt ## Custom message that is simply 'float32 position'

class MinimalPublisher(Node):

    def __init__(self):
        super().__init__('minimal_publisher')
        self.publisher_ = self.create_publisher(Bt, 'topic')
        timer_period = 0.5  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.i = 0

    def timer_callback(self):
        msg = Bt()
        msg.position = 3.14159265
        self.publisher_.publish(msg)
        self.i += 1

rclpy.init(args=None)

minimal_publisher = MinimalPublisher()
rclpy.spin(minimal_publisher)
minimal_publisher.destroy_node()
rclpy.shutdown()

```

#### Expected behavior
Topic should be published. 

#### Actual behavior
I get the error message: 
```
RuntimeError: Failed to create publisher: type support not from this implementation, at /Users/osrf/jenkins-agent/workspace/packaging_osx/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_publisher.cpp:81, at /Users/osrf/jenkins-agent/workspace/packaging_osx/ws/src/ros2/rcl/rcl/src/rcl/publisher.c:173
```

This error occurs at ``` self.publisher_ = self.create_publisher(Bt, 'topic') ```
I get the same message when publishing from the command line. I'm not sure where to start on debugging. The '/User/osrf' paths seem to be some indication of issues. Those aren't local to this machine.  

#### Additional information

zhaxiche, 2019-11-28:
Hi, capaulson, have you figured out what's the error there? I'm looking for a tutorial or something about creating a custom message type in ROS2 using python. Right now I think they only support ament_cmake? If you have cleaned the bug, mind if you could teach me how to create a custom message?

jacobperron, 2019-12-02:
Sorry for the delayed response.
I'm not able to reproduce the issue with Linux in Dashing or Eloquent.

@capaulson, are you still running into this error? If so, perhaps you can share the steps you did to build the custom message type?

---

> Right now I think they only support ament_cmake?

@zhaxiche Correct, currently the only way to build custom messages is from an ament_cmake package. Unfortunately, our [tutorial](https://index.ros.org/doc/ros2/Tutorials/Defining-custom-interfaces-(msg-srv)/) is lacking at the moment. Slightly related is [this tutorial on defining a custom action](https://index.ros.org/doc/ros2/Tutorials/Actions/Creating-an-Action/). For now, I suggest you take a look at [common_interfaces](https://github.com/ros2/common_interfaces) for packages as examples.

capaulson, 2019-12-03:
I'll need to dig through notes on how I fixed this. I recall that the apparent local path in the error was a bit of a red herring. Ultimately I had made a mistake in the message definition and that error was the outcome.

jacobperron, 2019-12-03:
Sounds like user error and not a bug, so I'll close this for now. Posting the your resolution would be good for future readers, thanks!

closed 2019-12-03 by jacobperron