ros2/ros2 #684: Crystal Patch Release 5

nuclearsandwich, 2019-04-04:
This issue will track the list of changes that should be included in the next Crystal patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

---

#### Changes proposed for backport this patch release

#### Changes in progress for this release

---

When the patch release date is set, we'll start tracking the list of every repository that needs to be released and review changes to those repositories for compatibility with Crystal to prevent breaking ABI/API unintentionally.

dirk-thomas, 2019-04-04:
Consider backporting ros2/rosidl_typesupport_opensplice#23 to address ros2/rmw_opensplice#259.

Karsten1987, 2019-04-07:
rosbag2 release for sourcing ros1 capabilities: https://github.com/ros2/rosbag2/issues/94

dirk-thomas, 2019-04-08:
> These PRs (ros2-gbp/rosbag2-release#3) should fix the problem when releasing the package the next time.

Since this patch in the release repo affects only the Debian package I would suggest to do a new release right away and sync the fix - there is no reason to wait with it until the next patch release.

dirk-thomas, 2019-04-17:
Consider backporting ros2/rcl#418 to address ros2/rclpy#302.

dirk-thomas, 2019-04-17:
> Since this patch in the release repo affects only the Debian package I would suggest to do a new release right away and sync the fix - there is no reason to wait with it until the next patch release.

@Karsten1987 where you able to release a new deb inc of `rosbag2`?

dirk-thomas, 2019-05-01:
> where you able to release `rosbag2`?

Done in ros/rosdistro#21068.

dirk-thomas, 2019-07-29:
Consider this backport ros2/rosidl_typesupport_opensplice#36 to stay compatible with newer versions of OpenSplice.

nuclearsandwich, 2019-08-15:
> Consider backporting [ros2/rosidl_typesupport_opensplice#23](https://github.com/ros2/rosidl_typesupport_opensplice/pull/23) to address [ros2/rmw_opensplice#259](https://github.com/ros2/rmw_opensplice/issues/259).

Backported in https://github.com/ros2/rosidl_typesupport_opensplice/pull/39

nuclearsandwich, 2019-08-21:
Linux Bionic AMD64 [image]
Linux Xenial AMD64 [image]

Linux Bionic ARM64 [image]
Linux Xenial ARM64 [image]

Mac OS [image]

Windows Debug [image]
Windows Release [image]

nuclearsandwich, 2019-08-21:
The release is [out](https://discourse.ros.org/t/patch-release-and-new-packages-for-ros-2-crystal-2019-08-21/10367)!

PR #763 (2019-08-21): Crystal Patch 5
Closes #684
closed 2019-08-21 by nuclearsandwich