ros2/ros2 #696: ROS2 rviz2 launch failed: Segmentation fault: 11

zuxinrui, 2019-05-02:
## Bug report
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
Segmentation fault: 11
**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
First install binary packages and dependencies according to the official guide, then run source ~/ros2_install/ros2-ox/setup.bash
then run rviz2 directly
```

#### Expected behavior
rviz2 GUI successfully launched 
#### Actual behavior
rviz2 GUI failed to start up
#### Additional information
I tried to reinstall the whole procedure again and still got the issue. Then I run “rviz2” in a quick repetition several times :), then it went well once...

----
## Feature request

#### Feature description

#### Implementation considerations

Karsten1987, 2019-05-02:
This might be related to https://github.com/ros2/rviz/pull/380. Please have a look at the open PR and see if this is working for you. If not, please open another issue on the rviz repository. In any case it helps to provide a few more details about which operating system you're using and if you're following the binary instructions or compiling it from source.

closed 2019-05-02 by Karsten1987

gaurishg, 2024-02-25:
It could be because of unsupported GPU drivers.
Try `LIBGL_ALWAYS_SOFTWARE=1 rviz2`