ros2/ros2 #697: ROS2 fails installation on MAC on package rviz_ogre_vendor: java.swg

asimonov, 2019-05-05:
## Bug report

**Required Info:**

- Operating System:
  - MasOS High Sierra
- Installation type:
  - crystal from source
- Version or commit hash:
  - crystal patch 4
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
just following https://index.ros.org/doc/ros2/Installation/OSX-Development-Setup/ 

#### Expected behavior
colcon succeeds

#### Actual behavior
```
...
Starting >>> ament_clang_format
Starting >>> ament_pclint
Starting >>> ros_environment
--- stderr: rviz_ogre_vendor                                                                                                                    
:3: Error: Unable to find 'java.swg'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/Components/RTShaderSystem/include/OgreRTShader.i:8: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:19: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:20: Error: Unable to find 'std_pair.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:21: Error: Unable to find 'std_map.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:25: Error: Unable to find 'std_vector.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:27: Error: Unable to find 'typemaps.i'
:3: Error: Unable to find 'java.swg'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/Components/Overlay/include/OgreOverlay.i:20: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:19: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:20: Error: Unable to find 'std_pair.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:21: Error: Unable to find 'std_map.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:25: Error: Unable to find 'std_vector.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:27: Error: Unable to find 'typemaps.i'
:3: Error: Unable to find 'java.swg'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:19: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:20: Error: Unable to find 'std_pair.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:21: Error: Unable to find 'std_map.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:25: Error: Unable to find 'std_vector.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:27: Error: Unable to find 'typemaps.i'
:3: Error: Unable to find 'java.swg'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/Components/Bites/include/OgreBites.i:20: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:19: Error: Unable to find 'std_string.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:20: Error: Unable to find 'std_pair.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:21: Error: Unable to find 'std_map.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:25: Error: Unable to find 'std_vector.i'
/Users/alexeysimonov/dev/ros2_crystal_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.i:27: Error: Unable to find 'typemaps.i'
make[5]: *** [Components/Java/CMakeFiles/OgreJNI.dir/OgreJAVA.stamp] Error 1
make[5]: *** Deleting file `Components/Java/CMakeFiles/OgreJNI.dir/OgreJAVA.stamp'
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [Components/Java/CMakeFiles/OgreJNI.dir/OgreRTShaderJAVA.stamp] Error 1
make[5]: *** Deleting file `Components/Java/CMakeFiles/OgreJNI.dir/OgreRTShaderJAVA.stamp'
make[5]: *** [Components/Java/CMakeFiles/OgreJNI.dir/OgreOverlayJAVA.stamp] Error 1
make[5]: *** Deleting file `Components/Java/CMakeFiles/OgreJNI.dir/OgreOverlayJAVA.stamp'
make[5]: *** [Components/Java/CMakeFiles/OgreJNI.dir/OgreBitesJAVA.stamp] Error 1
make[5]: *** Deleting file `Components/Java/CMakeFiles/OgreJNI.dir/OgreBitesJAVA.stamp'
make[4]: *** [Components/Java/CMakeFiles/OgreJNI_swig_compilation.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
Aborted  <<< ament_clang_format
Aborted  <<< ament_pclint
Aborted  <<< ros_environment
...
```

Looks like this is due to default swig in homebrew being on version 4 now:

```
(ros2_crystal) alexeysimonov@courage:~/dev/ros2_crystal_ws$ ll /usr/local/share
...
lrwxr-xr-x    1 alexeysimonov  admin    31  5 May 17:31 swig@ -> ../Cellar/swig/4.0.0/share/swig
...
```

but looks like 3.0.12 is expected in build cmakelist:

```
//Path to a file.
SWIG_DIR:PATH=/opt/local/share/swig/3.0.12

//Path to a program.
SWIG_EXECUTABLE:FILEPATH=/opt/local/bin/swig

//Swig version
SWIG_VERSION:STRING=3.0.12
```

The following seem to fix it (going back to 3.0.12 brew version of swig)
```
brew unlink swig
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/97d5c0caeab23c3f7765986f01b2f5ea5afbfdbb/Formula/swig.rb
```

sloretz, 2019-05-20:
Possibly related to #700

asimonov, 2019-05-26:
@sloretz i think this is different from #700, that's why i created both issues

Karsten1987, 2019-07-24:
In https://github.com/ros2/rviz/pull/414 we disabled the java components of OGRE all together. Can you verify that this issue here still persists?

asimonov, 2019-07-25:
just checked my records - I did not see this issue while building ROS2 Dashing on brand new macbook pro with Mojave, right after Dashing was released

closed 2019-07-25 by asimonov