ros2/ros2 #698: Messages published too early after Python startup do not get sent

tobiasblass, 2019-05-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Both, source installation from https://github.com/ros2/ros2 and binary installation as described in https://index.ros.org/doc/ros2/Installation/Linux-Install-Binary/
- Version or commit hash:
  - ros2-crystal-20190408
- DDS implementation:
  - The default, i.e., FastRTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1. Copy the following python script to `/tmp/reproduce_bug.py`:
```Python
#!/usr/bin/env python3
import rclpy
import rclpy.node
import time
from std_msgs.msg import String

rclpy.init()
n = rclpy.node.Node('test')
#time.sleep(1)
pub = n.create_publisher(String, "/test")
pub.publish(String(data="This only arrives if the time.sleep(1) is executed"))
```

2. Run `ros2 topic echo /test std_msgs/String`

3. Run the python script
#### Expected behavior
- The message is sent on the `/test` channel and is printed by the ros2 topic call.
- Adding the `time.sleep()` statement does not change anything
#### Actual behavior
- Nothing happens
- If the `time.sleep()` statement is added, the message is sent successfully.
#### Additional information
The bug does not appear under the previous release `ros2-crystal-20190314`.

dirk-thomas, 2019-05-07:
That is to be expected behavior since it takes a non-zero amount of time to discover and establish a connection between the publisher and potential subscribers. The timing determines which of tho events happens first - the subscribers being matched or the message being publisher without any subscribers being matched yet.

tobiasblass, 2019-05-07:
Thanks for the explanation, that makes sense. I guess there were some changes in the lower layers between the two versions that  extended this window, which explains why it seemed to work before.

Is that behavior documented somewhere? I would have expected publishers to look for potential subscribers before publishing the first message. Or is this behavior simply inherited from the DDS backend?

dirk-thomas, 2019-05-07:
It isn't explicitly documented at the moment.

This behavior is not specific to DDS but even present in ROS 1. It is common to all anonymous publish / subscribe systems where the publisher doesn't make assumptions about subscribers.

tobiasblass, 2019-05-07:
Ah, I see. So because your code shouldn't assume that there even is a subscriber to begin with, it doesn't make sense to put in extra effort to locate any subscribers before the first publication. That makes a lot of sense but isn't entirely obvious, so it should probably be written down somewhere. I'll see if I can find a good place to add it in the tutorial or the examples.

Let me put in one last question before I close the ticket: Is the correct way to do this to only use topics if you can handle the message getting lost (e.g., because the data is periodic anyway), and to use services in combination with `service_is_ready()` for any messages that have to arrive and where you want to wait for a receiver?

closed 2019-05-13 by tobiasblass