ros2/ros2 #699: Fails to publish message. failed to load shared library of rmw implementation.

fldvrees, 2019-05-08:
## Bug report
- Operating System:
  - Windows 10 version 1809
- Installation type:
  - binaries
- Version or commit hash:
  - ros2-**crystal**-20190408-windows-release-amd64.zip e0e4b6b42d1b200891971b1cadeac90f0c39aa09
- DDS implementation:
  - Open Splice 6.9.190403
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1. Follow the [windows binary install instructions](https://index.ros.org/doc/ros2/Installation/Windows-Install-Binary/#windows-install-binary-installing-prerequisites).
2.  Run `call C:\dev\ros2\local_setup.bat`
3. Run `ros2 topic pub /chatter std_msgs/String "data: Hello"`

#### Expected behavior
ros2 starts publishing to the /chatter topic.

#### Actual behavior
Returns error message:
``` powershell
C:\Users\flori>call C:\dev\ros2\local_setup.bat
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.

C:\Users\flori>ros2 topic pub /chatter std_msgs/String "data: Hello"
C:\dev\ros2\Lib\site-packages\ros2topic\verb\pub.py:77: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  values_dictionary = yaml.load(values)
Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:136, at C:\J\workspace\packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:55
```

#### Additional information
- the `C:\dev\ros2\bin/rmw_fastrtps_cpp.dll` exists.
- Running the same scenario from a Administrator Command Prompt doesnt help
- When doing the same with the ROS2 debug binaries (ros2-crystal-20190408-windows-**debug**-amd64.zip), I get the same error, and some extra error messages:
 ``` powershell
C:\Users\flori>call C:\dev\ros2\local_setup.bat
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.

C:\Users\flori>ros2 topic pub /chatter std_msgs/String "data: Hello"
C:\dev\ros2\Lib\site-packages\ros2cli\entry_points.py:89: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  .format_map(locals()))
Failed to load entry point 'create_key': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
Failed to load entry point 'create_keystore': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
Failed to load entry point 'create_permission': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
Failed to load entry point 'distribute_key': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
Failed to load entry point 'generate_policy': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
Failed to load entry point 'list_keys': cannot import name 'etree' from 'lxml' (c:\python37\lib\site-packages\lxml\__init__.py)
C:\dev\ros2\Lib\site-packages\ros2topic\verb\pub.py:77: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  values_dictionary = yaml.load(values)
Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\packaging_windows_debug\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:136, at C:\J\workspace\packaging_windows_debug\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:55
```
- Could this have anything to do with the combination of `\` and `/` in `C:\dev\ros2\bin/rmw_fastrtps_cpp.dll`? 
- This issue is somewhat simulate to these threads, but none of the proposed solutions worked for me.
  - ["failed to create process" when trying to run ros2 command.](https://github.com/ros2/ros2/issues/525#issuecomment-403319623)
  - [Windows run ros2 demo FAILED The C extension 'D:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system](https://github.com/ros2/ros2/issues/421)
  - [R2B3 Windows - Failed to create process - Installation Test](https://discourse.ros.org/t/r2b3-windows-failed-to-create-process-installation-test/2896/3)
  - [ROS2 Windows failed to load shared library of rmw implementation](https://answers.ros.org/question/301140/ros2-windows-failed-to-load-shared-library-of-rmw-implementation/)

clalancette, 2020-08-13:
Sorry, this somehow fell through the cracks.  I know this is really old now, but are you still having problems with the latest ROS 2 Dashing, Eloquent, or Foxy releases?

FlorisDevreese, 2020-08-19:
@clalancette  I'm currently not working with ROS anymore, so I don't have an environment to replicate this issue. Sorry for that.
Feel free to close this issue.

clalancette, 2020-08-19:
> @clalancette I'm currently not working with ROS anymore, so I don't have an environment to replicate this issue. Sorry for that.
> Feel free to close this issue.

Thanks for the response.  Sorry we missed this issue, but feel free to reach out again if you start working with ROS again.

closed 2020-08-19 by clalancette