ros2/ros2 #700: ROS2 Crystal or master build on MAC. which XCode to use?

asimonov, 2019-05-08:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 10.13.6 High Sierra
- Installation type:
  - from source
- Version or commit hash:
  - tried both crystal patch 4 and master (as of May 8, 2019)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Followed instructions here: https://index.ros.org/doc/ros2/Installation/OSX-Development-Setup/

Started with fresh Homebrew.

Tried XCode 9.4.1, 10.0, 10.1. They all give the same error below:

```
alexeysimonov@courage:~/dev/ros2_dashing_ws$ colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_flake8
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> ament_cppcheck
Starting >>> tinydir_vendor
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Finished <<< ament_package [1.23s]                                                                                                                            
Starting >>> ament_cmake_core                                                                                                               
Finished <<< ament_flake8 [1.49s]
Starting >>> ament_pep257
Finished <<< ament_cppcheck [1.51s]
Starting >>> tinyxml2_vendor                                                                                                                                 
Finished <<< osrf_pycommon [2.46s]                                                                                                                            
Finished <<< fastcdr [2.47s]                                                                                                      
Finished <<< poco_vendor [2.47s]
Finished <<< gtest_vendor [2.48s]
Starting >>> fastrtps
Starting >>> gmock_vendor
Starting >>> tinyxml_vendor
Starting >>> urdfdom_headers
Finished <<< tinydir_vendor [2.53s]
Starting >>> orocos_kdl
Finished <<< ament_cmake_core [1.53s]                                                                                                                   
Starting >>> ament_cmake_test
Finished <<< ament_pep257 [1.51s]
Starting >>> ament_cmake_libraries
Finished <<< tinyxml2_vendor [1.53s]
Starting >>> ament_cmake_export_libraries
Finished <<< tinyxml_vendor [0.70s]                                                                                              
Starting >>> ament_cmake_include_directories
Finished <<< gmock_vendor [1.06s]                                                                                                 
Starting >>> ament_cmake_export_definitions                                                                                        
Finished <<< urdfdom_headers [1.14s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_test [0.84s]                                                                                              
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_export_libraries [1.21s]                                                                                       
Finished <<< ament_cmake_libraries [1.24s]
Starting >>> ament_cmake_export_dependencies                                                                                                         
Starting >>> ament_cmake_python
Finished <<< orocos_kdl [2.15s]                                                                                                                        
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_include_directories [1.53s]                                                               
[... 116 lines cut ...]
Aborted  <<< test_interface_files
Aborted  <<< test_osrf_testing_tools_cpp
Aborted  <<< ament_clang_format
Aborted  <<< ament_pclint                                                                           
Aborted  <<< urdfdom                                              

Summary: 58 packages finished [21.6s]
  1 package failed: rviz_ogre_vendor
  7 packages aborted: ament_clang_format ament_cmake_uncrustify ament_pclint shared_queues_vendor test_interface_files test_osrf_testing_tools_cpp urdfdom
  1 package had stderr output: rviz_ogre_vendor
  186 packages not processed
The file /Users/alexeysimonov/dev/ros2_dashing_ws/None does not exist.

```

#### Expected behavior

Build success

#### Actual behavior

Build Fail

So, my question is: 
did anyone build MacOS version recently and what is your setup? (OS version, XCode)

sloretz, 2019-05-20:
Hi @asimonov, the OSX machines on https://ci.ros2.org are currently using macOS Sierra version 10.12 with xcode 9.2 .

The package that is failing to build on your machine `rviz_ogre_vendor` is located here: https://github.com/ros2/rviz/ . If you're getting the same error on all versions then it's likely something else. I don't use OSX myself. For more help you might have better luck with a question on https://answers.ros.org .

Karsten1987, 2019-05-20:
Can you try to build the latest master? Especially having this commit might be crucial:
https://github.com/ros2/rviz/pull/380

asimonov, 2019-05-25:
just tried current master.
see the same error on the same config as in my original message above.

```
--- stderr: rviz_ogre_vendor                                                                                                                       
Undefined symbols for architecture x86_64:
  "Ogre::ZipArchive::ZipArchive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, _zzip_plugin_io*)", referenced from:
      _wrap_new_ZipArchive(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::addEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned char const*, unsigned long, bool (*)(unsigned long, void*, unsigned long))", referenced from:
      _wrap_EmbeddedZipArchiveFactory_addEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::removeEmbbeddedFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      _wrap_EmbeddedZipArchiveFactory_removeEmbbeddedFile(_object*, _object*) in OgrePYTHON_wrap.cxx.o
  "Ogre::EmbeddedZipArchiveFactory::EmbeddedZipArchiveFactory()", referenced from:
      _wrap_new_EmbeddedZipArchiveFactory(_object*, _object*) in OgrePYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/_Ogre.so] Error 1
make[4]: *** [Components/Python/CMakeFiles/_Ogre.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
```

asimonov, 2019-05-26:
ok, I have an update.
I started from fresh homebrew install and followed the instructions (https://index.ros.org/doc/ros2/Installation/Crystal/OSX-Development-Setup/) verbatim.
and I got the errors above.

however, if I add the following into my homebrew:
```
brew tap osrf/simulation
brew install ogre
```
then I can build!

I still see a bunch of linker warning like the following:
```
ld: warning: text-based stub file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/OSServices.framework/Versions/A/OSServices.tbd and library file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/OSServices.framework/Versions/A/OSServices are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/AE.framework/Versions/A/AE.tbd and library file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/AE.framework/Versions/A/AE are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/LaunchServices.framework/Versions/A/LaunchServices.tbd and library file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/LaunchServices.framework/Versions/A/LaunchServices are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/SharedFileList.framework/Versions/A/SharedFileList.tbd and library file /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/SharedFileList.framework/Versions/A/SharedFileList are out of sync. Falling back to library file for linking.
```

they are coming from:
```
Summary: 260 packages finished [37min 52s]
  7 packages had stderr output: qt_gui_cpp rviz2 rviz_common rviz_default_plugins rviz_rendering rviz_rendering_tests rviz_visual_testing_framework
```

but rviz2 works!

vmayoral, 2019-05-26:
Thanks for reporting @asimonov, this fixed another connected issue for me.

Karsten1987, 2019-07-24:
@asimonov Can we close this issue?

asimonov, 2019-07-25:
just checked my records - I did not see any build issues while building ROS2 Dashing on brand new macbook pro with Mojave, right after Dashing was released. so happy to close

closed 2019-07-25 by asimonov

Gatsby23, 2019-08-19:
> just checked my records - I did not see any build issues while building ROS2 Dashing on brand new macbook pro with Mojave, right after Dashing was released. so happy to close

Really ? What is your documentation?  I build it on my mac, but there are a lot of failure

sandhavel, 2021-09-21:
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos

Use this link to install the latest package, to avoid those kind of errors