ros2/ros2 #703: [ROS2 meta-ticket] Fast-RTPS 1.8.0 upgrade

nburek, 2019-05-10:
This issue is to track the upgrade of ROS2 to using Fast-RTPS 1.8.0. We are targeting the Dashing release for this update. We will update this issue with status updates and general timelines as we discover things.

nburek, 2019-05-10:
eProsima has a release candidate ready for 1.8.0 and it currently working on their own release testing (https://github.com/eProsima/Fast-RTPS/pull/518). They are targeting May 14th for the release. 

Meanwhile, we have made updates to rmw_fastrtps to compile and run with the 1.8.0 changes and have opened a PR for that here: https://github.com/ros2/rmw_fastrtps/pull/272. We are currently having issues getting the CI to a full green state. The majority of the problems seem to be related to when the memory testing tools are running. We don't know if this is an actual issue with Fast-RTPS or if the issue is with how the memory test is interacting with Fast-RTPS. It seems like there may also be an issue with some of the tests timing out and just needing a few more seconds added to the timeout in order to be successful. We need to investigate this more to know where the regression is that is causing the additional time.

emersonknapp, 2019-05-13:
Memory tools problem is being tracked here https://github.com/osrf/osrf_testing_tools_cpp/issues/30

For the timing - adding extra time to the tests is not the ideal solution - considering that FastRTPS is the default solution, adding multiple seconds to application startup almost seems unacceptable. Hopefully we can find what's causing it in FastRTPS and solve that.

nburek, 2019-05-14:
After talking with eProsima this morning, they think that the timing regression we are seeing might be related to a bug that they fixed earlier today with a new QoS setting that might have slowed down endpoint discovery. We are working on validating if running with the latest version of their code fixes the issue. 

We have a PR open to fix the issue we were seeing with the failing memory tests. 

The release/1.8.0 branch of FastRTPS has also been merged back into the master branch of Fast-RTPS. They are temporarily holding off tagging this commit as the release in case we need to get a hot fix in for ROS2. 

I will post the results of the retest here after I have them.

MiguelCompany, 2019-05-14:
Hi, I found one discovery timing related issue. I've just open eProsima/Fast-RTPS#531 to fix it.

nburek, 2019-05-14:
Below are the latest test results using the master branch of Fast-RTPS (commit 89c7c7e). I will rerun with the @MiguelCompany 's fix and see if it improves. 

This build is using the repositories list found here: https://gist.github.com/nburek/ebd0b0bbf6fd6c23997c584659555321
This switches out the rmw_fastrtps and osrf_testing_tools_cpp repositories. 
```
build/demo_nodes_cpp/test_results/demo_nodes_cpp/test_tutorial_parameter_events_async__rmw_fastrtps_cpp.xunit.xml: 2 tests, 1 error, 0 failures, 0 skipped
build/demo_nodes_cpp/test_results/demo_nodes_cpp/test_tutorial_parameter_events_async__rmw_fastrtps_dynamic_cpp.xunit.xml: 2 tests, 1 error, 0 failures, 0 skipped
build/launch_testing/pytest.xml: 1 test, 1 error, 0 failures, 0 skipped
build/rclcpp/test_results/rclcpp/test_time_source.gtest.xml: 10 tests, 0 errors, 4 failures, 0 skipped
build/rclcpp_components/test_results/rclcpp_components/test_component_manager_api.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/rclpy/test_results/rclpy/test_action_client.xunit.xml: 16 tests, 0 errors, 2 failures, 0 skipped
build/rosbag2_transport/test_results/rosbag2_transport/test_play.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/rosbag2_transport/test_results/rosbag2_transport/test_record_all.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/rosbag2_transport/test_results/rosbag2_transport/test_rosbag2_node.gtest.xml: 6 tests, 0 errors, 4 failures, 0 skipped
build/test_cli/test_results/test_cli/test_params_yaml.xunit.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/test_cli_remapping/test_results/test_cli_remapping/test_cli_remapping.xunit.xml: 12 tests, 0 errors, 4 failures, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclcpp__rclpy__rmw_fastrtps_cpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclcpp__rclpy__rmw_fastrtps_dynamic_cpp__rmw_fastrtps_cpp.xunit.xml: 24 tests, 0 errors, 2 failures, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclcpp__rmw_fastrtps_cpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclcpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclcpp__rmw_fastrtps_dynamic_cpp__rmw_fastrtps_cpp.xunit.xml: 24 tests, 0 errors, 2 failures, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rclcpp__rmw_fastrtps_cpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rclcpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rclcpp__rmw_fastrtps_dynamic_cpp__rmw_fastrtps_cpp.xunit.xml: 24 tests, 0 errors, 2 failures, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rmw_fastrtps_cpp__rmw_fastrtps_dynamic_cpp.xunit.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber__rclpy__rmw_fastrtps_dynamic_cpp__rmw_fastrtps_cpp.xunit.xml: 24 tests, 0 errors, 2 failures, 0 skipped
build/test_communication/test_results/test_communication/test_publisher_subscriber_cpp__rmw_fastrtps_dynamic_cpp__WStrings.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/test_rclcpp/test_results/test_rclcpp/gtest_multithreaded__rmw_fastrtps_cpp.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/test_rclcpp/test_results/test_rclcpp/test_parameter_server_cpp__rmw_fastrtps_cpp.xunit.xml: 2 tests, 0 errors, 2 failures, 0 skipped
build/tf2_ros/test_results/tf2_ros/tf2_ros_test_message_filter.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped
```

nburek, 2019-05-14:
@thomas-moulard - please run the following CI job to test timeouts with the fix from eProsima and the memory test fix:

 * Gist file: https://gist.github.com/nburek/ebd0b0bbf6fd6c23997c584659555321
 * Additional BUILD flags: --packages-up-to rclcpp test_rclcpp
 * Additional TEST flags: --packages-up-to rclcpp test_rclcpp
* * CI Job: ci_launcher

thomas-moulard, 2019-05-14:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2019-05-14:
See latest round of CI builds: https://github.com/osrf/osrf_testing_tools_cpp/pull/31#issuecomment-492443719

nburek, 2019-05-15:
We've been digging into the test failures from build https://ci.ros2.org/job/ci_linux/7085/ that Dirk posted on https://github.com/eProsima/Fast-RTPS/pull/531. Specifically we've been looking at the failures in rclcpp that are in the test_time_source.cpp test suite. It appears that the failures in these tests are also stemming from a latency regression with using the parameter service. The tests that are failing are using the parameter service to change to using sim_time. It is using the synchronous client, but after the call returns it waits for up to 1 second for ros_time to be updated. If we increase the time it waits from 1 second to 4 seconds by changing this line (https://github.com/ros2/rclcpp/blob/master/rclcpp/test/test_time_source.cpp#L109) then all the tests in rclcpp are passing locally. It's not clear yet what is causing the regression.

emersonknapp, 2019-05-16:
I have a smaller repro case. It's still fairly far from Fast-RTPS but it's a little more revealing than the time_source test. Now, I don't know if it will repro on others' machines, so that would be useful to find out. 

```
#include <chrono>
#include <cinttypes>
#include <memory>
#include "rclcpp/parameter.hpp"
#include "rclcpp/rclcpp.hpp"

void do_test(int argc, char ** argv, bool declare_before_subscribe=true) {
  rclcpp::init(argc, argv);
  static const char * param_name = "timing_test_param";
  rclcpp::Node::SharedPtr node = std::make_shared<rclcpp::Node>("my_node");
  auto start = std::chrono::system_clock::now();

  if (declare_before_subscribe) {
    node->declare_parameter(param_name, rclcpp::ParameterValue(false));
  }
  auto client = std::make_shared<rclcpp::AsyncParametersClient>(node);

  auto sub = client->on_parameter_event([start](const rcl_interfaces::msg::ParameterEvent::SharedPtr event) -> void {
    (void) event;
    auto now = std::chrono::system_clock::now();
    auto diff = now - start;
    RCUTILS_LOG_INFO_NAMED("minimal_param", "Got event %ld ms after set", std::chrono::duration_cast<std::chrono::milliseconds>(diff).count());
  });

  start = std::chrono::system_clock::now();

  if (!declare_before_subscribe) {
    node->declare_parameter(param_name, rclcpp::ParameterValue(false));
  }

  client->set_parameters({
    rclcpp::Parameter(param_name, rclcpp::ParameterValue(true))
  },
  [](std::shared_future<std::vector<rcl_interfaces::msg::SetParametersResult>>) -> void {
    RCUTILS_LOG_INFO_NAMED("minimal_param", "SetParameters callback");
  });

  rclcpp::spin(node);
  rclcpp::shutdown();
}

int main(int argc, char * argv[])
{
  // Setting up subscriber before any publishes, I get 2 "Got event" callbacks at about 80ms
  do_test(argc, argv, false);
  // Setting up subscriber after declaration publish, I get 1 "Got event" callback at about 3000ms
  do_test(argc, argv, true);
  return 0;
}
```

I have uploaded this as the package `examples_rclcpp_minimal_param` on `master` branch here https://github.com/emersonknapp/examples

```
> colcon build --packages-up-to examples_rclcpp_minimal_param ros2run
> source install/local_setup.bash
> ros2 run examples_rclcpp_minimal_param param_main minimal_param __log_level:=INFO
[INFO] [minimal_param]: Running declare _after_ subscribe
[INFO] [minimal_param]: Got event 81 ms after set
[INFO] [minimal_param]: Got event 81 ms after set

<<<Ctrl+C>>>
[INFO] [rclcpp]: signal_handler(signal_value=2)

[INFO] [minimal_param]: Running declare _before_ subscribe
[INFO] [minimal_param]: Got event 3000 ms after set

nburek, 2019-05-16:
I have additional tests that show that if a subscriber is already subscribed before a publisher sends a message that the latency is less than a couple of milliseconds. This holds for ROS service clients as well. 

This 3 second delay seems to line up with the default heartbeat period that Fast-RTPS is using for writers. 
@MiguelCompany Is it possible that new readers that come online after there are already messages published are not receiving any notification about them until after the timer for the heartbeat waits the initial 3 seconds and then sends them the information about existing entries in the history? 

I'll do a bit more digging to see if I can find more evidence around this theory.

emersonknapp, 2019-05-16:
I've updated the branch (see https://github.com/emersonknapp/examples/blob/master/rclcpp/minimal_param/main.cpp#L66) to show an option where I briefly sleep (~100ms) after creating the subscription, before publishing, and then the messages are received very quickly.

nburek, 2019-05-16:
I made a change locally to Fast-RTPS where I set this line (https://github.com/eProsima/Fast-RTPS/blob/89c7c7e61a803015f8e49205ef677204a4f575b1/include/fastrtps/rtps/attributes/WriterAttributes.h#L61) to be 1 second instead of 3 seconds and then I observed that the latency issue was reduced from ~3000ms to ~1000ms. Increasing that heartbeat period to 5 seconds increases the latency using @emersonknapp 's test to ~5000ms. This seems to be a pretty strong indication that it's related to the initial heartbeat.

MiguelCompany, 2019-05-16:
We're looking at this and doing some tests with Fast-RTPS directly. It seems the issue only happens when the publisher and the subscriber are on the same participant. It may be the case that the subscriber receives the initial heartbeat before receiving the discovery information about the publisher, thus discarding the initial heartbeat. We don't reproduce the issue when we create a different participant for each endpoint

MiguelCompany, 2019-05-16:
One more question: are the parameter publishers or subscribers volatile? In that case, new subscribers will not receive old samples

MiguelCompany, 2019-05-16:
As I suspected, the parameter events publisher and subscriber are VOLATILE (see [here](https://github.com/ros2/rmw/blob/dc7b2f49f1f961d6cf2c173adc54736451be8938/rmw/include/rmw/qos_profiles.h#L82)). Will dig more into that use case.

MiguelCompany, 2019-05-16:
It seems we have finally found the issue. See eProsima/Fast-RTPS#532 for details

dirk-thomas, 2019-05-16:
> It seems we have finally found the issue. See eProsima/Fast-RTPS#532 for details

* Linux build testing `rclcpp`: [image]

dirk-thomas, 2019-05-16:
Running a full set of builds to check for regressions:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2019-05-16:
@nburek With the latest fix merged upstream I think this can be closed?

nburek, 2019-05-16:
Yes, I am fine with closing this issue. It looks like the last two failing tests were already known to be a bit flaky (https://github.com/ros2/build_cop/issues/166) and this has just exacerbated the problem so that they fail consistently. We can track fixing those on that issue instead of tracking them as part of the Fast-RTPS 1.8.0 upgrade.

closed 2019-05-16 by dirk-thomas