ros2/ros2 #706: TiXmlElement::SetAttribute function match problems met in building ROS2 Crystal Clemmys Patch Release 4

43970117, 2019-05-20:
TiXmlElement::SetAttribute match problems met in building ROS2 Crystal Clemmys Patch Release 4
**Required Info:**
The failure info:
home/qaz/ros2-CrystalClemmysPR4/src/ros2/urdfdom/urdf_parser/src/link.cpp:696:43: error: no matching function for call to ‘TiXmlElement::SetAttribute(const char [5], std::__cxx11::string&)’
   link_xml->SetAttribute("name", link.name)

/home/qaz/ros2-CrystalClemmysPR4/src/ros2/urdfdom/urdf_parser/src/pose.cpp:127:43: error: no matching function for call to ‘TiXmlElement::SetAttribute(const char [4], std::__cxx11::string&)’
   origin->SetAttribute("xyz", pose_xyz_str);

- Operating System:
  - Ubuntu 16.04
- Installation type:
  -  from source
- Version or commit hash:
  The official ROS2 Crystal Clemmys Patch Release 4. 
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: 0.6.1
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: 0.5.1
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: 0.6.4
  ament/ament_package:
    type: git
    url: https://github.com/ament/ament_package.git
    version: 0.6.0
  ament/googletest:
    type: git
    url: https://github.com/ament/googletest.git
    version: c8b6f2be853002b9a0abfb30ea537402383bf1f4
  ament/osrf_pycommon:
    type: git
    url: https://github.com/osrf/osrf_pycommon.git
    version: 0.1.6
  ament/uncrustify_vendor:
    type: git
    url: https://github.com/ament/uncrustify_vendor.git
    version: 1.1.0
  eProsima/Fast-CDR:
    type: git
    url: https://github.com/eProsima/Fast-CDR.git
    version: v1.0.8
  eProsima/Fast-RTPS:
    type: git
    url: https://github.com/eProsima/Fast-RTPS.git
    version: v1.7.2
  osrf/osrf_testing_tools_cpp:
    type: git
    url: https://github.com/osrf/osrf_testing_tools_cpp.git
    version: 1.1.0
  ros/class_loader:
    type: git
    url: https://github.com/ros/class_loader.git
    version: 1.2.0
  ros/pluginlib:
    type: git
    url: https://github.com/ros/pluginlib.git
    version: 2.2.1
  ros/resource_retriever:
    type: git
    url: https://github.com/ros/resource_retriever.git
    version: 2.1.0
  ros/ros_environment:
    type: git
[... 253 lines cut ...]
    version: 5.0.0

- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
 The followed wiki: https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/
```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2019-05-21:
I just tried following the instructions in a fresh Xenial Docker container and the build of `urdfdom` passed just fine from. Please double check you environment and the installation steps and provide more information if the problem persists.

43970117, 2019-06-17:
The reason is that my build environment has tinyxml/tinyxml2 source code.

closed 2019-06-17 by 43970117