ros2/ros2 #708: Can't load rmw_fastrtps_cpp on Windows

hidmic, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From fat binary archive
- Version or commit hash:
  - [Dashing pre-release](https://ci.ros2.org/view/packaging/job/packaging_windows/1380/)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
After [installation](https://github.com/ros2/ros2_documentation/blob/611f3aed7659e4689f5121fb098178100ae15dc6/source/Installation/Dashing/Windows-Install-Binary.rst), just attempt to:

```
ros2 topic list
```

#### Expected behavior

No output at all (assuming no node is running anywhere).

#### Actual behavior

```
[ERROR] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137', exiting with 1.
```

#### Additional information

`rmw_fastrtps_cpp.dll` can be nonetheless found in the archive and it's in the `Path`.

hidmic, 2019-05-24:
So, Fast-RTPS was bringing in `msvcr20.dll`, which is part of the Visual C++ 2013 redistributable packages. Apparently, said library is present on regular Windows 10 installation but it was not on my VM (which has the Windows Server). After installing them, everything went back to normal.

I'll close this as we don't officially support Windows Server atm.

closed 2019-05-24 by hidmic

hidmic, 2019-05-24:
After a quick discussion with @nuclearsandwich, we think it may be worth to eventually leave a note in the Troubleshooting section of the Windows Install instructions for Dashing in case someone runs into this issue again.

dzywater, 2019-12-25:
I installed Windows 10 , but had the same problem.

[ERROR] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137', exiting with 1.

Edition: Windows 10 Enterprise
Version: 1909
OS Build: 18363.476

nuclearsandwich, 2019-12-26:
@dzywater do you have vcredist2013 installed via Chocolatey or manually?