ros2/ros2 #710: Could not find a package configuration file provided by "ament_cmake_core" or any other ament files

yueyeah, 2019-05-27:
**Required Info:
- Operating System: Ubuntu 18.04
- Installation type: building from source, following the turtlebot3 ros2 tutorial here: http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#simulation 
- Version or commit hash: master
- DDS implementation: N/A

#### Steps to reproduce issue
I followed the steps on the TurtleBot3 manual on Robotis (http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#simulation) (this is the same link as above), except for the script to install Gazebo because I already have Gazebo at the time of installation.

```

#### Expected behavior
Successful build of turtlebot3 packages.

#### Actual behavior
This is the section where I couldn't build the packages.
`$ mkdir -p ~/turtlebot3_ws/src
$ cd ~/turtlebot3_ws
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
$ vcs import src < turtlebot3.repos
$ colcon build --symlink-install
`
After colcon build, the following errors came out.
`Starting >>> nav2_common
Starting >>> behaviortree_cpp
Starting >>> nav_2d_msgs
Starting >>> angles
--- stderr: nav2_common
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findament_cmake_core.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_core", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_core"
  with any of the following names:

    ament_cmake_coreConfig.cmake
    ament_cmake_core-config.cmake

  Add the installation prefix of "ament_cmake_core" to CMAKE_PREFIX_PATH or
  set "ament_cmake_core_DIR" to a directory containing one of the above
  files.  If "ament_cmake_core" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< nav2_common	[ Exited with code 1 ]
--- stderr: angles                               
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Aborted  <<< angles
--- stderr: nav_2d_msgs
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findgeometry_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "geometry_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "geometry_msgs"
  with any of the following names:

    geometry_msgsConfig.cmake
    geometry_msgs-config.cmake

  Add the installation prefix of "geometry_msgs" to CMAKE_PREFIX_PATH or set
  "geometry_msgs_DIR" to a directory containing one of the above files.  If
  "geometry_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Aborted  <<< nav_2d_msgs
--- stderr: behaviortree_cpp                                     
CMake Warning at CMakeLists.txt:79 (message):
   GTest not found!

CMake Warning:
  Manually-specified variables were not used by the project:

    AMENT_CMAKE_SYMLINK_INSTALL

---
Aborted  <<< behaviortree_cpp

Summary: 0 packages finished [2.15s]
  1 package failed: nav2_common
  3 packages aborted: angles behaviortree_cpp nav_2d_msgs
  4 packages had stderr output: angles behaviortree_cpp nav2_common nav_2d_msgs
  51 packages not processed`

Since the error mentioned that I was missing some ament_cmake_core, I decided to copy an ament directory over from an existing ros2_ws build that I had (this ros2 build was built from source successfully after following the ros2 tutorial here: https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Development-Setup/ ). 

The earlier error on ament_cmake_core was averted, but a new error appeared: 
`--- stderr: turtlebot3_description                                        
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findurdf.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "urdf", but
  CMake did not find one.

  Could not find a package configuration file provided by "urdf" with any of
  the following names:

    urdfConfig.cmake
    urdf-config.cmake

  Add the installation prefix of "urdf" to CMAKE_PREFIX_PATH or set
  "urdf_DIR" to a directory containing one of the above files.  If "urdf"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< turtlebot3_description	[ Exited with code 1 ]
Aborted  <<< uncrustify_vendor                      
Aborted  <<< behaviortree_cpp                           
--- stderr: cartographer                                                        
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/2d/submap_2d.cc.o] Interrupt
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/3d/range_data_inserter_3d.cc.o] Interrupt
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/3d/submap_3d.cc.o] Interrupt
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/detect_floors.cc.o] Interrupt
make[1]: *** [CMakeFiles/cartographer.dir/all] Interrupt
make: *** [all] Interrupt
---
Aborted  <<< cartographer
                                         
Summary: 37 packages finished [1min 11s]
  1 package failed: turtlebot3_description
  3 packages aborted: behaviortree_cpp cartographer uncrustify_vendor
  2 packages had stderr output: cartographer turtlebot3_description
  65 packages not processed
`

I realised that this ament folder required the other folders in the ros2_ws build as well, so I copied all of them over to turtlebot3_ws. (The folders are: osrf, ros2, ros-planning, eProsima, ros, ros-perception, ros-visualization).
Then I built them using colcon build --symlink-install and the following error occurred: 
`--- stderr: turtlebot3_description            
CMake Error at CMakeLists.txt:52 (find_package):
  By not providing "Findament_lint_auto.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_lint_auto", but CMake did not find one.

  Could not find a package configuration file provided by "ament_lint_auto"
  with any of the following names:

    ament_lint_autoConfig.cmake
    ament_lint_auto-config.cmake

  Add the installation prefix of "ament_lint_auto" to CMAKE_PREFIX_PATH or
  set "ament_lint_auto_DIR" to a directory containing one of the above files.
  If "ament_lint_auto" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< turtlebot3_description	[ Exited with code 1 ]
Aborted  <<< fastrtps                                    
--- stderr: cartographer                                   
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping/internal/3d/scan_matching/real_time_correlative_scan_matcher_3d.cc.o] Interrupt
make[1]: *** [CMakeFiles/cartographer.dir/all] Interrupt
make: *** [all] Interrupt
---
Aborted  <<< cartographer
--- stderr: rviz_ogre_vendor
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreGpuProgramParams.cpp.o] Interrupt
make[4]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Interrupt
make[3]: *** [all] Interrupt
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Interrupt
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Interrupt
make: *** [all] Interrupt
---
Aborted  <<< rviz_ogre_vendor

Summary: 96 packages finished [4min 33s]
  1 package failed: turtlebot3_description
  3 packages aborted: cartographer fastrtps rviz_ogre_vendor
  7 packages had stderr output: cartographer rmw_connext_cpp rmw_connext_shared_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rviz_ogre_vendor turtlebot3_description
  186 packages not processed
`
This is despite the ament_lint_auto package existing in the ament folder and already built before arriving at the turtlebot3_description package.

The latter two steps of troubleshooting may not be correct, but may I know how else do you prevent the first error of not finding the ament_cmake_core package configuration file? 

Thank you very much for reading all of this.

dirk-thomas, 2019-05-28:
It looks like you haven't source ROS 2 before trying to build your workspace and therefore none of the ROS 2 packages can be found. Assuming you are using the Debian packages you need to `source /opt/ros/<distro>/setup.bash` (or `.sh` or any other extension).

If this step is missing in the TurtleBot3 tutorial you might want to report this somewhere with the TurtleBot3 maintainers.

closed 2019-05-28 by dirk-thomas

nikboy111, 2022-01-23:
Hello there, 

I'm having the same problem like @yueyeah while trying to install the TurtleBot3 packages following the same manual yueyeah did. 

`CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findaction_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "action_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "action_msgs" with
  any of the following names:

    action_msgsConfig.cmake
    action_msgs-config.cmake

  Add the installation prefix of "action_msgs" to CMAKE_PREFIX_PATH or set
  "action_msgs_DIR" to a directory containing one of the above files.  If
  "action_msgs" provides a separate development package or SDK, be sure it
  has been installed.
`
I'm using ROS2 Foxy on Ubuntu 20.04 and even though I source the /opt/ros/foxy/setup.bash as well as the setup.sh I'm getting this error. 
I spent hours searching for a troubleshoot that works for me, but everyone is just suggesting source ROS2. 
Is there anything else I might have forgotten before building the turtlebot workspace? 

Thank you so much for your support!

Nik

clalancette, 2022-01-23:
Please open a question at https://answers.ros.org.  You'll get better answers there than on a closed issue.

nikboy111, 2022-01-23:
> Please open a question at https://answers.ros.org. You'll get better answers there than on a closed issue.

Thank you, i did [here](https://github.com/ros2/ros2/issues/1236).

mandulaj, 2023-10-26:
Also make sure that you have the package installed:
```bash
sudo apt install ros-${ROS_DISTRO}-ament-cmake-clang-format
```