ros2/ros2 #711: best-effort reliability not working in crystal

realdream, 2019-05-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  -  release crystal e0e4b6b4
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp
- Network Env
  - local loopback only
  - wired LAN
  - WiFi

#### Steps to reproduce issue

```
ros2 run image_tools cam2image -b -x 640 -y 480 -f 30 -r 0
ros2 run image_tools showimage -r 0
```
never get message
```
ros2 run image_tools cam2image -b -x 640 -y 480 -f 30 -r 0
ros2 run image_tools showimage -r 1
```
never get message (that's ok it shouldn't work)
```
ros2 run image_tools cam2image -b -x 640 -y 480 -f 30 -r 1
ros2 run image_tools showimage -r 0
```
sometime get message but freeze after a while
```
ros2 run image_tools cam2image -b -x 640 -y 480 -f 30 -r 1
ros2 run image_tools showimage -r 1
```
the last one can get image stably

according to :

| Combination | Publisher | Subscriber | Connection | Result |
| :----: | :----:  | :----: | :----:  | :----: |
| 1 | Best effort | Best effort | Yes | Best effort |
| 2 | Best effort | Reliable | No | x |
| 3 | Reliable | Best effort | Yes | Best effort |
| 4 | Reliable | Reliable | Yes | Reliable |

the combination 1/3/4 should work
but the result is only the last one Reliable2Reliable can work stably in crystal release(e0e4b6b4).
meanwhile, the bouncy release has no such issue.

realdream, 2019-05-30:
if I lower the image size to 128x128 everything works fine
the binary release also have the same behavior

realdream, 2019-05-30:
find some new clue

#### Steps to reproduce issue
terminal1
```
ros2 run image_tools cam2image -b -x 256 -y 256 -f 30 -r 1

```

terminal2
```
ros2 run image_tools showimage -r 0
```
with no image received
then use ctrl-C to stop cam2image
and restart with
```
ros2 run image_tools cam2image -b -x 64 -y 64 -f 30 -r 1
```
then you get error msg from showimage like this
```
2019-05-30 10:47:18.121 [RTPS_MSG_IN Error] (ID:140339708937984) Problem reserving CacheChange in reader: 0.0.1c.4 -> Function processDataMsg
```
2019-05-30 10:47:18.088 [RTPS_MSG_IN Error] (ID:140339708937984) Problem reserving CacheChange in reader: 0.0.1c.4 -> Function processDataMsg
2019-05-30 10:47:18.121 [RTPS_MSG_IN Error] (ID:140339708937984) Problem reserving CacheChange in reader: 0.0.1c.4 -> Function processDataMsg

clalancette, 2019-05-30:
For what it is worth, all of your above examples (with the exception of best effort -> reliable) work fine for me with the Crystal binaries.  Let's focus on the binary installation to reduce potential environmental effects.  Can you run `dpkg -l | grep fastrtps` and post the output here?  I just want to make sure the packages are up-to-date.

realdream, 2019-05-31:
I'm using the crystal binaries from github release [ros2-crystal-20190408-linux-xenial-amd64.tar.bz2](https://github.com/ros2/ros2/releases/download/release-crystal-20190408/ros2-crystal-20190408-linux-xenial-amd64.tar.bz2) 
(source code commitid e0e4b6b) and the fast-rtps version is 1.7.2

if I only change fast-rtps to 1.7.0 (same as previous crystal release before patch4 )and keep everything else same as patch4, it works fine.
I will try binary installation later

realdream, 2019-05-31:
I have just try the binary installation on a fresh-installed ubuntu 18.04
and I got the similar result:  if image size is large than 300x300 best-effort not working, the smaller the image is the more likely it works.
the fast-rtps version you required is followed
```
dpkg -l | grep fastrtps
ii  ros-crystal-fastrtps                               1.7.2-0bionic.20190327.141923      amd64        Implementation of RTPS standard.
ii  ros-crystal-fastrtps-cmake-module                  0.6.1-0bionic.20190310.051033      amd64        Provide CMake module to find eProsima FastRTPS.
ii  ros-crystal-rmw-fastrtps-cpp                       0.6.2-0bionic.20190402.194841      amd64        Implement the ROS middleware interface using eProsima FastRTPS static code generation in C++.
ii  ros-crystal-rmw-fastrtps-shared-cpp                0.6.2-0bionic.20190402.194652      amd64        Code shared on static and dynamic type support of rmw_fastrtps_cpp.
ii  ros-crystal-rosidl-typesupport-fastrtps-c          0.6.1-0bionic.20190402.194631      amd64        Generate the C interfaces for eProsima FastRTPS.
ii  ros-crystal-rosidl-typesupport-fastrtps-cpp        0.6.1-0bionic.20190402.194358      amd64        Generate the C++ interfaces for eProsima FastRTPS.

```

realdream, 2019-05-31:
add my new test command here
Steps to reproduce issue
terminal1
```
ros2 run image_tools cam2image -b -x 300 -y 300 -r 1
```
terminal2
```
ros2 run image_tools showimage -r 0
```
with no image received
then use ctrl-C to stop cam2image
and restart with
```
ros2 run image_tools cam2image -b -x 64 -y 64 -r 1
```
then you get error msg from showimage like this
```
2019-05-31 10:46:24.755 [RTPS_MSG_IN Error] (ID:140396922513152) Problem reserving CacheChange in reader: 0.0.1c.4 -> Function processDataMsg
```

jacobperron, 2019-06-04:
I can reproduce this with Crystal debs on Ubuntu 18.04.

Through some experimentation, it would appear that I am unable to receive images with a size greater than 65025. For example, the following works,

```sh
ros2 run image_tools cam2image -b -x 64 -y 1016 -f 30 -r 0
```

But not a slightly larger image:

```sh
ros2 run image_tools cam2image -b -x 65 -y 1016 -f 30 -r 0
```

jacobperron, 2019-06-04:
The problem happens when I build Crystal from source as well, though a little different. Eventually the images start being received, but there is a noticeable delay (5-10s) before it starts working.

I've tried with Connext and Opensplice without problems. Also no problems with any RMW implementation in Dashing. This leads me to believe that it is an issue with the version of Fast-RTPS shipped with Crystal.

clalancette, 2019-06-05:
I guess the reason I'm still confused is that this all seems to work for me on Crystal.  Additionally, I know one of my clients does pub/sub of 1280x720 images on Crystal, and that seems to work as well.  There is some possibility that the delay is there; I haven't checked closely.  But I'm just not sure why it works sometimes and not others.

jacobperron, 2019-06-05:
I'll note that I reproduced in a Docker container, I'll try without a container and report back.

jacobperron, 2019-06-05:
I encounter the same issue on my non-containerized Ubuntu 18.04.

condor21, 2020-06-15:
In ROS2 Eloquent I have run into the same issue. May I ask if this problem is not resolved yet?

fujitatomoya, 2020-06-15:
@condor21 

could you try with foxy release? I believe there are many improvement including shared memory feature in Fast-DDS(Fast-RTPS).

clalancette, 2021-06-10:
I did some basic testing with Galactic here, and things seem to work fine for me, both with CycloneDDS and Fast-DDS.  Certainly there may be things to improve here, but given that:

1. This case seems to work now, 
2. The age of the initial issue
3. There's been no response for a year

I'm going to close this out.  If you are still running into problems with Foxy or Galactic, please feel free to open another issue.

closed 2021-06-10 by clalancette