ros2/ros2 #712: Calling `setup.bat` of an overlay workspace doesn't bring RTI connext license

ivanpauno, 2019-05-28:
## Bug report

**Required Info:**

- Operating System:
  - Windows server 2019 Datacenter
- Installation type:
  - from source
- Version or commit hash:
  - current master
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
A ros2 workspace built with RTI connext is needed.
In a VS command prompt:
- Source the ros2 workspace.
- Create a new empty workspace.
- Build it.

In another command prompt:
- Source the new workspace `setup.bat`.
- Run:

```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run demo_nodes_cpp talker
```
or

```bat
set RMW_IMPLEMENTATION=rmw_connext_cpp
ros2 run demo_nodes_py talker
```

#### Expected behavior
Both commands should run without an error.

#### Actual behavior
I get this output for demo_nodes_cpp talker:
```
RTI Data Distribution Service No source for License information
Please contact support@rti.com with any questions or comments.
DDSDomainParticipant_impl::create_disabledI:!create participant
DDSDomainParticipant_impl::createI:!create participant
DomainParticipantFactory_impl::create_participant():!create failure creating participant

>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:

  'failed to create participant, at C:\dev\ros2\src\ros2\rmw_connext\rmw_connext_shared_cpp\src\node.cpp:269, at C:\dev\ros2\src\ros2\rcl\rcl\src\rcl\node.c:326'

with this new error message:

  'rcl node's rmw handle is invalid, at C:\dev\ros2\src\ros2\rcl\rcl\src\rcl\node.c:464'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [rcl]: Failed to fini publisher for node: 1
```

Similar problem with demo_nodes_py talker.

#### Additional information

If after sourcing the `setup.bat` in the new workspace I do:
```bat
call "C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\rticommon.bat"
```
The problems disappears.

I tried with both fastrtps and opensplice, and a similar problem doesn't happen.
For some reason, the overlay workspace is not setting the connext license correctly (maybe something it's not being sourced??).

Only tested on windows, no idea if `setup.bash` has a similar problem on linux or not.

dirk-thomas, 2019-05-28:
Can you please check what the following line was expanded to in your install space: https://github.com/ros2/rosidl_typesupport_connext/blob/51ee8205380d06a56b63ae69c471db79f4ef3216/connext_cmake_module/env_hook/connext.bat.in#L29

dirk-thomas, 2019-05-28:
Nvm. the expansion needs to be VS2017 even if we build with 2019...

dirk-thomas, 2019-05-28:
Please check which files are being sourced when sourcing the workspace level setup file with the following env var being set: `AMENT_TRACE_SETUP_FILES=1`. Please post the files related to the `connext_cmake_module` package.

ivanpauno, 2019-05-28:
In the `setup.bat` of the new workspace I'm seeing the following:

```
:: source chained prefixes
call:_colcon_prefix_chain_bat_call_script "c:\dev\ros2\install\local_setup.bat"
call:_colcon_prefix_chain_bat_call_script "c:\dev\ros2\install\\local_setup.bat"
```
Is that normal?

This three files appear:

```
call "c:\dev\ros2\install\share\connext_cmake_module\environment\ament_prefix_path.bat"
call "c:\dev\ros2\install\share\connext_cmake_module\environment\connext.bat"
call "c:\dev\ros2\install\share\connext_cmake_module\environment\path.bat"
```

dirk-thomas, 2019-05-28:
> :: source chained prefixes

The duplication seems odd. Maybe you have duplicate entries in your environment variables: `env | grep _PREFIX_PATH` (to-be-converted to Windows cmd).

> `call "c:\dev\ros2\install\share\connext_cmake_module\environment\connext.bat"`

Since it sources the `connext.bat` file it should source the script provided by RTO for you: https://github.com/ros2/rosidl_typesupport_connext/blob/51ee8205380d06a56b63ae69c471db79f4ef3216/connext_cmake_module/env_hook/connext.bat.in#L39

If it still doesn't please debug what goes wrong in the `connext.bat` file in your case.

ivanpauno, 2019-05-28:
> If it still doesn't please debug what goes wrong in the `connext.bat` file in your case.

I have debugged. Actually, the `connext.bat` file is calling the `C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\rtisetenv_x64Win64VS2017.bat` script correctly.

But I still have the same error when running the talker demo. Actually, if i do:
```
call C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\rtisetenv_x64Win64VS2017.bat
```
In that terminal and try again, it doesn't work.
I have to source `rticommon.bat` and not `rtisetenv_x64Win64VS2017.bat` for making it work.
```
call C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\rticommon.bat
```

Don't know why it works in the original workspace and not in the overlay.

> The duplication seems odd. Maybe you have duplicate entries in your environment variables: env | grep _PREFIX_PATH (to-be-converted to Windows cmd).

I have checked, and the environment variables aren't duplicated.

dirk-thomas, 2019-05-29:
Just to clarify: can you run the talker and listener from your underlay workspace? If yes, is the environment variable for the RTI licence file set in that case? If yes, where was it set - manually or somewhere by the sourced `setup` file?

ivanpauno, 2019-05-29:
> Just to clarify: can you run the talker and listener from your underlay workspace?

Yes

> If yes, is the environment variable for the RTI licence file set in that case?

This are the connext related variables in my enviroment:
```
installDir=C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\..\..
scriptDir=C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\
scriptName=rtisetenv_x64Win64VS2017
hostVersion=5.3.1
NDDSHOME=C:\Program Files\rti_connext_dds-5.3.1
architectureName=x64Win64VS2017
```

> If yes, where was it set - manually or somewhere by the sourced `setup` file?

Somewhere sourced from `setup` file.

----

In the overlay workspace, the connext related env variables are:
```
installDir=C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\..\..
scriptDir=C:\Program Files\rti_connext_dds-5.3.1\resource\scripts\
scriptName=rtisetenv_x64Win64VS2017
hostVersion=5.3.1
NDDSHOME=C:\Program Files\rti_connext_dds-5.3.1
architectureName=x64Win64VS2017
```

Which looks the same! But it's still not working in the overlay, although in the underlay works well.

jacobperron, 2019-05-29:
I am unable to reproduce this on my Windows laptop.

ivanpauno, 2019-05-30:
> I am unable to reproduce this on my Windows laptop.

Thank you for trying it too.
I have already dug into this, and I have checked that the `*.bat` scripts which are supposed to setup connext license are being called. I have also checked the related environment variables, and they look good.

I don't know what more to check. If no one can reproduce it, I will proceed to close the issue.

closed 2019-05-30 by dirk-thomas

ivanpauno, 2019-05-31:
I finally found something else.
When I installed the last fat archives ([here](https://ci.ros2.org/job/packaging_windows/1388/)), I had the same problem when calling the `setup.bat`, without creating an overlay workspace!

After trying a little bit, I found how to solve the problem. When choosing the license, Connext gives you two options:
- `Copy to RTI install location (all users)` (default option).
- `Copy to RTI user home (just for user Administrator)` (the option I chose the first time).

When choosing the second option, calling the `setup.bat` of the binaries install or calling the `setup.bat` of an overlay workspace of the from source install doesn't work. And it works well when calling the `setup.bat` of the from source workspace.
I have never seen the problem when choosing the first option for installing the license.

I have manually deleted and reinstall the connext license a few times, and I always get the same behavior in each of the options.
My windows setup only have one `Administrator` user.

I will add a troubleshooting section to the windows install instructions, recommending using the first option and not the second one. This seems to be a problem in the connext launcher itself, it's strange that it works well with two of the three cases I mentioned before though.