ros2/ros2 #713: Installation instructions point to a wrong repo GPG key

artizirk, 2019-05-30:
## Bug report

**Required Info:**

- Operating System:
  - Debian 9
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS 2 Crystal

#### Steps to reproduce issue

Follow instructions on https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Debians/#setup-sources
```
sudo apt update && sudo apt install curl gnupg2 lsb-release
curl http://repo.ros2.org/repos.key | sudo apt-key add -

sudo sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'

export CHOOSE_ROS_DISTRO=crystal  # or bouncy or ardent
sudo apt update
```

#### Expected behavior

`apt update` should validate the repo

#### Actual behavior

Repo validation and package installs fail

#### Additional information
It looks like the gpg key file http://repo.ros2.org/repos.key has been changed recently from dsa1024 `421C365BD9FF1F717815A3895523BAEEB01FA116` `ROS Builder <rosbuild@ros.org>` to a new key rsa4096 `C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654` `Open Robotics <info@osrfoundation.org>` but apt repo Release file has not been re signed.

It would be nice if both old and the new key were to be present on the key file

gerkey, 2019-05-30:
We're in the process of rebuilding and resigning with a new key because of a security issue on the build farm: https://discourse.ros.org/t/security-issue-on-ros-build-farm/9342. Please bear with us while we bring things back online.

closed 2019-05-31 by dirk-thomas

nuclearsandwich, 2019-05-31:
We've rotated the keys used for ROS 2 as detailed here: https://discourse.ros.org/t/key-rotation-for-ros-2-apt-repositories/9363