ros2/ros2 #714: Exception sending message over network

glmnet, 2019-05-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 
- DDS implementation:
  - fast rtps? I'm not sure of having chosen one
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I'm publishing odometry data from a robot, it is a differential drive one, using mostly turtlebot3 as example, many messages get published then randomly after some seconds between 5 to more than 200 may be it will throw an exception.

#### Expected behavior
No exceptions!

#### Actual behavior

The communication is throwing an exception

terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/rosbert/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/rosbert/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:257

#### Additional information
````
48:  rmw_fastrtps_shared_cpp::SerializedData data;
49:  data.is_cdr_buffer = false;
50:  data.data = const_cast<void *>(ros_message);
51:  if (!info->publisher_->write(&data)) {
52:    RMW_SET_ERROR_MSG("cannot publish data");
53:    return RMW_RET_ERROR;
54:  }

....

    rcl_ret_t
    rcl_publish(
      const rcl_publisher_t * publisher,
      const void * ros_message,
      rmw_publisher_allocation_t * allocation)
    {
      if (!rcl_publisher_is_valid(publisher)) {
        return RCL_RET_PUBLISHER_INVALID;  // error already set
      }
      RCL_CHECK_ARGUMENT_FOR_NULL(ros_message, RCL_RET_INVALID_ARGUMENT);
      if (rmw_publish(publisher->impl->rmw_handle, ros_message, allocation) != RMW_RET_OK) {
257:    RCL_SET_ERROR_MSG(rmw_get_error_string().str);
        return RCL_RET_ERROR;
      }
      return RCL_RET_OK;
    }

````

sloretz, 2019-05-30:
Thanks for posting where the exception occurred. Would you mind posting the code you wrote that calls publish?

realdream, 2019-06-20:
I have try on arm64 or amd64 ,from bin or source, release-dashing or release-dashing-patch1 and using fastrtps as dds, all the combination get same result

This can also be reproduced by following cmd
```
ros2 run image_tools cam2image -b 
ros2 run image_tools showimage
```
however you may need to wait for hours to reproduce the crashing,
or you can also use higher resulation like 2000x9000, and you can reproduce it within one minute

```
ros2 run image_tools cam2image -b -x 2000 -y 9000 
ros2 run image_tools showimage
```
Morever, this is not a new issue:
https://github.com/ros2/ros2/issues/539

glmnet, 2019-06-20:
I can add some information. This exception happens more often when data is going both ways:
I have two laser scan sending data to rviz in a remote machine, then a joystick in this remote machine sending cmd_vel commands to the robot. When I start using the joystick the exception happens more often. Also odom data is now sent from robot to remote rviz.
I am now catching the exceptions on the publish call and it is running kind of stable.
The try/catch are installed in publish odom and I had to put them in publish laser scan too (different process)
It is happening with dashing release, I am running from source again for other reasons.

ich-will, 2019-07-10:
I have the same issue. 

Dashing, rclcpp (binary installation)
DDS: FastRTPS **(With OpenSplice or Connext the issue does not appears!!)** , 
Ethernet, Ubuntu 18.04 bionic (in docker)
Linux 4.15.0-51-generic #55-Ubuntu SMP Wed May 15 14:27:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

host1: single node/publisher, interval: 10ms, message content:
_int32 cycle_number_
_builtin_interfaces/Time timestamp_

host2: single node/subscriber

**host1 app crashes after ~500 sec with exception:**
_terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl/src/rcl/publisher.c:257
Aborted_

Publisher's source code:
```
#include <rclcpp/rclcpp.hpp>
#include <bzx68_msgs/msg/test_message.hpp>
using namespace rclcpp;
class bzx68_sender_node: public rclcpp::Node{
public:
    bzx68_sender_node(const std::string& topic_name, unsigned int interval_ms): rclcpp::Node("bzx68_sender"),counter(0){
        pub = create_publisher<bzx68_msgs::msg::TestMessage>(topic_name, QoS(KeepLast(10)));
        pub_timer = create_wall_timer(std::chrono::milliseconds(interval_ms), 
                    std::bind(&bzx68_sender_node::send_data_callback,this));
        local_clock = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
        std::cout<<"bzx68_sender: start to publish data to ["<<topic_name<<"] each "<<interval_ms<<"ms"<<std::endl;
    };
    ~bzx68_sender_node(){
    };
    void send_data_callback(){
        bzx68_msgs::msg::TestMessage msg;
        msg.timestamp = local_clock->now();
        msg.cycle_number = counter;        
        pub->publish(msg);
        counter++;
    };
private:
    unsigned int counter;
    rclcpp::Publisher<bzx68_msgs::msg::TestMessage>::SharedPtr pub;    
    rclcpp::TimerBase::SharedPtr pub_timer;
    rclcpp::Clock::SharedPtr local_clock;
};

int main(int argc,char** argv){
    rclcpp::init(argc, argv);    
    auto node = std::make_shared<bzx68_sender_node>("test_topic",10);
    rclcpp::spin(node);
}

```

Subscriber's code is mostly the same
```
......
        sub = create_subscription<bzx68_msgs::msg::TestMessage>("test_topic", QoS(KeepLast(10)),
                    std::bind(&bzx68_receiver_node::receive_data_callback,this,std::placeholders::_1));
......
```

Jconn, 2019-07-30:
This issue reproduces 100% for me when I launch gazebo, then launch the nav2 stack, then kill the nav2 stack, and then launch the nav2 stack again. Gazebo will faithfully terminate at this point with the following error message:

`[gazebo-1]   what():  failed to publish message: cannot publish data, at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.4/src/rmw_publish.cpp:52, at /tmp/binarydeb/ros-dashing-rcl-0.7.5/src/rcl/publisher.c:257`

ivanpauno, 2019-07-31:
> ```
> ros2 run image_tools cam2image -b -x 2000 -y 9000 
> ros2 run image_tools showimage
> ```

This is an interesting example. In my computer (Ubuntu 18.04, from source, master):
- The demo works perfectly using `opensplice`.
- It crashes with `fastrtps` after some messages (same error message as the others).
- It works really slowly in `connext`, but it doesn't crashes. Actually, is the only implementation that loses messages (which is acceptable in a best effort subscriber).

@richiware Could you look at this? It seems to be the same that the old https://github.com/ros2/ros2/issues/539.

SteveMacenski, 2019-08-06:
I've also seen it occur with the robot_state_publisher when up for very long periods of time with the navigation2 stack, I didn't notice any particular thing I've done as the trigger

maxlein, 2019-08-14:
Same problem for me when using ros1_bridge in a docker container, talking with a ros2 host and starting/stopping the container.

richiware, 2019-08-28:
I was able to reproduce the issue using your commands:

```bash
ros2 run image_tools cam2image -b -x 2000 -y 9000 
ros2 run image_tools showimage
```
The publication fails because it exceeds the `max_blocking_time` QoS (100 milliseconds by default). We will investigate why it is blocked so much time. Thanks for your help.

clalancette, 2019-09-10:
I can reproduce this in a slightly different way by forcing a crash on a subscriber enough times, which eventually causes the publisher to crash.  Compile https://github.com/clalancette/ros2-fake-imu-publisher and https://github.com/clalancette/ros2-subscriber-crasher under Dashing, then run:

Terminal 1:
```
ros2 run fake_imu fake_imu_node
```

Terminal 2:
```
while true ; do ros2 run crasher crasher_node ; done
```

Eventually the fake_imu_node will crash (about 1-2 minutes).

ivanpauno, 2019-09-13:
I'm transferring the issue to https://github.com/ros2/rmw_fastrtps, if someone doesn't like the idea I can transfer it back here.