ros2/ros2 #718: ROS 2 Dashing build from source error in OS X (rosidl_generator_c)

vmayoral, 2019-06-02:
## Bug report

**Required Info:**

- Operating System:
  - OS X 10.14.3 Mojave
- Installation type:
  - From source
- Version or commit hash:
  - latest-repos (at the time of writing)
- DDS implementation:
  - default, Fast-RTPS

#### Steps to reproduce issue
```bash
| victor:~/ros2_ws$
| => colcon build --merge-install
Starting >>> ament_package
Starting >>> ament_flake8
Starting >>> osrf_testing_tools_cpp
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> ament_cppcheck
...
Finished <<< rosidl_cmake [2.67s]
Starting >>> rosidl_generator_c
Finished <<< python_qt_binding [7.18s]
Starting >>> rosidl_generator_dds_idl
--- stderr: rosidl_generator_c
'Token' object has no attribute 'children' /Users/victor/ros2_ws/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c/msg/BoundedArrayPrimitives.idl
Error processing idl file: /Users/victor/ros2_ws/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c/msg/BoundedArrayPrimitives.idl
Traceback (most recent call last):
  File "/Users/victor/ros2_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 40, in <module>
    sys.exit(main())
  File "/Users/victor/ros2_ws/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 35, in main
    args.generator_arguments_file,
  File "/Users/victor/ros2_ws/src/ros2/rosidl/rosidl_generator_c/rosidl_generator_c/__init__.py", line 37, in generate_c
    generate_files(generator_arguments_file, mapping)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_cmake/__init__.py", line 90, in generate_files
    raise(e)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_cmake/__init__.py", line 70, in generate_files
    idl_file = parse_idl_file(locator)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 64, in parse_idl_file
    content = parse_idl_string(string, png_file=png_file)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 73, in parse_idl_string
    content = extract_content_from_ast(tree)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 144, in extract_content_from_ast
    add_message_members(msg, struct_defs[0])
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 365, in add_message_members
    abstract_type = get_abstract_type_from_type_spec(type_spec)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 406, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 439, in get_abstract_type
    basetype = get_abstract_type_from_type_spec(type_spec)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 406, in get_abstract_type_from_type_spec
    return get_abstract_type(child)
  File "/Users/victor/ros2_ws/install/lib/python3.7/site-packages/rosidl_parser/parser.py", line 415, in get_abstract_type
    while len(child.children) == 1:
AttributeError: 'Token' object has no attribute 'children'
make[2]: *** [rosidl_generator_c/rosidl_generator_c/msg/bool.h] Error 1
make[2]: *** Deleting file `rosidl_generator_c/rosidl_generator_c/msg/bool.h'
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
```

dirk-thomas, 2019-06-02:
Please check the version of the Python package `lark-parser`. It needs to be 0.7+.

vmayoral, 2019-06-03:
Right, thanks @dirk-thomas. That did it.
Apparently I wasn't installing the last version as per the instructions 
```bash
python3 -m pip install lark-parser
Requirement already satisfied: lark-parser in /usr/local/lib/python3.7/site-packages (0.6.5)
```
but
```bash
python3 -m pip install lark-parser==0.7.1
Requirement already satisfied: lark-parser==0.7.1 in /usr/local/lib/python3.7/site-packages (0.7.1)
```

I've sent a patch https://github.com/ros2/ros2_documentation/pull/269

closed 2019-06-03 by vmayoral