ros2/ros2 #719: [ROS 2 Dashing] Which versions of class_loader and pluginlib are deployed via deb package installation?

LanderU, 2019-06-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-release
- DDS implementation:
  - N/A 
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue

We're trying to update the CI for Moveit 2, using the latest version of the dashing. We've succsesfully installed the packages via `apt`. The problem appears when some packages try to link against the class_loader and pluginlib that are installed in our system, via `apt`. We always need to remove the current installed version and compile this packages again using the same versions that are linked on ros2.repos.

* https://github.com/AcutronicRobotics/moveit2/blob/master/moveit2.repos#L66-L73

#### Expected behavior

Compiles without issues with the packages installed.

#### Actual behavior

```bash
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `pluginlib::ClassLoader<move_group::MoveGroupCapability>::getRegisteredLibraries[abi:cxx11]()':
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE22getRegisteredLibrariesB5cxx11Ev[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE22getRegisteredLibrariesB5cxx11Ev]+0x20): undefined reference to `class_loader::MultiLibraryClassLoader::getRegisteredLibraries[abi:cxx11]() const'
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `pluginlib::ClassLoader<move_group::MoveGroupCapability>::unloadLibraryForClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE21unloadLibraryForClassERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE21unloadLibraryForClassERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0xb5): undefined reference to `class_loader::MultiLibraryClassLoader::unloadLibrary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `pluginlib::ClassLoader<move_group::MoveGroupCapability>::loadLibraryForClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE19loadLibraryForClassERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE19loadLibraryForClassERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x17b): undefined reference to `class_loader::MultiLibraryClassLoader::loadLibrary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `pluginlib::ClassLoader<move_group::MoveGroupCapability>::getAllLibraryPathsToTry(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x484): undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()'
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x4ed): undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()'
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x757): undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()'
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x774): undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()'
list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x796): undefined reference to `class_loader::systemLibrarySuffix[abi:cxx11]()'
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o:list_capabilities.cpp:(.text._ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_[_ZN9pluginlib11ClassLoaderIN10move_group19MoveGroupCapabilityEE23getAllLibraryPathsToTryERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESB_]+0x7b3): more undefined references to `class_loader::systemLibrarySuffix[abi:cxx11]()' follow
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > class_loader::impl::getAvailableClasses<move_group::MoveGroupCapability>(class_loader::ClassLoader const*)':
list_capabilities.cpp:(.text._ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE[_ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE]+0x2b): undefined reference to `class_loader::impl::getPluginBaseToFactoryMapMapMutex()'
list_capabilities.cpp:(.text._ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE[_ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE]+0x73): undefined reference to `class_loader::impl::getFactoryMapForBaseClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
list_capabilities.cpp:(.text._ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE[_ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE]+0x142): undefined reference to `class_loader::impl::AbstractMetaObjectBase::isOwnedBy(class_loader::ClassLoader const*) const'
list_capabilities.cpp:(.text._ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE[_ZN12class_loader4impl19getAvailableClassesIN10move_group19MoveGroupCapabilityEEESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EEPKNS_11ClassLoaderE]+0x150): undefined reference to `class_loader::impl::AbstractMetaObjectBase::isOwnedBy(class_loader::ClassLoader const*) const'
CMakeFiles/list_move_group_capabilities.dir/src/list_capabilities.cpp.o: In function `pluginlib::ClassLoader<move_group::MoveGroupCapability>::isClassLoaded(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
[... cut ...]
Version: 2.3.1-1bionic.20190531.005319
Depends: ros-dashing-ament-index-cpp, ros-dashing-class-loader, ros-dashing-rcutils, ros-dashing-tinyxml2-vendor, ros-dashing-ros-workspace
Description: The pluginlib package provides tools for writing and dynamically loading plugins using the ROS build infrastructure.
 To work, these tools require plugin providers to register their plugins in the package.xml of their package.
Description-md5: 174d9ecc2e9af226f45303394420e880
Homepage: http://www.ros.org/wiki/pluginlib
```

Thanks in advance!

ahcorde, 2019-06-03:
if I compare the `class_loader.so` installed with the packages with the one compiled from `dashing branch` I had these differences:

### packages

```
readelf -Ws  /opt/ros/dashing/lib/libclass_loader.so | grep getAll
    59: 000000000000c220   174 FUNC    GLOBAL DEFAULT   12 _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
    79: 0000000000008e10   447 FUNC    GLOBAL DEFAULT   12 _ZN12class_loader4impl32getAllLibrariesUsedByClassLoaderB5cxx11EPKNS_11ClassLoaderE
```

### sources

```
readelf -Ws  ./install/lib/libclass_loader.so | grep getAll
   357: 00000000000396ee   226 FUNC    GLOBAL DEFAULT   12 _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
   722: 000000000002d9e3   477 FUNC    GLOBAL DEFAULT   12 _ZN12class_loader4impl32getAllLibrariesUsedByClassLoaderB5cxx11EPKNS_11ClassLoaderE
   752: 00000000000396ee   226 FUNC    GLOBAL DEFAULT   12 _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
   835: 000000000002d9e3   477 FUNC    GLOBAL DEFAULT   12 _ZN12class_loader4impl32getAllLibrariesUsedByClassLoaderB5cxx11EPKNS_11ClassLoaderE
```

ahcorde, 2019-06-03:
Using "fat" archive is returning the same output as when I compiled from sources:

### "fat" archive

```
    59: 000000000000c250   174 FUNC    GLOBAL DEFAULT   12 _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
    79: 0000000000008e40   447 FUNC    GLOBAL DEFAULT   12 _ZN12class_loader4impl32getAllLibrariesUsedByClassLoaderB5cxx11EPKNS_11ClassLoaderE
   205: 000000000000c250   174 FUNC    GLOBAL DEFAULT   12 _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
   224: 0000000000008e40   447 FUNC    GLOBAL DEFAULT   12 _ZN12class_loader4impl32getAllLibrariesUsedByClassLoaderB5cxx11EPKNS_11ClassLoaderE

```

ahcorde, 2019-06-04:
I think we can focus in the error

 ```
/usr/bin/ld: CMakeFiles/move_group.dir/src/move_group.cpp.o: referencia sin definir al símbolo '_ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv'
//opt/ros/dashing/lib/libclass_loader.so: error adding symbols: DSO missing from command line
```

If we use the "fat" archive version or when we compile it from sources it works. It seems that something is not been well deployed with the packages.

Can you gives us some idea of how to provide you a minimal reproduction setup?

nuclearsandwich, 2019-06-04:
> Can you gives us some idea of how to provide you a minimal reproduction setup?

If you link me to a single example package that works when built against pluginlib from source or the archives but fails when using the debs that will be much easier for me than replicating all of MoveIt's CI.

LanderU, 2019-06-06:
@nuclearsandwich, finally we figure out how to fix the issue. The problem were in our package.xml. Now we don't need to compile from sources the packages. Thank you for your support!

closed 2019-06-06 by LanderU

gavanderhoorn, 2019-06-06:
@LanderU: what was the cause in the end?

LanderU, 2019-06-06:
This was the key: https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/package.xml#L88