ros2/ros2 #720: Add type enforcement of parameters

ivanpauno, 2019-06-03:
## Feature request

#### Feature description
Meta ticket for discussing how to add type enforcement of parameters both in rclcpp and rclpy.
See discussion in [range enforcement PR](https://github.com/ros2/rclcpp/issues/728).

##### Summary of the previous discussion

Currently, the [parameter descriptor message](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterDescriptor.msg) has a `type` field, which is updated each time the parameter is set. For adding type enforcement, some different proposals have been done:

- Add a new field to the parameter descriptor, called `allowed_types`. The `type` field will continue working as before, and `allowed_types` is a vector with the admitted types. (api compatible but breaking abi)
- Similar to the above option, but the member name is `allowed_type`. Only one type can be specified. (api compatible but breaking abi)
- Change the `type` field meaning. It's not more updated, and it's used for type enforcing. When the specified `type` is `PARAMETER_NO_TYPE`, type enforcement is not applied (this could be applied in the other options). (breas abi and api)
- Do not apply type enforcement in undeclared parameters, and continue updating the `type` field in that case. Stop updating it with declared parameters, and use it for type enforcement. (breaks api)
- Add an `is_type_enforced` field to the parameter descriptor (default to false). When it is `false`, type enforcement is not applied, and `type` field is updated as before. When it is `true`, it's used for type enforcement (no need to update it, as the type is enforced). (api compatible, but breaking abi)

#### Implementation considerations
Any of the options should be easy to implement both in `rclcpp` and `rclpy`.

ivanpauno, 2019-06-03:
cc @jacobperron @wjwwood @jubeira @hidmic @clalancette @tfoote.
If something is missing in the discussion summary, let me know and I will update it.

Although my original proposal was the third member of the list, I suggest going on with the last option (which does not break api).

jubeira, 2019-06-03:
I'd propose to discuss this in our next meeting.
IMO:
- For most use-cases, allowing a single type should do the job as @tfoote [mentioned](https://github.com/ros2/rclcpp/issues/728#issuecomment-495815119).
  - Having a way to bypass type enforcement explicitly would be a nice to have (shouldn't be the recommended option though). Then, a user would be able to enforce more than one type with the callback in case it's really required.
  - If we are enforcing the type defined in the descriptor, having a variable with the actual type and a variable with the enforced type feels a bit redundant, because the actual type and the enforced type should always be the same by design. I know this wouldn't be the case if we are bypassing type enforcement (e.g. using [PARAMETER_NOT_SET](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterType.msg#L1)), but since this shouldn't be the recommended case and that the type can be retrieved in a [different way](https://github.com/ros2/rclcpp/issues/728#issuecomment-494982912), I would be fine with changing the meaning of `type` for this particular case.

In other words, if a slight change in the meaning of `type` is acceptable, I would use the existing `type` inside `ParameterDescriptor` to enforce the parameter type to avoid adding more fields to the message.
- When type is enforced, the enforced type will be the actual parameter type (which doesn't change the current meaning of the variable). This should be the most common use-case and the recommended one.
- When the type is not enforced, `type` will not give any information about the actual type. This would change the meaning of the current `type` variable. The actual type information can be retrieved using `GetParameterTypes` service.

If that change of meaning for `type` is not acceptable, I'd be fine with either of the other options or new proposals.

jacobperron, 2019-06-04:
> 5. Add an is_type_enforced field to the parameter descriptor (default to false). When it is false, type enforcement is not applied, and type field is updated as before. When it is true, it's used for type enforcement (no need to update it, as the type is enforced). (api compatible, but breaking abi)

This option makes the most sense to me. We keep the same intention of `type` and are backwards compatible. But, maybe it is better to go the recommended way and enforce the type by default. This would mean that we postpone releasing this until Eloquent (or just flip the switch after backporting to Dashing).

ivanpauno, 2019-06-25:
@wjwwood Friendly ping.

alsora, 2019-08-16:
> Change the type field meaning. It won't be updated anymore when a parameter is set, and it would be used for type enforcing. When the specified type is PARAMETER_NO_TYPE, type enforcement is not applied (this could be applied in the other options). 

One of the concerns for this release was that it breaks APIs. 
Now that this would go into Eloquent, is the option 3 more interesting for you?

clalancette, 2021-05-24:
@ivanpauno Now that we are enforcing types, can we close this one out?

jacobperron, 2021-05-25:
I'm pretty sure we can close this.

clalancette, 2021-05-25:
All right, will do.  Feel free to reopen if there is further work we need for this feature.

closed 2021-05-25 by clalancette