ros2/ros2 #721: armhf support is incomplete

nuclearsandwich, 2019-06-03:
For ROS 2 Dashing we have set armhf as a tier 2 platform with nightly build and binary archive support provided by https://ci.ros2.org/ as discovered over the weekend the armhf configuration on that service is incomplete and the last couple weeks of armhf builds and archives are actually being built in a 64-bit ARM environment.

I have removed the package labeled armhf from the first [Dashing release on GitHub](https://github.com/ros2/ros2/releases/tag/release-dashing-20190531) to avoid confusing folks and opened https://github.com/ros2/ci/pull/299 which gets builds running on armhf in earnest. However there are some unresolved issues there.

- [ ] Resolve warnings on armhf ([test CI warnings report](https://ci.ros2.org/view/All/job/test_ci_linux-armhf/8/warnings23Result/new/))

- [ ] Resolve GCC ABI warnings throughout:

```
In file included from /usr/include/c++/7/vector:63:0,
                from /home/jenkins-agent/workspace/test_ci_linux-armhf/ws/build/osrf_testing_tools_cpp/googletest-1.8.0-extracted/googletest-1.8.0-src/googletest/include/gtest/gtest.h:56,
                from /home/jenkins-agent/workspace/test_ci_linux-armhf/ws/build/osrf_testing_tools_cpp/googletest-1.8.0-extracted/googletest-1.8.0-src/googletest/src/gtest-all.cc:39:
/usr/include/c++/7/bits/stl_uninitialized.h: In function ‘_ForwardIterator std::__uninitialized_copy_a(_InputIterator, _InputIterator, _ForwardIterator, std::allocator<_Tp>&) [with _InputIterator = __gnu_cxx::__normal_iterator<const double*, std::vector<double> >; _ForwardIterator = double*; _Tp = double]’:
/usr/include/c++/7/bits/stl_uninitialized.h:287:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<const double*, std::vector<double> >’ changed in GCC 7.1
    __uninitialized_copy_a(_InputIterator __first, _InputIterator __last,
    ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/7/bits/stl_uninitialized.h:287:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<const double*, std::vector<double> >’ changed in GCC 7.1
/usr/include/c++/7/bits/stl_uninitialized.h:289:63: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<const double*, std::vector<double> >’ changed in GCC 7.1
    { return std::uninitialized_copy(__first, __last, __result); }
                                                              ^
```

emersonknapp, 2019-06-04:
Mentioning https://github.com/eProsima/Fast-CDR/issues/52, which has been opened to track the remaining warnings in Fast-CDR

emersonknapp, 2019-06-05:
Opened https://github.com/eProsima/Fast-CDR/pull/53 - hoping to have a bit of discussion about it though before merging - it seems good but I'm not 100% convinced of the completeness of the fix.

Running [image] to see if it goes green

emersonknapp, 2019-06-05:
Feeling better about the change after another revision - 

* Linux-armhf [image]

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

prajakta-gokhale, 2019-06-05:
> * Resolve GCC ABI warnings throughout:
>
> ```
> In file included from /usr/include/c++/7/vector:63:0,
>                 from [...]

I and @thomas-moulard had a discussion about the logged `note:`s on incompatible ABI and here are our thoughts:
* The `note` indicates that code compiled with gcc versions 6.x and prior **may** not work correctly when called from code built with 7.x and vice versa, and not a problem with the generated binary
* There won't be a change in the `gcc` compiler major version during the lifetime of this distro, so there won't be any new incompatibilities
* Having this `note` in the build log file would be useful for the package maintainers, as it will serve as a reminder to not use/bundle this package with another that uses a different `gcc` version with an incompatible ABI

If the logs get too verbose for the build farm, we can consider suppressing the `note:`s by using the ```-Wno-psabi``` flag if required.

clalancette, 2019-06-06:
>     * The `note` indicates that code compiled with gcc versions 6.x and prior **may** not work correctly when called from code built with 7.x and vice versa, and not a problem with the generated binary

I guess this is the part that confuses me the most.  Does gcc 7.x really spew this warning every single time you call one of these methods?  Answering my own question, it does look like this is the case: https://github.com/gcc-mirror/gcc/blob/e741947/gcc/config/arm/arm.c#L6501 (at least for containers with doubles).  With that being the case, I'd strongly suggest that we use `-Wno-psabi`, since it is not an actionable item and just noise as far as we are concerned.

prajakta-gokhale, 2019-06-06:
> I'd strongly suggest that we use `-Wno-psabi`, since it is not an actionable item and just noise as far as we are concerned.

My only argument for keeping it around was that it may be good to remember later in case we add more libraries in future distros.
But you're right, it is not actionable right now and we can create a colcon mixin to enable the flag that suppresses the notes in this build.

prajakta-gokhale, 2019-06-07:
@nuclearsandwich @clalancette I have a patch that adds `-Wno-psabi` to build arguments in this AWS fork [commit](https://github.com/aws-ros-dev/ci/commit/f322adad1c5a30e6feddf7da67d0d409e62afdb3). What would be the best way to test it on the buildfarm?

clalancette, 2019-06-07:
We don't have an easy way for our CI jobs to grab branches from forks, so I cherry-picked that commit onto https://github.com/ros2/ci/tree/suppress-armhf-abi-warn , then kicked off a build: https://ci.ros2.org/job/ci_linux-armhf/3/

emersonknapp, 2019-06-07:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
* Linux-armhf [image]

emersonknapp, 2019-06-18:
i wonder if i mention @emersonknapp myself if this will show up in my mentioned issues list

emersonknapp, 2019-06-21:
Linux-armhf on master: [image]

christianrauch, 2019-07-04:
Are there plans to support Debian / Raspbian as armhf target? Raspbian has a large user base and many Raspberry Pi users would benefit from a binary distribution of ROS2.

emersonknapp, 2019-07-04:
Due to https://github.com/ros2/ros2/issues/735 I don't think there will be raspbian stretch. However, the armhf builds https://ci.ros2.org/view/packaging/job/packaging_linux-armhf/ should work fine on buster, assuming you install the dependencies via rosdep.

christianrauch, 2019-07-04:
Debian/Raspbian `buster` provides python 3.7, i.e. this should not be an issue.

The CI output `ros2-package-linux-armhf.tar.bz2` does not mention the target distribution. I assume it is Ubuntu Bionic and I don't think that Debian buster and Ubuntu Bionic are binary compatible. E.g. Debian buster is missing `libPocoFoundation.so.50`. And even after working around this with a symlink to `libPocoFoundation.so.60`, it is still compiled for the wrong architecture, e.g. the `talker` example exits with `Illegal instruction`.

I think that Debian/Raspbian buster would need its dedicated CI job.

nuclearsandwich, 2019-07-15:
@christianrauch there's a separate discussion about Debian support [on ROS Discourse](https://discourse.ros.org/t/debian-as-a-tier-1-platform-for-ros-2/9431/2). As @gerkey mentioned there Debian support will need to be provided by the community. 

> The CI output `ros2-package-linux-armhf.tar.bz2` does not mention the target distribution. I assume it is Ubuntu Bionic and I don't think that Debian buster and Ubuntu Bionic are binary compatible. E.g. Debian buster is missing `libPocoFoundation.so.50`.

It is indeed Ubuntu Bionic. It's hard to spot if you're not looking for it but it is visible in the job parameters. Buster and Bionic are not binary compatible. As Debian is currently a [tier 3 platform](http://www.ros.org/reps/rep-2000.html#support-tiers) our regular CI does not support it but from-source builds were reported working on previous versions of Debian.

>  it is still compiled for the wrong architecture, e.g. the `talker` example exits with `Illegal instruction`.

Prior to deploying https://github.com/ros2/ci/pull/299 armhf builds were actually just armv8 builds with a different job name. Now that the correctly targeted binaries are available you should be able to use them with Ubuntu Bionic.

closed 2019-07-18 by Karsten1987