ros2/ros2 #724: Fail if `BUILD_TESTING` is set and test dependencies are missing

sloretz, 2019-06-05:
Some cmake packages check if test targets exist before trying to use them.
They do this because macros like `ament_add_pytest()` [only create the test target if `pytest` is found](https://github.com/ament/ament_cmake/blob/3c432f29b80fd203883a5d83b2b4db3bfb015924/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake#L73-L79), even if `BUILD_TESTING` is set.
This makes it possible for a CI machine to be misconfigured (missing `pytest`), but tests CI jobs will still succeed.

## 

cmake macros with this behavior
* `ament_add_pytest()` [with pytest](https://github.com/ament/ament_cmake/blob/3c432f29b80fd203883a5d83b2b4db3bfb015924/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake#L73-L79)
* `ament_add_nose_test()` [with nose](https://github.com/ament/ament_cmake/blob/3c432f29b80fd203883a5d83b2b4db3bfb015924/ament_cmake_nose/cmake/ament_add_nose_test.cmake#L46-L49)
* `ament_add_gmock()` [with gmock](https://github.com/ament/ament_cmake/blob/3c432f29b80fd203883a5d83b2b4db3bfb015924/ament_cmake_gmock/cmake/ament_add_gmock.cmake#L48-L51)
* `ament_add_gtest()` [with gtest](https://github.com/ament/ament_cmake/blob/3c432f29b80fd203883a5d83b2b4db3bfb015924/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake#L35-L38)

Packages ignoring missing test targets
* [rcutils](https://github.com/ros2/rcutils/blob/e49e277efb2e58cc9df664827b16f519cc6d9c95/CMakeLists.txt#L224)
* [rcl](https://github.com/ros2/rcl/blob/8085c6adbf30d76f8073632a4e2d2452aac366d9/rcl/test/cmake/rcl_add_custom_gtest.cmake#L77-L80)
* [rclcpp](https://github.com/ros2/rclcpp/blob/24080a458dd6cc227e29a97127209acca1ceb060/rclcpp/CMakeLists.txt#L148-L149)
* [test_cli](https://github.com/ros2/system_tests/blob/a245038d9087b9f6e9bb0e73ba603b88f9836a55/test_cli/CMakeLists.txt#L44)
* [intra_process_demo](https://github.com/ros2/demos/blob/c88aa8e8c859fbde49b3638cf1105fb585a8738a/intra_process_demo/CMakeLists.txt#L144)
* [pendulum_control](https://github.com/ros2/demos/blob/c88aa8e8c859fbde49b3638cf1105fb585a8738a/pendulum_control/CMakeLists.txt#L125)
* [test_communication](https://github.com/ros2/system_tests/blob/a245038d9087b9f6e9bb0e73ba603b88f9836a55/test_communication/CMakeLists.txt#L433)
* [test_rclcpp](https://github.com/ros2/system_tests/blob/a245038d9087b9f6e9bb0e73ba603b88f9836a55/test_rclcpp/CMakeLists.txt#L168)
* [test_security](https://github.com/ros2/system_tests/blob/a245038d9087b9f6e9bb0e73ba603b88f9836a55/test_security/CMakeLists.txt#L255)
* [rosidl_generator_cpp](https://github.com/ros2/rosidl/blob/66f9a0e92962778edddc2ce6522ec259d7a76331/rosidl_generator_cpp/CMakeLists.txt#L91)
* [rclcpp_components](https://github.com/ros2/rclcpp/blob/24080a458dd6cc227e29a97127209acca1ceb060/rclcpp_components/CMakeLists.txt#L94)
* [rclcpp_action](https://github.com/ros2/rclcpp/blob/24080a458dd6cc227e29a97127209acca1ceb060/rclcpp_action/CMakeLists.txt#L72)
* [rclcpp_lifecycle](https://github.com/ros2/rclcpp/blob/24080a458dd6cc227e29a97127209acca1ceb060/rclcpp_lifecycle/CMakeLists.txt#L51)
* [rttest](https://github.com/ros2/realtime_support/blob/d9c7f16613d1a5a848df1beff71531d61269fd19/rttest/CMakeLists.txt#L39)
* [tlsf_cpp](https://github.com/ros2/realtime_support/blob/d9c7f16613d1a5a848df1beff71531d61269fd19/tlsf_cpp/CMakeLists.txt#L51)
* [rviz_rendering](https://github.com/ros2/rviz/blob/a20152c290739c4cf2ef5af06a7e498a744162ac/rviz_rendering_tests/CMakeLists.txt#L71)
* [rviz_default_plugins](https://github.com/ros2/rviz/blob/a20152c290739c4cf2ef5af06a7e498a744162ac/rviz_default_plugins/CMakeLists.txt#L332)
* [rviz_common](https://github.com/ros2/rviz/blob/a20152c290739c4cf2ef5af06a7e498a744162ac/rviz_common/CMakeLists.txt#L387)
* [tf2](https://github.com/ros2/geometry2/blob/234725f30ade83caa4bcf019ee854199aa31bef5/tf2/CMakeLists.txt#L43)
* [tf2_eigen](https://github.com/ros2/geometry2/blob/234725f30ade83caa4bcf019ee854199aa31bef5/tf2_eigen/CMakeLists.txt#L26)
* [tf2_geometry_msgs](https://github.com/ros2/geometry2/blob/234725f30ade83caa4bcf019ee854199aa31bef5/tf2_geometry_msgs/CMakeLists.txt#L37)
* [sensor_msgs](https://github.com/ros2/common_interfaces/blob/292eb2b1ff871c25baaa4156dfde8fd789be922b/sensor_msgs/CMakeLists.txt#L72)
* [message_filters](https://github.com/ros2/message_filters/blob/a18a34a2242e6d44dd6c66e4cb2679d0fe07a40b/CMakeLists.txt#L70)
* [rmw](https://github.com/ros2/rmw/blob/dc7b2f49f1f961d6cf2c173adc54736451be8938/rmw/test/CMakeLists.txt#L17)
* [rosbag2_transport](https://github.com/ros2/rosbag2/blob/8c38b88a73910b7eeb40f39ae1c7b2972353a64a/rosbag2_transport/CMakeLists.txt#L90)
* [rosbag2_converter_default_plugins](https://github.com/ros2/rosbag2/blob/8c38b88a73910b7eeb40f39ae1c7b2972353a64a/rosbag2_converter_default_plugins/CMakeLists.txt#L68)
* [rosbag2_storage](https://github.com/ros2/rosbag2/blob/8c38b88a73910b7eeb40f39ae1c7b2972353a64a/rosbag2_storage/CMakeLists.txt#L83)
* [rosbag2_storage_default_plugins](https://github.com/ros2/rosbag2/blob/8c38b88a73910b7eeb40f39ae1c7b2972353a64a/rosbag2_storage_default_plugins/CMakeLists.txt#L80)
* [rosbag2](https://github.com/ros2/rosbag2/blob/8c38b88a73910b7eeb40f39ae1c7b2972353a64a/rosbag2/CMakeLists.txt#L99)
* [rcl_action](https://github.com/ros2/rcl/blob/8085c6adbf30d76f8073632a4e2d2452aac366d9/rcl_action/CMakeLists.txt#L88)
* [rcl_lifecycle](https://github.com/ros2/rcl/blob/8085c6adbf30d76f8073632a4e2d2452aac366d9/rcl_lifecycle/CMakeLists.txt#L72)
* [rcl_yaml_param_parser](https://github.com/ros2/rcl/blob/8085c6adbf30d76f8073632a4e2d2452aac366d9/rcl_yaml_param_parser/CMakeLists.txt#L50)

### Option 1: Fail if pytest/gtest/gmock/nose isn't found

This option would cause cmake to fail at configure time if it is missing one of these dependencies.
It treats the test frameworks as any other required dependency.
However, if a necessary framework is missing then nothing would be built as the failure happens at configure time.

### Option 2: Create a test target that is marked as disabled or skipped

@dirk-thomas mentioned an option would be to create a test that is marked as skipped or disabled. This might mean calling `add_test()` and setting a test property. Starting in CMake 3.8 there is [DISABLED](https://cmake.org/cmake/help/v3.9/prop_test/DISABLED.html), but I don't know what is available on earlier versions.

### Option 3: Create a test target that always fails

This option would make the cmake macros always return a test target when `BUILD_TESTING` is set, but the command would fail. This could be done by calling `add_test()`, and setting an impossible [PASS_REGULAR_EXPRESSION](https://cmake.org/cmake/help/v3.5/prop_test/PASS_REGULAR_EXPRESSION.html).

dirk-thomas, 2019-06-05:
Imo option 1 isn't desired. In the past we had the case that users lacked one of the testing framework and because of that couldn't even configure the other tests.

Option 3 will make it fairly hard to find actual test failures in the noise.

The `DISABLED` flag of option 2 looks like the most appealing one to me. Since the feature is available in Ubuntu Bionic (but not in Debian Stretch) it could simply be used conditionally if it is available in the used CMake version. If the CMake version doesn't support the flag it would gracefully fallback to option 3.

clalancette, 2019-06-05:
> The `DISABLED` flag of option 2 looks like the most appealing one to me. Since the feature is available in Ubuntu Bionic (but not in Debian Stretch) it could simply be used conditionally if it is [...]

If we go this route, I think it is also important that Jenkins reports this as a "failed" test, so it is easy for the build cop to notice one of these things broke.

mikaelarguedas, 2019-06-05:
>  This makes it possible for a CI machine to be misconfigured (missing pytest), but tests CI jobs will still succeed.

To clarify the current behavior: for each target not created because of a missing test suite, a CMake Warning is generated.
So if a CI machine is missing one of these dependencies, the jobs will neither succeed nor fail, they will be reported as unstable with as many CMake warnings as test targets non generated.
It does not provide the list of faulty tests.

How it compares to the options above:

Option 1: Louder failures: job fails and stops at first error,  doesn't provide the list of faulty tests either
Option 2: Silent failure: job comes back green but there is (should be?) a way to get the list of skipped tests (e.g. xmllint tests were `DISABLED` on Windows for a long time using DISABLED)
Option 3: Same level of noise: job comes back yellow with the faulty tests all marked as failed

As a third-party user leveraging other CI platforms, I like the current behavior or option 3 as they both allow all my packages to compile and the available tests to run while reporting the fact that some things didn't go as expected.
For option 2, if `colcon test-result` provides an option to return non-zero on skipped/DISABLED tests that will be good to.

dirk-thomas, 2019-06-05:
The `Test.xml` file generated by `CTest` does list the disabled test with a `Status` attribute of `notrun`.

`colcon test-result` reports that test as `skipped`.

Jenkins not picking this up at the moment should be re-evaluated after updating the Jenkins plugin `xunit` to the latest release which comes with a lot of changes / improvements. Maybe the new version will show the failed test somehow.