ros2/ros2 #725: python3-lark-parser not available for Debian 9 Stretch

artizirk, 2019-06-06:
## Bug report

**Required Info:**

- Operating System:
  - Debian 9 Stretch
- Installation type:
  - Source
- Version or commit hash:
  - Dashing

#### Steps to reproduce issue

Follow From Source install instructions from https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/

https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/#install-dependencies-using-rosdep

When running `rosdep install` step then it says that `python3-lark-parser` is not available

This can be bypassed with manually installing Ubuntu version of this package from [deb repo](http://packages.ros.org/ros2/ubuntu/pool/main/l/lark-parser/) and adding `python3-lark-parser` to `--skip-keys` argument

#### Expected behavior
`python3-lark-parser` is installed
#### Actual behavior
`python3-lark-parser` is not packaged for Debian 9 Stretch and `rosdep` fails

dirk-thomas, 2019-06-06:
Please consider to contribute a note to the install instructions mentioning this for Debian: https://github.com/ros2/ros2_documentation/blob/master/source/Installation/Dashing/Linux-Development-Setup.rst

sousou1, 2019-07-14:
@dirk-thomas 

I have this problem too.
I skipped the python3-lark-parser. However, I got such an error in running ros2.
Is it difficult to add python3-lark-parser to the stretch package?
```
# ros2
Failed to load entry point 'test': invalid syntax (io_handler.py, line 137)
Failed to load entry point 'launch': invalid syntax (__init__.py, line 60)
Failed to load entry point 'send_goal': invalid syntax (executors.py, line 148)
Failed to load entry point 'show': invalid syntax (executors.py, line 148)
Failed to load entry point 'info': invalid syntax (executors.py, line 148)
Failed to load entry point 'list': invalid syntax (executors.py, line 148)
Failed to load entry point 'play': invalid syntax (node.py, line 133)
Failed to load entry point 'record': invalid syntax (node.py, line 133)
Failed to load entry point 'unload': invalid syntax (node.py, line 133)
Failed to load entry point 'standalone': invalid syntax (node.py, line 133)
Failed to load entry point 'types': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'load': invalid syntax (node.py, line 133)
Failed to load entry point 'status': invalid syntax (node.py, line 133)
Failed to load entry point 'start': invalid syntax (node.py, line 133)
Failed to load entry point 'stop': invalid syntax (node.py, line 133)
Failed to load entry point 'nodes': invalid syntax (node.py, line 133)
Failed to load entry point 'set': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'get': invalid syntax (node.py, line 133)
Failed to load entry point 'info': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'delete': invalid syntax (node.py, line 133)
Failed to load entry point 'set': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'get': invalid syntax (node.py, line 133)
Failed to load entry point 'generate_policy': invalid syntax (node.py, line 133)
Failed to load entry point 'call': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'pub': invalid syntax (node.py, line 133)
Failed to load entry point 'list': invalid syntax (node.py, line 133)
Failed to load entry point 'delay': invalid syntax (node.py, line 133)
Failed to load entry point 'info': invalid syntax (node.py, line 133)
Failed to load entry point 'hz': invalid syntax (node.py, line 133)
Failed to load entry point 'echo': invalid syntax (node.py, line 133)
Failed to load entry point 'bw': invalid syntax (node.py, line 133)
```

ivanpauno, 2019-07-15:
@sousou1 we are not supporting python 3.5 in Dashing (see https://github.com/ros2/ros2/issues/735).
The only changes that are needed for supporting it is changing variable annotation syntax (e.g. https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py#L136, https://github.com/ros2/launch/blob/3d34d3b50e43e9236fcfc5804966eee67e4d4c77/launch/launch/launch_description_source.py#L52) to annotation in comments (e.g.: https://github.com/ros2/launch/blob/d5c5f54b030e8795e3376c6ce80db5f6d3bdfe89/launch/launch/actions/execute_process.py#L223).
We will accept PRs to backport to Dashing.

nuclearsandwich, 2019-07-31:
The package has been imported to our Stretch repositories.

montanier, 2019-08-29:
Hello,

I think I still have the same issue. The setup is the same. 

Here is the command and the result:
``
$ rosdep install --from-paths src --ignore-src --rosdistro dashing -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers" --os=debian:stretch
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rosidl_parser: No definition of [python3-lark-parser] for OS [debian]
``

nuclearsandwich, 2019-08-29:
The package is there. The problem is that a rosdep definition for Debian Stretch is missing. https://github.com/ros/rosdistro/pull/20666/ is currently under review.

dirk-thomas, 2019-08-29:
I will reopen this until the rosdep key is merged and the package is also available in Buster.

sousou1, 2019-09-06:
@dirk-thomas 
Like rosdistro, I want to register python3-lark-parser binary package for Buster by sending a PR.

Where does Buster manage this [repository](http://packages.ros.org/ros2/ubuntu/dists/buster/) ?(not rosdistro?)

nuclearsandwich, 2019-09-13:
This package is now in the Stretch and Buster repositories and with https://github.com/ros/rosdistro/pull/22234 rosdep keys have been added.

nuclearsandwich, 2019-09-13:
> Where does Buster manage this [repository](http://packages.ros.org/ros2/ubuntu/dists/buster/) ?(not rosdistro?)

The repositories are not managed via Git. The import_upstream jobs on the buildfarm are configured to pull from the ROS bootstrap repository which is maintained by hand.

closed 2019-09-13 by nuclearsandwich

sousou1, 2019-09-15:
Noted with thanks.

yuriachermann, 2024-08-12:
I'm having a problem with that:

```
$ colcon build --cmake-clean-first
Starting >>> px4_msgs
--- stderr: px4_msgs
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 21, in <module>
    from rosidl_generator_type_description import generate_type_hash
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_generator_type_description/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
gmake[2]: *** [CMakeFiles/px4_msgs__rosidl_generator_type_description.dir/build.make:441: rosidl_generator_type_description/px4_msgs/msg/ActionRequest.json] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:215: CMakeFiles/px4_msgs__rosidl_generator_type_description.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_msgs [12.9s, exited with code 2]

Summary: 0 packages finished [13.1s]
  1 package failed: px4_msgs
  1 package had stderr output: px4_msgs
  1 package not processed
```