ros2/ros2 #726: Custom Srvs that use custom msgs in the same package build fails unless install and build folders are deleted.

EpicZa, 2019-06-06:
## Bug report
- Operating System:
  Windows 10
- Installation type:
  From source
- Version or commit hash:
  -repos file
- DDS implementation:
  - rti

Create a new package with a custom msg and custom srv folder, make the service use the massage type.

cmake_minimum_required(VERSION 3.5)

project(bug_sim)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(builtin_interfaces REQUIRED)
find_package(rosidl_default_generators REQUIRED)

set(msg_files
	"msg/PlayerState.msg"
)

set(srv_files
	"srv/SetPlayerState.srv"
)

rosidl_generate_interfaces(bug_sim
	${msg_files}
	${srv_files}
	DEPENDENCIES builtin_interfaces
)

ament_export_dependencies(rosidl_default_runtime)

ament_package()
---
note SetPlayerState.sv uses the msg as:
bug_sim/PlayerState ps

#### Expected behavior
Builds.
#### Actual behavior
Initial build will succeed, but if you build the msg first, commenting out $(srv_files) and after the custom message builds, rebuild the package with srvs included, it fails.
#### Additional information

ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\CL.exe /c /IC:\Projects\Playground\build\bug_sim\rosidl_generator_cpp /IC:\ros2\install\include /nologo /W3 /WX- /diagnostics:classic /O2 /Ob2 /D WIN32 /D _WINDOWS /D NDEBUG /D ROSIDL_TYPESUPPORT_INTROSPECTION_CPP_BUILDING_DLL /D "CMAKE_INTDIR=\"Release\"" /D bug_sim__rosidl_typesupport_introspection_cpp_EXPORTS /D _WINDLL /D _MBCS /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /std:c++14 /Fo"bug_sim__rosidl_typesupport_introspection_cpp.dir\Release\\" /Fd"bug_sim__rosidl_typesupport_introspection_cpp.dir\Release\vc141.pdb" /Gd /TP /FC /errorReport:queue C:\Projects\Playground\build\bug_sim\rosidl_typesupport_introspection_cpp\bug_sim\srv\set_cloud_layer_state__type_support.cpp
  set_cloud_layer_state__type_support.cpp
c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_introspection_cpp\bug_sim\srv\set_cloud_layer_state__type_support.cpp': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_introspection_cpp.vcxproj]
Done Building Project "C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_introspection_cpp.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default target) (3) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_generator_c.vcxproj" (default target) (5) ->
(ClCompile target) -> 
  c1 : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_generator_c\bug_sim\srv\set_cloud_layer_state__functions.c': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_generator_c.vcxproj]

"C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default target) (3) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_c.vcxproj" (default target) (7) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_cpp.vcxproj" (default target) (9) ->
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_connext_cpp\bug_sim\srv\dds_connext\set_cloud_layer_state__type_support.cpp': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_cpp.vcxproj]
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_connext_cpp\bug_sim\srv\dds_connext\SetCloudLayerState_.cxx': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_cpp.vcxproj]
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_connext_cpp\bug_sim\srv\dds_connext\SetCloudLayerState_Plugin.cxx': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_cpp.vcxproj]
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_connext_cpp\bug_sim\srv\dds_connext\SetCloudLayerState_Support.cxx': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_connext_cpp.vcxproj]

"C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default target) (3) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_cpp.vcxproj" (default target) (10) ->
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_cpp\bug_sim\srv\set_cloud_layer_state__type_support.cpp': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_cpp.vcxproj]

"C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default target) (3) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_fastrtps_c.vcxproj" (default target) (11) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_fastrtps_cpp.vcxproj" (default target) (12) ->
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_fastrtps_cpp\bug_sim\srv\dds_fastrtps\set_cloud_layer_state__type_support.cpp': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_fastrtps_cpp.vcxproj]

"C:\Projects\Playground\build\bug_sim\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Projects\Playground\build\bug_sim\bug_sim.vcxproj" (default target) (3) ->
"C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_introspection_cpp.vcxproj" (default target) (14) ->
  c1xx : fatal error C1083: Cannot open source file: 'C:\Projects\Playground\build\bug_sim\rosidl_typesupport_introspection_cpp\bug_sim\srv\set_cloud_layer_state__type_support.cpp': No such file or directory [C:\Projects\Playground\build\bug_sim\bug_sim__rosidl_typesupport_introspection_cpp.vcxproj]

jacobperron, 2019-06-07:
The error is referring to a service called `SetCloudLayerState`, which is not in your CMakeLists.txt. Is the CMakeLists.txt you've posted up-to-date?

EpicZa, 2019-06-07:
The error is from my actual project, the msgs and srvs names at the start were examples, and I ctrl+f'ed the project name so it is less identifying.

dirk-thomas, 2019-06-11:
This does work for me on Linux. Can you please share the output of the build command after re-enabling the service file (`colcon build --event-handlers console_direct+`). Since you change the `CMakeLists.txt` file it should reconfigure CMake. Can you also try to pass `--cmake-force-configure`.

dirk-thomas, 2019-06-12:
I can confirm the problem on Windows. The problem affects multiple generators. The problem is that the [custom command](https://github.com/ros2/rosidl/blob/9ffe2148b9c249b8539a27745062d6c2c1716f2b/rosidl_generator_c/cmake/rosidl_generator_c_generate_interfaces.cmake#L80-L87) isn't being triggered.

I double checked that the `OUTPUT` files of the custom command do contain the ones for the new service. But CMake still doesn't invoke the command.

Removing any of the files generated in the prrvious build for the message is sufficient to retrigger the command which then also generates the service specific files.

anacsousa1, 2021-05-14:
Hello, are there any updates related to this bug?
My current solution for a building is deleting 'build' and 'install' folders, and then it correctly builds.

jacobperron, 2021-05-17:
@anacsousa1 Since this issue is labeled as "backlog", I take it no one is currently looking at it (at least no one from the ROS 2 team).