ros2/ros2 #730: About contributing a new RMW implementation

clemjh, 2019-06-10:
Hi. We developed a new RMW implementation, `rmw_coredds`, and posted about this on [ROS Discourse](https://discourse.ros.org/t/rmw-coredds-new-rmw-implementation-for-crystal-and-dashing/9421). We would like to contribute this to official ROS2 repository. We have read this [guide](https://index.ros.org/doc/ros2/Contributing/), but still not sure what we should do to contribute a whole package.

What we would like to ask is:
- Can we contribute new packages to the official repository?
- If we can, what do we have to do to do so?

Thank you.

dirk-thomas, 2019-06-11:
For visibility we recommend to move both repos (`rmw_coredds` and `rosidl_typesupport_coredds`) into the `ros2` org unit. You would still continue to have admin access to these repos. If you want to proceed with that please transfer the repos to my personal account from where I can transfer them further. I would add you as admin to the repos so you can add other users to those repos.

For other RMW implementation we have nightly jobs ensuring they build and pass tests (http://build.ros2.org/view/Dci/). Those use publically accessible Debian packages of the vendor specific code - or build the vendor specific code from source. I assume that is not applicable to CoreDDS (if it is we are happy to set this up and if necessary work with you to create a Debian package)?

> Can we contribute new packages to the official repository?

Are you referring to the `apt` repository `packages.ros.org`? In order to build a Debian package you need to release your repositories with [bloom](https://index.ros.org/doc/ros2/Tutorials/Releasing-a-ROS-2-package-with-bloom/). But again the requirement would be that its dependencies (including CoreDDS) needs to be available as Debian packages too.

For the big archives we create on macOS and Windows we have the build machines provisioned with the publically downloadable version of the vendor packages. Again this might not apply to CoreDDS?

> If we can, what do we have to do to do so?

Independent of the packaging question I would recommend to write a page containing instructions how to build / setup / use ROS 2 with CoreDDS. You can either contribute that page beside the existing install instruction ([here](https://github.com/ros2/ros2_documentation/tree/master/source/Installation), either in a distro agnostic way or specific pages for each distro) or you can of course host them anywhere else. The advantage of the former option is that the content is searchable on index.ros.org.

clemjh, 2019-06-12:
Thank you for your response.

> For visibility we recommend to move both repos (rmw_coredds and rosidl_typesupport_coredds) into the ros2 org unit.

We will transfer the repos to your account soon. We appreciate your help.

> For other RMW implementation we have nightly jobs ensuring they build and pass tests.

I'm afraid those jobs are not applicable for now. Since we aim for [Tier 3](http://www.ros.org/reps/rep-2000.html#support-tiers) for now, I assume this won't be a serious problem?

> In order to build a Debian package you need to release your repositories with bloom. But again the requirement would be that its dependencies (including CoreDDS) needs to be available as Debian packages too.

Since CoreDDS is not available as a debian package (We are working on it and it will be done very soon), I think building Debian packages will take some time.

> Independent of the packaging question I would recommend to write a page containing instructions how to build / setup / use ROS 2 with CoreDDS.

We will prepare the instructions soon.

Again, thank you for your answers.

dirk-thomas, 2019-06-12:
> We will transfer the repos to your account soon.

* https://github.com/ros2/rmw_coredds
* https://github.com/ros2/rosidl_typesupport_coredds

> I'm afraid those jobs are not applicable for now. Since we aim for Tier 3 for now, I assume this won't be a serious problem?

Sounds good - no problem.

> Since CoreDDS is not available as a debian package (We are working on it and it will be done very soon), I think building Debian packages will take some time.

Great, let us know if we can help.

> We will prepare the instructions soon.

Thank you.