ros2/ros2 #732: Error: 'charmap' codec can't decode byte 0x8f in position 156

minggangw, 2019-06-16:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 ONLY
- Installation type:
  - Binary of nightly build (http://ci.ros2.org/view/packaging/job/packaging_windows/lastSuccessfulBuild/artifact/ws/ros2-package-windows-AMD64.zip)
- Version or commit hash:
- DDS implementation:
- Client library (if applicable):

#### Steps to reproduce issue

Pass the path of [WStrings.msg](https://github.com/ros2/test_interface_files/blob/master/msg/WStrings.msg) as the parameter of Python method [parse_message_file](https://github.com/ros2/rosidl/blob/ced307012bda3413d6c495914bf517d0d03748f1/rosidl_adapter/rosidl_adapter/parser.py#L427)

#### Expected behavior
Execute successfully
#### Actual behavior
Error: 'charmap' codec can't decode byte 0x8f in position 156: character maps to <undefined>
#### Additional information
- Python version: Python3.7 x64
- Suspection: there are some special characters though commented out, and I found possible some potential reason, see https://stackoverflow.com/questions/9233027/unicodedecodeerror-charmap-codec-cant-decode-byte-x-in-position-y-character

Thanks!

sloretz, 2019-06-17:
This sounds little bit like https://github.com/ros2/system_tests/pull/362#issue-277436162 . Maybe `WStrings.msg` needs a [byte order mark](https://en.wikipedia.org/wiki/Byte_order_mark)? Alternatively, maybe `parse_message_file()` needs to expose an `encoding` keyword arg for the call to [`open()`](https://github.com/ros2/rosidl/blob/ced307012bda3413d6c495914bf517d0d03748f1/rosidl_adapter/rosidl_adapter/parser.py#L430)?

minggangw, 2019-06-18:
Maybe it's kind of Python version of it.

dirk-thomas, 2019-07-15:
Any generated source file which contains non-ASCII characters (coming from e.g. default values in the interface file) needs to include a BOM.